wolfSSL/wolfssl #5154: Espressif ESP-IDF component install fails when target directory path has embedded spaces

gojimmypi, 2022-05-18:
### Version

current version

### Description

The [Windows Espressif ESP-IDF setup_win.bat](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/setup_win.bat) component install does not properly handle spaces in the `IDF_PATH` or command-line parameter.

For example when someone has a `first last` login name and the ESP-IDF is installed for the current user desktop directory, the install will fail. 

credit @ejohnstown for finding this. 

See also https://github.com/wolfSSL/wolfssh/pull/405

gojimmypi, 2022-06-04:
Although there are plans to address this, at this time it is currently a low priority as the [Espressif install](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/) states "_ESP-IDF does not support spaces in paths_".

[image]

PaulMartinsen, 2022-12-21:
Hi, I hit this same problem today and modified setup_win.bat to work with paths that contains spaces. I note that Espressif doesn't like spaces, but can confirm it does work for me. I made PR: #5919 . HTH.

gojimmypi, 2024-06-21:
I'm closing this lingering issue as the [Espressif docs](https://docs.espressif.com/projects/esp-idf/en/release-v5.2/esp32/get-started/windows-setup.html) clearly indicate:

[image]

Still, where possible, we'll try to support spaces. Please open a new issue if there's a specific, problematic instance.

It's also a bit less relevant now that [Managed Components](https://components.espressif.com/components/wolfssl/wolfssl) are the preferred mechanism to install wolfSSL instead of the setup script.

See also: 

https://www.wolfssl.com/wolfssl-now-available-in-espressif-component-registry/

closed 2024-06-21 by gojimmypi