wolfSSL/wolfssl #5185: I got "revcd alert fatal error" from wolfSSL_connect

kython28, 2022-05-26:
### Version

5.3.0

### Description

I am creating a client application, and I am trying to connect to: "[ws-api.kucoin.com](http://ws-api.kucoin.com/)"

´´´C
wolfSSL_library_init();
WOLFSSL_CTX *ctx = wolfSSL_CTX_new(wolfTLS_client_method());
wolfSSL_CTX_set_verify(ctx, WOLFSSL_VERIFY_NONE, 0);

// I create the fd here and then I connect ....

WOLFSSL *ssl = wolfSSL_new(ctx);
wolfSSL_set_fd(ssl, fd);

int ret = wolfSSL_connect(ssl);
if (ret != WOLFSSL_SUCCESS){
	printf("%d\n", wolfSSL_get_error(ssl, ret));

	// .....
}
´´´

When I run the above I always get: -313. I've been trying all of the WOLFSSL_METHODs, and none of them work. I always get -313 or -326

kareem-wolfssl, 2022-05-26:
Hi @kython28 ,

-326 (VERSION_ERROR) usually means you are missing support for the TLS version the site requires.  Looks like kucoin.com is using TLS1.3 + X25519 (disabled by default).  Please attach your build config (./configure or user_settings.h).  If you are missing any of the following, please add them and retry: `--enable-curve25519 --enable-tls13` (./configure)
```
#define HAVE_CURVE25519
#define WOLFSSL_TLS13
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES
```
(user_settings.h)

For WOLFSSL_METHOD, use wolfSSLv23_client_method, which will automatically use the highest supported protocol version.

Thanks,
Kareem

kython28, 2022-05-26:
Hello!

hello, I just recompiled with only the two options that you told me (previously I tried putting all of them) and I still can't connect, I always receive: -313

I also defined what you told me and added the CA certificate. Still not working :'(

Here my options:
```Bash
./configure --enable-all --enable-curve25519 --enable-tls13
```

```C

#define HAVE_CURVE25519
#define WOLFSSL_TLS13
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES

// ........

wolfSSL_library_init();
WOLFSSL_CTX *ctx = wolfSSL_CTX_new(wolfTLS_client_method());
wolfSSL_CTX_load_verify_locations(ctx, NULL, "...");

// I create the fd here and then I connect ....

WOLFSSL *ssl = wolfSSL_new(ctx);
wolfSSL_set_fd(ssl, fd);

int ret = wolfSSL_connect(ssl);
if (ret != WOLFSSL_SUCCESS){
    printf("%d\n", wolfSSL_get_error(ssl, ret));
    // ...
}
```

kareem-wolfssl, 2022-05-26:
Hi @kython28 ,

No need to include the defines since you are using ./configure.  We also do not recommend using `--enable-all`.
Please enable debug logging by adding `--enable-debug` and calling wolfSSL_Debugging_ON() before any other wolfSSL functions in your code, then attach your debug log here.  Please also confirm you're including wolfSSL settings with `#include <wolfssl/options.h>`.

kython28, 2022-05-26:
Ready. I did everything, I also confirm that I am including "wolfssl/options.h", and recompile with:
```
./configure --enable-curve25519 --enable-tls13 --enable-debug
```

here are the logs:
```
2 - wolfSSL Entering wolfCrypt_Init
2 - wolfSSL Entering SSLv23_client_method_ex
2 - wolfSSL Entering wolfSSL_CTX_new_ex
2 - wolfSSL Entering wolfSSL_Init
2 - wolfSSL Entering wolfSSL_CertManagerNew
3 - wolfSSL Leaving WOLFSSL_CTX_new, return 0
1 - wolfSSL_CTX_load_verify_locations_ex
1 - opendir path verify locations failed
2 - wolfSSL Entering SSL_new
3 - wolfSSL Leaving SSL_new, return 0
2 - wolfSSL Entering SSL_set_fd
2 - wolfSSL Entering SSL_set_read_fd
3 - wolfSSL Leaving SSL_set_read_fd, return 1
2 - wolfSSL Entering SSL_set_write_fd
3 - wolfSSL Leaving SSL_set_write_fd, return 1
2 - wolfSSL Entering wolfSSL_connect_TLSv13()
2 - wolfSSL Entering SendTls13ClientHello
1 - Adding signature algorithms extension
1 - Adding supported versions extension
2 - wolfSSL Entering EccMakeKey
3 - wolfSSL Leaving EccMakeKey, return 0
1 - growing output buffer
1 - Key Share extension to write
1 - Supported Versions extension to write
1 - Signature Algorithms extension to write
1 - Point Formats extension to write
1 - Supported Groups extension to write
1 - Encrypt-Then-Mac extension to write
1 - EMS extension to write
1 - Shrinking output buffer
3 - wolfSSL Leaving SendTls13ClientHello, return 0
1 - connect state: CLIENT_HELLO_SENT
1 - Server attempting to accept with different version
1 - received record layer msg
1 - got ALERT!
1 - Alert type: handshake_failure
0 - wolfSSL error occurred, error = 40
0 - wolfSSL error occurred, error = -313
2 - wolfSSL Entering SSL_get_error
3 - wolfSSL Leaving SSL_get_error, return -313
```

kython28, 2022-05-26:
Sorry, I uploaded the CA wrong, I upload the log again.

```
2 - wolfSSL Entering wolfCrypt_Init
2 - wolfSSL Entering SSLv23_client_method_ex
2 - wolfSSL Entering wolfSSL_CTX_new_ex
2 - wolfSSL Entering wolfSSL_Init
2 - wolfSSL Entering wolfSSL_CertManagerNew
3 - wolfSSL Leaving WOLFSSL_CTX_new, return 0
2 - wolfSSL Entering wolfSSL_CTX_set_verify
1 - wolfSSL_CTX_load_verify_locations_ex
1 - Getting dynamic buffer
1 - Processing CA PEM file
2 - wolfSSL Entering PemToDer
1 - Adding a CA
2 - wolfSSL Entering GetExplicitVersion
2 - wolfSSL Entering GetSerialNumber
1 - Got Cert Header
2 - wolfSSL Entering GetAlgoId
2 - wolfSSL Entering GetObjectId()
1 - Got Algo ID
1 - Getting Cert Name
1 - Getting Cert Name
1 - Getting Cert Name
1 - Getting Cert Name
1 - Got Subject Name
2 - wolfSSL Entering GetAlgoId
2 - wolfSSL Entering GetObjectId()
1 - Got Key
1 - Parsed Past Key
2 - wolfSSL Entering DecodeCertExtensions
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering DecodeAuthInfo
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering DecodeSubjKeyId
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering DecodeBasicCaConstraint
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering DecodeAuthKeyId
2 - wolfSSL Entering GetObjectId()
1 - Certificate Policy extension not supported yet.
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering DecodeCrlDist
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering DecodeKeyUsage
2 - wolfSSL Entering GetObjectId()
2 - wolfSSL Entering DecodeAltNames
2 - wolfSSL Entering GetAlgoId
2 - wolfSSL Entering GetObjectId()
1 - 	Parsed new CA
1 - 	Freeing Parsed CA
1 - 	Freeing der CA
1 - 		OK Freeing der CA
3 - wolfSSL Leaving AddCA, return 0
1 -    Processed a CA
2 - wolfSSL Entering PemToDer
1 - Adding a CA
2 - wolfSSL Entering GetExplicitVersion
2 - wolfSSL Entering GetSerialNumber
1 - Got Cert Header
2 - wolfSSL Entering GetAlgoId
2 - wolfSSL Entering GetObjectId()
1 - Got Algo ID
1 - Getting Cert Name
1 - Getting Cert Name
1 - Getting Cert Name
1 - Getting Cert Name
1 - Got Subject Name
[... 4470 lines cut ...]
1 - Adding supported versions extension
2 - wolfSSL Entering EccMakeKey
3 - wolfSSL Leaving EccMakeKey, return 0
1 - growing output buffer
1 - Key Share extension to write
1 - Supported Versions extension to write
1 - Signature Algorithms extension to write
1 - Point Formats extension to write
1 - Supported Groups extension to write
1 - Encrypt-Then-Mac extension to write
1 - EMS extension to write
1 - Shrinking output buffer
3 - wolfSSL Leaving SendTls13ClientHello, return 0
1 - connect state: CLIENT_HELLO_SENT
1 - Server attempting to accept with different version
1 - received record layer msg
1 - got ALERT!
1 - Alert type: handshake_failure
0 - wolfSSL error occurred, error = 40
0 - wolfSSL error occurred, error = -313
2 - wolfSSL Entering SSL_get_error
3 - wolfSSL Leaving SSL_get_error, return -313
```

Still, I keep getting the same error.

kareem-wolfssl, 2022-05-26:
Thanks for the additional information, looks like the server is using SNI.  Please try enabling SNI with `--enable-sni`, and call `wolfSSL_CTX_UseSNI(ctx, WOLFSSL_SNI_HOST_NAME, "ws-api.kucoin.com", strlen("ws-api.kucoin.com"));` before connecting.

kython28, 2022-05-26:
OHHHH! IT WORKS!!!, thank you so much!!!

kython28, 2022-05-26:
Now I have the question, if through the code there is a way to know if the server uses SNI, to call that function before making the connection.

kareem-wolfssl, 2022-05-26:
Set up an SNI callback using `wolfSSL_CTX_set_servername_callback(ctx, yourSniCallback);`.  We will call this callback if we detect the server is using SNI, you can then enable SNI in your callback.
`typedef int (*CallbackSniRecv)(WOLFSSL *ssl, int *ret, void* exArg);`
Expected return codes: `warning_return, fatal_return, noack_return` to indicate issues, any other return code to indicate success.

kython28, 2022-05-27:
Thank so much!, it works! :-)

closed 2022-05-27 by kython28