wolfSSL/wolfssl #5190: Please add "Microsoft x86-64 calling convention" support in sp_x86_64_asm.asm

ilka1999, 2022-05-30:
### Version

5.3.0

### Description

According to MS convention, registers XMM6-15 are nonvolatile. They must be saved and restored by a function that uses them

For example in the "wolfcrypt/src/sp_x86_64_asm.asm"

        sp_2048_get_from_table_avx2_16 PROC
           mov	rax, 1
           movd	xmm10, r8
           movd	xmm11, rax
           vpxor	ymm13, ymm13, ymm13
           vpermd	ymm10, ymm13, ymm10
           vpermd	ymm11, ymm13, ymm11
        
Should be added

        sp_2048_get_from_table_avx2_16 PROC
           push     rbp
           mov      rbp,rsp
           sub      rsp, N * 32  
       ; N is number ymm registers to save or (possible) we can save only xmm parts, in that case we should use 16 instead of 32
       ; for ymm we should align stack to 32 boundary
           and       @@rsp,0FFFFFFFFFFFFFFE0h
        ; save nonvolatile registers
           vmovapd  [rsp+0],ymm6
           vmovapd  [rsp+32],ymm7
           mov	        rax, 1
           movd	xmm10, r8
           movd	xmm11, rax
           vpxor	ymm13, ymm13, ymm13
           vpermd	ymm10, ymm13, ymm10
           vpermd	ymm11, ymm13, ymm11

        ; restore nonvolatile registers and stack
           vmovapd  ymm6,[rsp+0]
           vmovapd  ymm7,[rsp+32]                
           mov      rsp,rbp
           pop      rbp        
           ret

SparkiDev, 2022-05-30:
Hi @ilka1999

I've modified the assembly code to put the xmm6-xmm15 registers on the stack and restore them.
Please try this out and let us know how it goes!

Sean

ilka1999, 2022-05-31:
Hi @SparkiDev 

1. Fixes are incomplete
    XMM is the low part of YMM
    There about 10 function in the code that change the ymm6-15 registers without saving XMM parts
    
2. I hope compilers will not use the entire YMM as non-volatile, if it is true we can save only xmm part.

3. My case. I am using clang 13.0.0, in the part of my code it uses xmm7 for zeroing data, but after wolfssl xmm7 is not zero

SparkiDev, 2022-05-31:
Hi @ilka1999
1. I've fixed the 10 functions.
2. The MSVC documentation is clear that XMM only is non-volatile and it the high parts of the YMM registers is known to be volatile.
3. Your right, I was too worried about other stack usage after the stored XMM registers. Fixed.
4. I've changed the usage of _udiv128 for clang with _WIN64 to use inline assembly. Let me know if this works.

Thanks for working with me on these fixes!

Sean :-)

SparkiDev, 2022-06-05:
Hi @ilka1999,,

Could you let me know if you've had a chance to try out the updated pull request?

Thanks,
Sean :-)

ilka1999, 2022-06-05:
Hi Sean

I was busy on other project.
I going to check your last changed on Monday

Best regards
Ilias

ilka1999, 2022-06-06:
Hi @SparkiDev 

I checked the code, now the registers are saved and restored correctly.

It would be possible to provide x64 stack alignment and use MOVDQA instead of MOVDQU , but I don't think this is critical, because. there is no call to any third-party functions from the code.

Unfortunately, at the moment I can not check the execution of code version 5.3.0 (I use version 5.1.0). In version 5.3.0 there were some problems with encryption, I haven’t figured it out yet. I am currently working on other projects
I think within a week I will be able to try version 5.3.0 I'll tell you about the results.

Best regards
Ilias

SparkiDev, 2022-06-06:
Hi @ilka1999 
Using MOVDQA over MOVDQU is not a big advantage with newer chips. We don't consider it worthwhile at this time.

Thanks for the update about encryption. Please let me know which algorithm you are using. The WIN64 asm code for AES is different from for Linux x64 and so may need to be changed separately.

Sean

PR #5191 (2022-06-06): SP Winx64 ASM: XMM6-15 are non-volatile
# Description

Put the used non-volatile XMM registers on the stack at the start and
get them off again at the end.

Fixes #5190

# Testing
Standard testing.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-06-06 by dgarske