wolfSSL/wolfssl #5208: Question about support for TLS v1.3 handshake with Ed25519/X25519

petertorelli, 2022-06-03:
### Version

5.1.1

### Description

Config: `$ ./configure CFLAGS=-DNO_RSA -pg -g --enable-ecc --enable-ed25519 --enable-eccencrypt --enable-curve25519`

I am trying to understand how TLS might use Ed25519, considering the caveats around ph and ctx versions of the signing algorithm. I see there are premade `certs/ed25519` server and CA certs/keys generated. However, I don't see a way to force the signature algorithm in the server startup line, nor do I see a `BUILD_*` macor for a TLS ciphersuite containing Ed25519. The existence of the certs leads me to believe this can be configured, however I cannot identify the correct way to launch the example server, and subsequently connect with the client.

More to the point, I'm trying to do this to understand the following: does WolfSSL uses PureEdDsa and constructs an identity hash (no hash) sending this to the sign algorithm, or does it open a SHA512 context, use hash updates per TLS handshake step, and then send the SHA512 to the basic `wc_ed25519_sign_msg` command?

Even one more layer above this, I'm surveying SSL libraries to understand what the common practice is (open an update hash, or send the huge message to the sign) given the concern about ph and ctx vulnerabilities (as outlined in your own blog post).

Thanks,
Peter

SparkiDev, 2022-06-03:
Hi @petertorelli

Ed25519 certificates can be passed to the server example as you would any other certificate using the option '-c' and you will also need to pass in the key with '-k'.

wolfSSL performs signatures with Ed25519 in the handshake as per the standards.
For TLS 1.2 and below: RFC 8422.
For TLS 1.3: RFC 8446 (TLS 1.3 RFC).

There are no specific cipher suites for Ed25519. The ECDSA cipher suites can be for any ECC signature schemes.

For TLS 1.3, we cache the messages that are to be hashed.

Does this help?

Sean
--
Sen Parkinson, wolfSSL

petertorelli, 2022-06-03:
@SparkiDev 

Hi Sean, yes, this comment is what interests me, which isn't (AFAICT) specified in RFC8446 (esp. section 4.4.3):

"For TLS 1.3, we cache the messages that are to be hashed."

Since the only function used by `internal.c:Ed25519Sign` (as callback) and `ssl.c:wolfSSL_ED25519_sign` (unused?) is `wc_ed25519_sign_msg`, this confirms that the lib is not constructing the SHA512 hash first, then signing it, but using the PureEdDsa definition as stated in RFC8446:4.4.2, and passing the cached message to the signature function.

Thank you,
Peter

closed 2022-06-03 by petertorelli