wolfSSL/wolfssl #5225: [Bug]: DTLS Multiple Length fields Non-conformances

bathooman, 2022-06-09:
### Version

5.3.0

### Description

**Configure Options**

> ./configure --disable-asm --enable-singlethreaded --enable-dtls --enable-dtls-mtu --enable-psk --disable-extended-master --prefix=$(pwd)/build --enable-static --enable-opensslextra --enable-debug

**Expected behavior**

Fundamentally, a length field in a record structure should specify the number of bytes for the respective object. Moreover, The DTLS 1.2 RFC states that:
> Unlike TLS, DTLS is resilient in the face of invalid records (e.g., invalid formatting, **length**, MAC, etc.).  In general, invalid records SHOULD be silently discarded, thus preserving the association; however, an error MAY be logged for diagnostic purposes.

**Actual behavior**    

According to the RFC, invalid records (records containing invalid fields such as invalid length fields) should be discarded. However, On different occasions, we see that WolfSSL does not discard invalid records and proceed with the handshake.

##### 1- Handshake Length

One can proceed with a handshake in the presence of an invalid handshake length for the `CH0`.

[image]

In the following, you can find the Wireshark log.

[handshake-length-witness.zip](https://github.com/wolfSSL/wolfssl/files/8869121/handshake-length-witness.zip)

##### 2- Fragment Offset
 
It appears that WolfSSL completely ignores fragment offset when it processes incoming records. One could complete a handshake with an invalid offset in all messages if the hash check when `Finished` is processed was disabled.

In the following, you can find the Wireshark log.

[fragment-offset-witness.zip](https://github.com/wolfSSL/wolfssl/files/8869277/fragment-offset-witness.zip)

##### 3- Fragment Length

Communicating with a server, the `CH2` and `CKE` messages with invalid fragment lengths are not properly discarded. One could complete a handshake with an invalid fragment length in the `CH2` and `CKE` messages if the hash check when `Finished` is processed was disabled.

[image]

In the following, you can find the Wireshark log.

[Fragment-length-witness.zip](https://github.com/wolfSSL/wolfssl/files/8869327/Fragment-length-witness.zip)

##### 4- Extensions Length

Last but not least, the handshake could be completed in the presence of an invalid extension length. 

[image]

In the following, you can find the Wireshark log.

[Extensions-length-witness.zip](https://github.com/wolfSSL/wolfssl/files/8869350/Extensions-length-witness.zip)

rizlik, 2022-06-09:
Hi @bathooman ,

Thanks a lot for the detailed report, we are investigating the issues.
Can you provide a quick way to reproduce the issue you mentioned? 
That would really helps. 
If you want to communicate with us in a more private manner, you can reach us directly at support@wolfssl.com.
Thanks

rizlik, 2022-06-16:
Hi @bathooman,
#5246 should have fixed this. Can you confirm that? 
Thanks,

bathooman, 2022-06-21:
Hello,

That fixes the problem. I close the issue if no further action is required.

closed 2022-06-21 by bathooman