wolfSSL/wolfssl #5233: [Bug]: Turning on WOLFSSL_NO_CONST_CMP causes compile failure

gojimmypi, 2022-06-11:
### Contact Details

jim@wolfssl.com

### Version

Latest master branch

### Description

When turning on `WOLFSSL_NO_CONST_CMP`, in my case for the ESP32, with this [user_settings.h](https://github.com/gojimmypi/wolfssh/blob/a1f15195bfb0f3b2edc413a8d337cb674a8e7d10/examples/ESP32-SSH-Server/components/wolfssl/include/user_settings.h#L38) file, I get compile errors for missing `ConstantCompare` that is not wrapped in `#ifndef WOLFSSL_NO_CONST_CMP` such as in [misc.c#L327](https://github.com/wolfSSL/wolfssl/blob/afc63a3bfa7b1ddc21d0ab4125bc9d031e4ce505/wolfcrypt/src/misc.c#L327)

I'm building with this [CMakeLists.txt](https://github.com/gojimmypi/wolfssh/blob/ESP32_Development/examples/ESP32-SSH-Server/CMakeLists.txt) and turning on `WOLFSSL_NO_CONST_CMP` like this:
```
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -DWOLFSSL_USER_SETTINGS -DWOLFSSH_NO_RSA -DNO_RSA -DWOLFSSH_SHELL -DDEBUG_WOLFSSL -DDEBUG_WOLFSSH -DWOLFSSL_ESP32WROOM32_CRYPT_DEBUG -DWOLFSSL_NO_CONST_CMP")
```

Compile fails on:
[fe_operations.c#L1353](https://github.com/wolfSSL/wolfssl/blob/afc63a3bfa7b1ddc21d0ab4125bc9d031e4ce505/wolfcrypt/src/fe_operations.c#L1353)
[ed25519.c#L675](https://github.com/wolfSSL/wolfssl/blob/afc63a3bfa7b1ddc21d0ab4125bc9d031e4ce505/wolfcrypt/src/ed25519.c#L675)

Is this expected, and is this how I should be using `WOLFSSL_NO_CONST_CMP` ?

### Reproduction steps

This was built with VisualGDB like this:
```
cmd.exe /C "cd /D C:\workspace\wolfssh-gojimmypi-ESP32\examples\ESP32-SSH-Server\build\VisualGDB\Debug\bootloader\esp-idf\esptool_py && python C:/SysGCC/esp32/esp-idf/v4.4.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/workspace/wolfssh-gojimmypi-ESP32/examples/ESP32-SSH-Server/build/VisualGDB/Debug/bootloader/bootloader.bin"
```
but is expected to have similar results with the Espressif ESP-IDF.

### Relevant log output

```shell
../../../components/wolfssl/wolfcrypt/src/ed25519.c: In function 'ed25519_verify_msg_final_with_sha':
../../../components/wolfssl/wolfcrypt/src/ed25519.c:675:11: error: implicit declaration of function 'ConstantCompare' [-Werror=implicit-function-declaration]
     ret = ConstantCompare(rcheck, sig, ED25519_SIG_SIZE/2);
           ^~~~~~~~~~~~~~~
../../../components/wolfssl/wolfcrypt/src/fe_operations.c: In function 'fe_isnonzero':
../../../components/wolfssl/wolfcrypt/src/fe_operations.c:1353:10: error: implicit declaration of function 'ConstantCompare' [-Werror=implicit-function-declaration]
   return ConstantCompare(s,zero,32);
          ^~~~~~~~~~~~~~~
```

dgarske, 2022-06-13:
Hi @gojimmypi ,

This is an improper use of `WOLFSSL_NO_CONST_CMP`. ConstantCompare is required for many algorithms. What prompted you to consider using `WOLFSSL_NO_CONST_CMP`?

This was added to provide a reduced build option for RSA/SHA2-256 only (see https://github.com/wolfSSL/wolfssl/pull/1960).

Thanks,
David Garske, wolfSSL

gojimmypi, 2022-06-13:
Hi @dgarske 

I would think that if someone wanted to squeeze every drop of performance and knew the concerns of `ConstantCompare()` timing, that the `WOLFSSL_NO_CONST_CMP` could be defined as desired.

In my case, I stumbled upon this problem in desperation while working on the SHA256 acceleration features when failing the [wc_RNG_HealthTest() at the second Hash_DRBG_Generate()](https://github.com/wolfSSL/wolfssl/blob/0b78961111ea926f8ea918d12dd1222c8ecbfca5/wolfcrypt/src/random.c#L1173) due to a mismatched value.

Clearly the [implementation of ConstantCompare](https://github.com/wolfSSL/wolfssl/blob/0b78961111ea926f8ea918d12dd1222c8ecbfca5/wolfcrypt/src/misc.c#L329) should do what is desired all the time, but I wasn't sure if there were other side effects elsewhere. In any case, the compare method is not the cause of my SHA hardware computation discrepancy. 

Is there a proper way to use `WOLFSSL_NO_CONST_CMP` or is that generally an internal setting?

Cheers

dgarske, 2022-06-13:
Hi @gojimmypi ,

Your understanding of `WOLFSSL_NO_CONST_CMP` is wrong. It is not to change behavior, but to reduce code size when the function ConstantCompare is not used.

The function `ConstantCompare` is a way to compare bytes using a constant time implementation where all bytes are checked even if a mismatch is found early. The typical memcmp exists early if a difference is found. For compare cases against private data the constant version does not leak timing data.

Your issue with the RNG health test was because your SHA2-256 result expected from the DRBG was not valid.

Thanks,
David Garske, wolfSSL

gojimmypi, 2022-06-13:
aha! thanks for the clarification.

closed 2022-06-13 by gojimmypi