wolfSSL/wolfssl #5247: [Bug]: mp_exptmod_ex does not fail if modulus is 0

guidovranken, 2022-06-15:
### Version

Git repository checkout

### Description

It would be more appropriate to fail because a value modulo 0 has no solution.

### Reproduction steps

```
git clone --depth 1 https://github.com/wolfSSL/wolfssl.git
cd wolfssl/
autoreconf -ivf
./configure --enable-static
make
gcc -I . poc.c src/.libs/libwolfssl.a
./a.out
```

```c
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/integer.h>

#define MP_CHECK(x) if ( (x) != MP_OKAY ) abort();

int main(void)
{
    mp_int a;
    mp_int b;
    mp_int c;
    mp_int result;
    char str[8192];

    MP_CHECK(mp_init(&a));
    MP_CHECK(mp_init(&b));
    MP_CHECK(mp_init(&c));
    MP_CHECK(mp_init(&result));
    
    MP_CHECK(mp_read_radix(&a, "0", 10));
    MP_CHECK(mp_read_radix(&b, "1", 10));
    MP_CHECK(mp_read_radix(&c, "0", 10));

    MP_CHECK(mp_exptmod_ex(&a, &b, b.used, &c, &result));
    
    MP_CHECK(mp_toradix(&result, str, 10));

    printf("%s\n", str);

    /* noret */ mp_clear(&a);
    /* noret */ mp_clear(&b);
    /* noret */ mp_clear(&c);
    /* noret */ mp_clear(&result);
    return 0;
}
```c

embhorn, 2022-06-15:
Hi @guidovranken 

Thanks for this report. I was able to reproduce and I have assigned this to be reviewed.

SparkiDev, 2022-06-16:
Hi @guidovranken 

A fix has been put up for this: https://github.com/wolfSSL/wolfssl/pull/5252
I could only see an issue with the implementation in tfm.c.

Thanks,
Sean

guidovranken, 2022-06-16:
Confirmed fixed.

closed 2022-06-16 by guidovranken