wolfSSL/wolfssl #5264: [Bug]: SIGSEGV as (sub)struct in WOLFSSL not aligned to 16 bytes

space88man, 2022-06-19:
### Contact Details

shihping.chan@gmail.com

### Version

5.3.0

### Description

I found a case where gcc(8.5.0)  generates `movaps` asm  in `-O2` and `-O` and the structure `ssl->buffers.serverDH_P` was not aligned to 16 bytes
Flags

```
CFLAGS="-g -O2" ./configure --enable-optflags=no --enable-all --enable-pkcs11 --disable-examples --disable-silent-rules
```

```
internal.c: 6028
6028        ssl->buffers.serverDH_P = ctx->serverDH_P;
(gdb) print &ctx->serverDH_P
$4 = (buffer *) 0x7fe7bd824110
(gdb) print &ssl->buffers.serverDH_P #not aligned to 16 bytes
$3 = (buffer *) 0x7fe7bebd0768

(gdb) disassemble 0x00007fe7b9e5b6ef,+64
Dump of assembler code from 0x7fe7b9e5b6ef to 0x7fe7b9e5b72f:
=> 0x00007fe7b9e5b6ef <SetSSL_CTX+1279>:        movaps %xmm0,0x330(%rbx)
```

I don't know why gcc chose `movaps` instead of `movups` since it didn't ensure the 
struct-in-struct-in-struct (the `buffer` members)  was aligned at 16 bytes.

In other testing gcc would generate `movups`.

space88man, 2022-06-19:
clang 13 generates `movups`:

```
    ssl->buffers.serverDH_P = ctx->serverDH_P;
    37fe:       41 0f 10 46 38          movups 0x38(%r14),%xmm0
```

space88man, 2022-06-20:
Update: I am using  a custom memory allocator but the allocations are guaranteed on 8-byte boundaries.

The location which is affected by `-DWOLFSSL_USE_ALIGN`

```
// internal.h
   1864 
   1865 typedef struct {
   1866     ALIGN16 byte staticBuffer[STATIC_BUFFER_LEN];
   1867     byte*  buffer;       /* place holder for static or dynamic buffer */
   1868     word32 length;       /* total buffer length used */
   1869     word32 idx;          /* idx to part of length already consumed */
   1870     word32 bufferSize;   /* current buffer size */
   1871     byte   dynamicFlag;  /* dynamic memory currently in use */
   1872     byte   offset;       /* alignment offset attempt */
   1873 } bufferStatic;
```

This `ALIGN16` macro causes gcc to emit `movaps`. With clang the macro is not defined so it always emits `movups`.

SparkiDev, 2022-06-20:
Hi @space88man,

Just a thought but I recently found a similar issue where I was prepending length to the allocated data.
To get it to work with GCC on x86_64 I needed to make the prepended amount 16 bytes.
I assumed that GCC is allocating on a 16 byte boundary from this.

Sean

space88man, 2022-06-20:
I am integrating this into kamailio which provides its own `wolfSSL_SetAllocators`; possibly the memory allocations are aligned to 8-bytes instead of 16-bytes.
> Hi @space88man,
> 
> Just a thought but I recently found a similar issue where I was prepending length to the allocated data. To get it to work with GCC on x86_64 I needed to make the prepended amount 16 bytes. I assumed that GCC is allocating on a 16 byte boundary from this.
> 
> Sean

space88man, 2022-06-20:
NOTABUG - my custom memory allocator should follow malloc and return memory aligned on 16 bytes but instead does so at 8 bytes. Thanks @SparkiDev .

closed 2022-06-20 by space88man

space88man, 2022-06-21:
Hello @SparkiDev ­— here is an interesting thread in jemalloc https://github.com/jemalloc/jemalloc/issues/1533; if users of wolfSSL use jemalloc (this was 2019, not sure if it now returns 16-byte aligned pointers) they might be surprised.

Maybe the docs should emphasise more the impact of `--enable-aligndata=yes".