wolfSSL/wolfssl #5270: --enable-aligndata=yes doesn't test for clang

space88man, 2022-06-21:
### Version

master

### Description

When using `--enable-aligndata=yes` some  macros are defined that result in various alignment optimisations.

In `wolfssl/wolfcrypt/types.h`
```
    #ifdef WOLFSSL_USE_ALIGN
        #if !defined(ALIGN16)
            #if defined(__IAR_SYSTEMS_ICC__) || defined(__GNUC__)
                #define ALIGN16 __attribute__ ( (aligned (16)))
            #elif defined(_MSC_VER)
                /* disable align warning, we want alignment ! */
                #pragma warning(disable: 4324)
                #define ALIGN16 __declspec (align (16))
            #else
                #define ALIGN16
            #endif
        #endif /* !ALIGN16 */
```

embhorn, 2022-06-21:
Hello @space88man 

Thanks for this suggestion. Adding `|| defined(__llvm__)` in all cases would solve this from your perspective?

```
    #ifdef WOLFSSL_USE_ALIGN
        #if !defined(ALIGN16)
            #if defined(__IAR_SYSTEMS_ICC__) || defined(__GNUC__) || defined(__llvm__)
                #define ALIGN16 __attribute__ ( (aligned (16)))
            #elif defined(_MSC_VER)
                /* disable align warning, we want alignment ! */
                #pragma warning(disable: 4324)
                #define ALIGN16 __declspec (align (16))
            #else
                #define ALIGN16
            #endif
        #endif /* !ALIGN16 */
```

space88man, 2022-06-22:
> Hello @space88man
> 
> Thanks for this suggestion. Adding `|| defined(__llvm__)` in all cases would solve this from your perspective?
> 

Yes this would be great.

PR #5290 (2022-06-29): Add support for aligned data with clang llvm
# Description

Add support for alignment with clang llvm compiler.

Fixes #5270

# Testing

`./configure CC=clang --enable-aligndata`

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-06-29 by douzzer