wolfSSL/wolfssl #5275: [RFE] Expose runtime features flag :  alignment requirements of structs

space88man, 2022-06-22:
### Version

master / 5.3.0

### Description

[RFE] Expose at runtime a features flag such as alignment requirements for structs

I am attempting to integrate wolfSSL into [kamailio](https://github.com/kamailio/kamailio) as a TLS transport.

I encountered the following impedance mismatch:
* kamailio provides custom memory allocators on 8-byte alignment
* wolfSSL when compiled with the default (`--enable-aligndata=yes`) has 16-byte alignment requirement on `WOLFSSL` This flag is used by Debian packaging.

The result is that some structs are aligned on 8-byte boundaries by the custom allocators. Optimisation flags
like `-O -O2 -Og` cause gcc to emit `movaps`. This caused alignment segfaults.

It would be useful at runtime to be able to detect this mismatch. kamailio also happens to have 16-byte aligned memory allocators (not turned on by default).

space88man, 2022-06-24:
@kareem-wolfssl  - oh I did not intend to change at runtime - more to expose to the consuming application features baked into the library such as (my example) alignment requirement.

kareem-wolfssl, 2022-06-24:
@space88man - Thanks for the update, I understand now.  The calling application should be including our build-time settings, eg. options.h/user_settings.h depending on how wolfSSL was built.  With these settings included, you just need to check if `WOLFSSL_USE_ALIGN` is defined to determine if we need any alignment.  Let me know if this is sufficient.

closed 2022-07-07 by space88man