wolfSSL/wolfssl #5282: [Bug]: -Werror  on current master

space88man, 2022-06-24:
### Contact Details

shihping.chan@gmail.com

### Version

master@00b82888bc

### Description

Basic build errors out due to `-Werror=array-bounds`

### Reproduction steps

```
./configure --enable-all --enable-pkcs11
make
```

### Relevant log output

```shell
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-asn.lo
In file included from ./wolfssl/wolfcrypt/asn.h:37,
                 from wolfcrypt/src/asn.c:87:
In function ‘SetCurve’,
    inlined from ‘wc_BuildEccKeyDer’ at wolfcrypt/src/asn.c:28449:19:
./wolfssl/wolfcrypt/types.h:620:39: error: ‘memcpy’ forming offset [22, 131] is out of the bounds [0, 22] of object ‘curve’ with type ‘byte[22]’ {aka ‘unsigned char[22]’} [-Werror=array-bounds]
  620 |             #define XMEMCPY(d,s,l)    memcpy((d),(s),(l))
      |                                       ^~~~~~~~~~~~~~~~~~~
wolfcrypt/src/asn.c:13175:5: note: in expansion of macro ‘XMEMCPY’
13175 |     XMEMCPY(output+idx, key->dp->oid, oidSz);
      |     ^~~~~~~
wolfcrypt/src/asn.c: In function ‘wc_BuildEccKeyDer’:
wolfcrypt/src/asn.c:28425:12: note: ‘curve’ declared here
28425 |     byte   curve[MAX_ALGO_SZ+2];
      |            ^~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:5783: wolfcrypt/src/src_libwolfssl_la-asn.lo] Error 1
make[2]: Leaving directory '/work/richard/Projects/kamailio/wolfssl-git'
make[1]: *** [Makefile:6759: all-recursive] Error 1
make[1]: Leaving directory '/work/richard/Projects/kamailio/wolfssl-git'
make: *** [Makefile:3989: all] Error 2
```
```

dgarske, 2022-06-24:
@kareem-wolfssl please review this and see if there is an issue here. I recall seeing this a while back and confirmed the array was properly sized and seemed to be occurring due to the compiler inlining a function.

kareem-wolfssl, 2022-06-24:
@space88man Thanks for the report, but I am unable to reproduce the error here, what compiler and environment are you building on?  I am testing with Fedora Linux + GCC 11.
I do see how this could happen, but I am not clear how the compiler is determining oidSz is 132 in this case.

space88man, 2022-06-24:
@kareem-wolfssl I just tried 8dfcc76f5039 with
```
gcc (GCC) 12.1.1 20220507 (Red Hat 12.1.1-1)
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
on Fedora 36
```
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
In file included from ./wolfssl/wolfcrypt/asn.h:37,
                 from wolfcrypt/src/asn.c:87:
In function ‘SetCurve’,
    inlined from ‘wc_BuildEccKeyDer’ at wolfcrypt/src/asn.c:28449:19:
./wolfssl/wolfcrypt/types.h:620:39: error: ‘memcpy’ forming offset [22, 131] is out of the bounds [0, 22] of object ‘curve’ with type ‘byte[22]’ {aka ‘unsigned char[22]’} [-Werror=array-bounds]
  620 |             #define XMEMCPY(d,s,l)    memcpy((d),(s),(l))
      |                                       ^~~~~~~~~~~~~~~~~~~
wolfcrypt/src/asn.c:13175:5: note: in expansion of macro ‘XMEMCPY’
13175 |     XMEMCPY(output+idx, key->dp->oid, oidSz);
      |     ^~~~~~~
wolfcrypt/src/asn.c: In function ‘wc_BuildEccKeyDer’:
wolfcrypt/src/asn.c:28425:12: note: ‘curve’ declared here
28425 |     byte   curve[MAX_ALGO_SZ+2];
      |            ^~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:5783: wolfcrypt/src/src_libwolfssl_la-asn.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:6759: all-recursive] Error 1
make: *** [Makefile:3989: all] Error 2
```

kareem-wolfssl, 2022-06-25:
@space88man I am able to reproduce with GCC 12.  I'm still not clear where GCC is coming up with a size of 132, I will continue to look into this.

PR #5309 (2022-07-02): Pass in and check output length in SetCurve.
# Description

Fixes #5282 and ZD# 14390

# Testing

Built and ran testsuite.  Confirmed warning/error no longer occurs.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-07-02 by dgarske