wolfSSL/wolfssl #5302: wolfSSL_BN_bn2dec: XMALLOC/XFREE type mismatch

space88man, 2022-06-30:
### Version

master

### Description

Why is the type different in `XMALLOC()` and `XFREE()` in this function?

`XMALLOC` → `DYNAMIC_TYPE_OPENSSL`
`XFREE` → `DYNAMIC_TYPE_ECC`

```
// src/ssl.c:35491
char *wolfSSL_BN_bn2dec(const WOLFSSL_BIGNUM *bn)
{
    int len = 0;
    char *buf;

    WOLFSSL_MSG("wolfSSL_BN_bn2dec");

    if (bn == NULL || bn->internal == NULL) {
        WOLFSSL_MSG("bn NULL error");
        return NULL;
    }

    if (mp_radix_size((mp_int*)bn->internal, MP_RADIX_DEC, &len) != MP_OKAY) {
        WOLFSSL_MSG("mp_radix_size failure");
        return NULL;
    }

    buf = (char*) XMALLOC(len, NULL, DYNAMIC_TYPE_OPENSSL);
    if (buf == NULL) {
        WOLFSSL_MSG("BN_bn2dec malloc buffer failure");
        return NULL;
    }

    if (mp_todecimal((mp_int*)bn->internal, buf) != MP_OKAY) {
        XFREE(buf, NULL, DYNAMIC_TYPE_ECC);
        return NULL;
    }

    return buf;
}
```

SparkiDev, 2022-06-30:
Hi @space88man 

I've made some changes in a PR : #5303
Tried to identify some more mismatches as well.

Let us know if this fixes your issue!

Sean

--
Sean Parkinson, wolfSSL
Senior Software Engineer

space88man, 2022-06-30:
LGTM - thanks.

closed 2022-06-30 by dgarske