wolfSSL/wolfssl #5320: Failure to run a library built with `--enable-aesni` on older systems.

cotequeiroz, 2022-07-05:
### Version

5.3.0

### Description

A libarary built with `./configure --enable-aesni`  should run on systems that do not have the extension.  It is currently failing because code that uses sse4 is being generated due to the addition of `-maes -msse4 -mpclmul` to compile all source files.

I've tested this with Linux, running qemu.  The error was discussed in https://github.com/openwrt/openwrt/pull/10103.

It appears that the only source file that needs `-maes` with Linux is `wolfcrypt/src/aes.c`, while the other flags are not needed.  Looking at the code, I could easily identify intrinsics requiring `-mpclmul` and `-maes`; it possibly needs `-msse4`, but I haven't tested this, or looked at every single intrinsic function being called.

However, because `-msse4` is added to `CFLAGS`, at gcc will generate code for it elsewhere, and then older systems will fail with illegal instructions.

There are a number of ways to deal with it:
- patching `configure.ac` to using only `-maes` to compile the whole thing, except for Windows where flags will be kept as is.
- patching `configure.ac` and `src/include.am` so that only `aes.c` is built with the flags ( Windows will still use `-maes -msse4 -mcplmul`, gcc/clang will get `-maes`).
- removing the flags completely from `configure.ac`, and adding them to the source files with `#pragma`, as it is already done with RDRAND.

I have tested the first two options on Linux, but I am a noob with Windows, and have no SDK for it.  It's not clear to me even how to single it out with `configure.ac`.  Is `case $host_os in mingw*)` enough?

kareem-wolfssl, 2022-07-06:
Hey @cotequeiroz ,

Thanks for reporting this, indeed AES-NI does not depend on SSE4 so we should be able to decouple these.  I'm working with the team to determine the best way to do this.
In the meantime, you can use this as a workaround: `CFLAGS='-mno-sse4' ./configure --enable-aesni ...`.  Of course this will lead to a loss in performance over SSE4 support, but in return it will run on older CPUs without SSE4.

cotequeiroz, 2022-07-06:
Here's my first idea of resolving this:  Use the added flags for compiling `aes.c` only.    Flags are set by autoconf/libtool are used to compile all sources in a library, and can't be set on a source file basis.  The recommended way to change the flags for a file only is adding `aes.c` to its own static library, then using the library in the final libwolfssl linkage.

Windows will use `-mpclmul` in addition to `-maes`:
``` diff
diff --git a/configure.ac b/configure.ac
index 12388db48..1e9d7cffa 100644
--- a/configure.ac
+++ b/configure.ac
@@ -2131,21 +2131,20 @@ then
     if test "$ENABLED_AESNI" = "yes" || test "$ENABLED_INTELASM" = "yes"
     then
         AM_CFLAGS="$AM_CFLAGS -DWOLFSSL_AESNI"
-        if test "$GCC" = "yes"
+        # GCC & clang need this flag, icc doesn't
+        # opt levels greater than 2 may cause problems on systems w/o
+        # aesni
+        if test "$GCC" = "yes" && test "$CC" != "icc"
         then
-            # clang needs these flags
-            if test "$CC" = "clang"
-            then
-                AM_CFLAGS="$AM_CFLAGS -maes -mpclmul"
-            else
-                # GCC needs these flags, icc doesn't
-                # opt levels greater than 2 may cause problems on systems w/o
-                # aesni
-                if test "$CC" != "icc"
-                then
-                    AM_CFLAGS="$AM_CFLAGS -maes -msse4 -mpclmul"
-                fi
-            fi
+            # Windows uses pclmul itrinsics in wolfcrypt/src/aes.c
+            case $host_os in
+            mingw*)
+                CFLAGS_AESNI="-maes -mpclmul"
+                ;;
+            *)
+                CFLAGS_AESNI="-maes"
+                break;;
+            esac
         fi
         AS_IF([test "x$ENABLED_AESGCM" != "xno"],[AM_CCASFLAGS="$AM_CCASFLAGS -DHAVE_AESGCM"])
     fi
@@ -2156,6 +2155,7 @@ then
         ENABLED_AESNI=yes
     fi
 fi
+AC_SUBST([CFLAGS_AESNI])
 
 AC_ARG_ENABLE([aligndata],
     [AS_HELP_STRING([--enable-aligndata],[align data for ciphers (default: enabled)])],
diff --git a/src/include.am b/src/include.am
index 361bf6a89..5a6bd0599 100644
--- a/src/include.am
+++ b/src/include.am
@@ -34,6 +34,11 @@ src_libwolfssl_la_LIBADD = $(LIBM) $(LIB_ADD) $(LIB_STATIC_ADD)
 src_libwolfssl_la_CFLAGS = -DBUILDING_WOLFSSL $(AM_CFLAGS) -DLIBWOLFSSL_GLOBAL_EXTRA_CFLAGS="\" $(EXTRA_CFLAGS)\""
 src_libwolfssl_la_CPPFLAGS = -DBUILDING_WOLFSSL $(AM_CPPFLAGS)
 
+EXTRA_LTLIBRARIES = wolfcrypt/src/libaes.la
+wolfcrypt_src_libaes_la_SOURCES = wolfcrypt/src/aes.c
+wolfcrypt_src_libaes_la_CFLAGS = $(CFLAGS_AESNI) $(src_libwolfssl_la_CFLAGS)
+wolfcrypt_src_libaes_la_CPPFLAGS = $(src_libwolfssl_la_CPPFLAGS)
+
 # install the packaged IPP libraries
 if BUILD_FAST_RSA
 
@@ -184,7 +189,11 @@ src_libwolfssl_la_SOURCES += wolfcrypt/src/ecc.c
 endif
 
 if BUILD_AES
+if BUILD_AESNI
+src_libwolfssl_la_LIBADD += wolfcrypt/src/libaes.la
+else
 src_libwolfssl_la_SOURCES += wolfcrypt/src/aes.c
+endif
 if BUILD_ARMASM
 src_libwolfssl_la_SOURCES += wolfcrypt/src/port/arm/armv8-aes.c
 endif
@@ -374,7 +383,11 @@ endif
 
 if !BUILD_FIPS_CURRENT
 if BUILD_AES
+if BUILD_AESNI
+src_libwolfssl_la_LIBADD += wolfcrypt/src/libaes.la
+else
 src_libwolfssl_la_SOURCES += wolfcrypt/src/aes.c
+endif
 if BUILD_ARMASM
 src_libwolfssl_la_SOURCES += wolfcrypt/src/port/arm/armv8-aes.c
 endif
diff --git a/wolfcrypt/src/aes.c b/wolfcrypt/src/aes.c
index 38b786b67..6eb308263 100644
--- a/wolfcrypt/src/aes.c
+++ b/wolfcrypt/src/aes.c
@@ -4875,7 +4875,7 @@ void AES_GCM_decrypt_avx2(const unsigned char *in, unsigned char *out,
 #ifdef __clang__
     /* With Clang the __m128i in emmintrin.h is union using:
      *     "unsigned __int64 m128i_u64[2];"
-     * Notes: Must add "-maes -msse4.1 -mpclmul" to compiler flags.
+     * Notes: Must add "-maes -mpclmul" to compiler flags.
      *        Must mark "aes_asm.asm" as included/compiled C file.
      */
     #define M128_INIT(x,y) { (long long)x, (long long)y }
```

Perhaps using `#pragma` to add the necessary flags may be simpler.

SparkiDev, 2022-07-06:
Hi @cotequeiroz

I've put up a pull request that should fix the issue.
Only Windows builds need the SSE4 instructions.
See PR #5330

Sean

PR #5330 (2022-07-07): AESNI: fix configure to use minimal compiler flags
# Description

Only AESNI builds on Windows use the GCM intrinsics implementation which uses AES, SSE4 and PCLMUL instructions.
The only place that the intrinsics are used otherwise is in AES_set_decrypt_key() which only needs aes instructions.
Changed configure.ac to deal with this.

Fixes #5320

# Testing

Built on Linux x64 with --enable-aesni.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-07-07 by dgarske