wolfSSL/wolfssl #5331: [RFE] Use a PKCS#11 key in TLS

space88man, 2022-07-07:
### Version

master

### Description

RFE: treat a PKCS#11 private key as something that can be used with `wolfSSL_CTX_use_PrivateKey_buffer()` / `wolfSSL_use_PrivateKey_buffer()` like in OpenSSL(<3.0.0) using engine keys?

So a  hypothetical `SSL_FILETYPE_PKCS11` would be nice to use HSMs in DTLS/TLS connections.

dgarske, 2022-07-07:
Hi @space88man ,

That is a great suggestion to consider.

Our solution for PKCS11 was the addition of the following API's:

```
WOLFSSL_API int wolfSSL_CTX_use_PrivateKey_id(WOLFSSL_CTX* ctx,
                                                const unsigned char* id, long sz,
                                                int devId, long keySz);
WOLFSSL_API int wolfSSL_CTX_use_PrivateKey_Id(WOLFSSL_CTX* ctx,
                                                const unsigned char* id, long sz,
                                                int devId);
WOLFSSL_API int wolfSSL_CTX_use_PrivateKey_Label(WOLFSSL_CTX* ctx, const char* label,
                                                    int devId);
```

You can find examples for using these here:
https://github.com/wolfSSL/wolfssl-examples/blob/master/pkcs11/server-tls-pkcs11.c
https://github.com/wolfSSL/wolfssl-examples/blob/master/pkcs11/server-tls-pkcs11-ecc.c

Let me know if that does not fulfill your requirement.

Thanks,
David Garske, wolfSSL

space88man, 2022-08-15:
@dgarske sorry for the late reply; I could indeed use these APIs.

I'll leave this issue open now if the RFE label is of use to the team but feel free to close it.

dgarske, 2022-08-15:
Hi @space88man ,

Note: If you have PKCS11 enabled the API's you mentioned `wolfSSL_CTX_use_PrivateKey_buffer` and `wolfSSL_use_PrivateKey_buffer ` allow passing a public key only.

This was added in PR: https://github.com/wolfSSL/wolfssl/pull/4921

Thanks,
David Garske, wolfSSL

closed 2022-08-15 by dgarske