wolfSSL/wolfssl #5333: Poor AES-CTR, AES-CCM performance with AESNI

cotequeiroz, 2022-07-07:
### Version

5.3.0

### Description

I was looking at the benchmark results, and noticed that the performance of AES-CTR and AES-CCM have poor performance when compared to the rest of the AES suites.  This happens independently of 9ba77300f9f5dea9f53aed00bf6c33d10b7b2fce.

```
AES-128-CBC-enc   1925 MB took 1.002 seconds, 1920.931 MB/s Cycles per byte =   1.29
AES-128-CBC-dec  12860 MB took 1.000 seconds, 12856.901 MB/s Cycles per byte =   0.19
AES-192-CBC-enc   1630 MB took 1.001 seconds, 1629.055 MB/s Cycles per byte =   1.52
AES-192-CBC-dec  10745 MB took 1.000 seconds, 10740.263 MB/s Cycles per byte =   0.23
AES-256-CBC-enc   1410 MB took 1.000 seconds, 1409.593 MB/s Cycles per byte =   1.75
AES-256-CBC-dec   9245 MB took 1.000 seconds, 9241.173 MB/s Cycles per byte =   0.27
AES-128-GCM-enc   7840 MB took 1.000 seconds, 7839.774 MB/s Cycles per byte =   0.32
AES-128-GCM-dec   7870 MB took 1.000 seconds, 7868.899 MB/s Cycles per byte =   0.31
AES-192-GCM-enc   7135 MB took 1.001 seconds, 7131.292 MB/s Cycles per byte =   0.35
AES-192-GCM-dec   7090 MB took 1.000 seconds, 7087.391 MB/s Cycles per byte =   0.35
AES-256-GCM-enc   6575 MB took 1.000 seconds, 6574.848 MB/s Cycles per byte =   0.38
AES-256-GCM-dec   6615 MB took 1.001 seconds, 6610.683 MB/s Cycles per byte =   0.37
GMAC Table 4-bit  2009 MB took 1.000 seconds, 2008.833 MB/s Cycles per byte =   1.23
AES-128-CTR        805 MB took 1.000 seconds,  804.656 MB/s Cycles per byte =   3.07
AES-192-CTR        750 MB took 1.005 seconds,  746.335 MB/s Cycles per byte =   3.31
AES-256-CTR        695 MB took 1.006 seconds,  691.186 MB/s Cycles per byte =   3.58
AES-CCM-Enc        740 MB took 1.002 seconds,  738.429 MB/s Cycles per byte =   3.35
AES-CCM-Dec        740 MB took 1.002 seconds,  738.871 MB/s Cycles per byte =   3.35
```

It struck me that GCM performs much faster than CTR.  It does not appear right.  For comparison, I've built a table with the relative performance of AES-128-CTR/AES-128-GCM-enc using different builds, just to ensure that practice is not contradicting the theory:

| SW/HW combination | CTR/GCM |
|---------------|----|
| WolfSSL on x86-64 with --enable-intelasm | 0.10 |
| WolfSSL on x86-64 with --enable-aesni | 0.13 |
|WolfSSL on x86-64 without aesni/asm | 2.45 |
| WolfSSL running on armv8 with --enable-armasm | 2.04 |
| OpenSSL on x86-64-aesni | 1.58 |

cotequeiroz, 2022-07-07:
GCM can be roughly defined as counter-mode + ghash.  So how can the addition of ghash make it 10 times faster than plain counter-mode?

It is not a simple matter of using AESNI or not.  While enabling AESNI makes CTR faster than without it, it is not up to par with GCM.  This appears wrong to me.  In order to verify my assumption, I made some comparisons of GCM and CTR speeds using different software and hardware implementations.  In all of them, except for wolfssl with AESNI enabled, CTR is considerably faster than GCM.  When AESNI is enabled, how can CTR become 10x slower than GCM if they are both using the same base cipher, and the same counter-mode of operation?  The arm-CE implementation, which is bascially arm's implementation of AESNI does not suffer from this, and openssl does not as well.  This led me to believe there's something wrong here.

This was  a drive-by observation, as I am not directly using wolfssl on x86 hardware, but I am helping to maintain the wolfssl package in OpenWRT.

SparkiDev, 2022-07-08:
Hi @cotequeiroz,

The issue is one of scaling.
If you build with --enable-intelasm and run benchmark to perform encrypts/decrypts on 16 bytes at a time, that is
  ./wolfcrpyt/benchmark/benchmark -aes-gcm -aes-ctr 16
you will get figures like this:

wolfCrypt Benchmark (block bytes 16, min 1.0 sec each)
AES-128-GCM-enc    350 MB took 1.008 seconds,  347.126 MB/s Cycles per byte =  10.15
AES-128-GCM-dec    330 MB took 1.001 seconds,  329.811 MB/s Cycles per byte =  10.69
AES-192-GCM-enc    345 MB took 1.008 seconds,  342.143 MB/s Cycles per byte =  10.30
AES-192-GCM-dec    330 MB took 1.013 seconds,  325.818 MB/s Cycles per byte =  10.82
AES-256-GCM-enc    335 MB took 1.003 seconds,  333.926 MB/s Cycles per byte =  10.56
AES-256-GCM-dec    325 MB took 1.007 seconds,  322.756 MB/s Cycles per byte =  10.92
GMAC Table 4-bit   296 MB took 1.000 seconds,  295.577 MB/s Cycles per byte =  11.93
AES-128-CTR        675 MB took 1.002 seconds,  673.431 MB/s Cycles per byte =   5.23
AES-192-CTR        620 MB took 1.008 seconds,  615.016 MB/s Cycles per byte =   5.73
AES-256-CTR        570 MB took 1.005 seconds,  566.960 MB/s Cycles per byte =   6.22
Benchmark complete

Where CTR/GCM = 1.94...

When you increase to 128 bytes:

wolfCrypt Benchmark (block bytes 128, min 1.0 sec each)
AES-128-GCM-enc   1485 MB took 1.003 seconds, 1481.292 MB/s Cycles per byte =   2.38
AES-128-GCM-dec   1785 MB took 1.002 seconds, 1780.837 MB/s Cycles per byte =   1.98
AES-192-GCM-enc   1385 MB took 1.001 seconds, 1383.058 MB/s Cycles per byte =   2.55
AES-192-GCM-dec   1640 MB took 1.001 seconds, 1638.190 MB/s Cycles per byte =   2.15
AES-256-GCM-enc   1295 MB took 1.001 seconds, 1293.109 MB/s Cycles per byte =   2.73
AES-256-GCM-dec   1505 MB took 1.003 seconds, 1501.131 MB/s Cycles per byte =   2.35
GMAC Table 4-bit  1029 MB took 1.000 seconds, 1029.431 MB/s Cycles per byte =   3.42
AES-128-CTR        675 MB took 1.002 seconds,  673.462 MB/s Cycles per byte =   5.23
AES-192-CTR        615 MB took 1.000 seconds,  614.822 MB/s Cycles per byte =   5.73
AES-256-CTR        570 MB took 1.008 seconds,  565.582 MB/s Cycles per byte =   6.23
Benchmark complete

CTR/GCM=0.454...

The issue is that the C code is performing the encrypt one block at a time while the GCM assembly code is doing it efficiently on the whole input.

By making a change to the AES-CTR implementation to do create a buffer with all counters and encrypt that in one go:

wolfCrypt Benchmark (block bytes 128, min 1.0 sec each)
AES-128-GCM-enc   1485 MB took 1.003 seconds, 1480.461 MB/s Cycles per byte =   2.38
AES-128-GCM-dec   1765 MB took 1.001 seconds, 1763.089 MB/s Cycles per byte =   2.00
AES-192-GCM-enc   1380 MB took 1.003 seconds, 1375.332 MB/s Cycles per byte =   2.56
AES-192-GCM-dec   1635 MB took 1.001 seconds, 1632.910 MB/s Cycles per byte =   2.16
AES-256-GCM-enc   1300 MB took 1.003 seconds, 1295.881 MB/s Cycles per byte =   2.72
AES-256-GCM-dec   1485 MB took 1.001 seconds, 1484.059 MB/s Cycles per byte =   2.38
GMAC Table 4-bit  1025 MB took 1.000 seconds, 1024.942 MB/s Cycles per byte =   3.44
AES-128-CTR       1255 MB took 1.002 seconds, 1252.773 MB/s Cycles per byte =   2.81
AES-192-CTR       1175 MB took 1.002 seconds, 1172.605 MB/s Cycles per byte =   3.01
AES-256-CTR       1110 MB took 1.001 seconds, 1108.898 MB/s Cycles per byte =   3.18
Benchmark complete

CTR/GCM = 0.846...

Accessing memory multiple times slows the code down.
Doing the whole operation in assembly would fix this.

Let us know if an assembly version is important to you.
The pull request with the improved AES-CTR implementation: #5336

Sean

PR #5336 (2022-07-08): AES-CTR: improve performance when multiple blocks
# Description

When in and out aren't the same pointer, for multiples of block size
input:
  - generate the counters into the output buffer
  - encrypt output buffer
  - XOR in the input
Faster than encrypting a block at a time.

Fixes #5333

# Testing

Standard testing including with and without HAVE_AES_ECB.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-07-08 by SparkiDev