wolfSSL/wolfssl #5344: SP math with custom curves

guidovranken, 2022-07-09:
### Version

Git repository checkout

### Description

Historically custom curves (`--enable-ecccustcurves` configure option) was not possible with SP math.

At commit 2111d6b179a08251ef0560d296d1adc68b482f08, SP math was allowed to use custom curves so I reconfigured the fuzzer at OSS-Fuzz: https://github.com/google/oss-fuzz/pull/7956/files

But currently custom curves with SP math is disabled again. Will it be re-enabled or will it stay this way?

SparkiDev, 2022-07-10:
Hi Guido,

ecccustcurve enables curves that are not in SP.
SP math is a minimal set of MP APIs that support the SP code.
So, SP math isn't meant to support ecccustcurve.
In the transition to having SP math all as the default, configuring for SP math was broken and you got the heap or integer maths instead. So that is probably why the option worked.

Sean

closed 2022-07-10 by SparkiDev

guidovranken, 2022-07-12:
@SparkiDev 

I'm now getting an error when I try to build SP math without assembly.

```
export CFLAGS="$CFLAGS -DHAVE_AES_ECB -DWOLFSSL_DES_ECB -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DWOLFSSL_ECDSA_SET_K -DWOLFSSL_ECDSA_SET_K_ONE_LOOP -DWOLFSSL_PUBLIC_ECC_ADD_DBL"
./configure --enable-static --enable-md2 --enable-md4 --enable-ripemd --enable-blake2 --enable-blake2s --enable-pwdbased --enable-scrypt --enable-hkdf --enable-cmac --enable-arc4 --enable-camellia --enable-aesccm --enable-aesctr --enable-xts --enable-des3 --enable-x963kdf --enable-harden --enable-aescfb --enable-aesofb --enable-aeskeywrap --enable-aessiv --enable-keygen --enable-curve25519 --enable-curve448 --enable-shake256 --disable-crypttests --disable-examples --enable-compkey --enable-ed448 --enable-ed25519 --enable-xchacha --enable-cryptocb --enable-eccencrypt --enable-smallstack --enable-ed25519-stream --enable-ed448-stream --enable-sp --enable-sp-math --disable-asm
```

Output:

```
configure: error: Assembly code turned off
```

Is it correct that SP math no longer supports `--disable-asm`?

Additionally, SP math fails to build on 32 bit:

```
+ make -j2
make -j3  all-recursive
make[1]: warning: -j3 forced in submake: resetting jobserver mode.
make[1]: Entering directory '/src/wolfssl-sp-math'
make[2]: Entering directory '/src/wolfssl-sp-math'
make[2]: warning: -j3 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-cryptocb.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
In file included from wolfcrypt/src/hmac.c:47:
In file included from ./wolfssl/wolfcrypt/cryptocb.h:39:
In file included from ./wolfssl/wolfcrypt/rsa.h:74:
In file included from ./wolfssl/wolfcrypt/integer.h:37:
./wolfssl/wolfcrypt/sp_int.h:272:13: error: unknown type name 'sp_uint128'
    typedef sp_uint128  sp_int_word;
            ^
./wolfssl/wolfcrypt/sp_int.h:273:14: error: unknown type name 'sp_int128'
    typedef  sp_int128  sp_int_sword;
             ^
2 errors generated.
make[2]: *** [Makefile:5409: wolfcrypt/src/src_libwolfssl_la-hmac.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from wolfcrypt/src/cryptocb.c:33:
In file included from ./wolfssl/wolfcrypt/cryptocb.h:39:
In file included from ./wolfssl/wolfcrypt/rsa.h:74:
In file included from ./wolfssl/wolfcrypt/integer.h:37:
./wolfssl/wolfcrypt/sp_int.h:272:13: error: unknown type name 'sp_uint128'
    typedef sp_uint128  sp_int_word;
            ^
./wolfssl/wolfcrypt/sp_int.h:273:14: error: unknown type name 'sp_int128'
    typedef  sp_int128  sp_int_sword;
             ^
2 errors generated.
make[2]: *** [Makefile:6130: wolfcrypt/src/src_libwolfssl_la-cryptocb.lo] Error 1
In file included from wolfcrypt/src/hash.c:31:
In file included from ./wolfssl/wolfcrypt/asn.h:46:
In file included from ./wolfssl/wolfcrypt/integer.h:37:
./wolfssl/wolfcrypt/sp_int.h:272:13: error: unknown type name 'sp_uint128'
    typedef sp_uint128  sp_int_word;
            ^
./wolfssl/wolfcrypt/sp_int.h:273:14: error: unknown type name 'sp_int128'
    typedef  sp_int128  sp_int_sword;
             ^
2 errors generated.
make[2]: *** [Makefile:5570: wolfcrypt/src/src_libwolfssl_la-hash.lo] Error 1
make[2]: Leaving directory '/src/wolfssl-sp-math'
make[1]: *** [Makefile:6756: all-recursive] Error 1
make[1]: Leaving directory '/src/wolfssl-sp-math'
```

SparkiDev, 2022-07-12:
I'll take a look at it.
The SP math all default changes have messed up SP math.

SparkiDev, 2022-07-12:
Hi @guidovranken

I believe the --disable-asm is correct.
The default for x86_64 and aatch64/amd64 is to build the assembly code version.

You will need to change your configurations to include --disable-sp-asm on these platforms if you want the C code version.

I'll look into the 32-bit issue, but I don't have the configuration line you used.

Sean

SparkiDev, 2022-07-12:
Hi @guidovranken 

https://github.com/wolfSSL/wolfssl/pull/5350
Should fix the issue with the 32-bit builds.

Sean

guidovranken, 2022-07-13:
> Hi @guidovranken
> 
> #5350 Should fix the issue with the 32-bit builds.
> 
> Sean

Fixed except that it is now necessary to run `setarch i386 ./configure ...` when compiling for 32 bit on a 64 bit machine, otherwise it will show errors like:

```
wolfcrypt/src/sp_x86_64_asm.S:10353:16: error: register %xmm13 is only available in 64-bit mode
        movdqu %xmm13, %xmm12
               ^~~~~~
wolfcrypt/src/sp_x86_64_asm.S:10354:17: error: register %xmm10 is only available in 64-bit mode
        pcmpeqd %xmm10, %xmm12
                ^~~~~~
wolfcrypt/src/sp_x86_64_asm.S:10355:17: error: register %rcx is only available in 64-bit mode
        movdqu (%rcx), %xmm0
                ^~~~
wolfcrypt/src/sp_x86_64_asm.S:10356:19: error: register %rcx is only available in 64-bit mode
        movdqu 16(%rcx), %xmm1
                  ^~~~
wolfcrypt/src/sp_x86_64_asm.S:10357:19: error: register %rcx is only available in 64-bit mode
        movdqu 32(%rcx), %xmm2
                  ^~~~
wolfcrypt/src/sp_x86_64_asm.S:10358:19: error: register %rcx is only available in 64-bit mode
        movdqu 48(%rcx), %xmm3
                  ^~~~
wolfcrypt/src/sp_x86_64_asm.S:10359:14: error: register %xmm12 is only available in 64-bit mode
        pand %xmm12, %xmm0
             ^~~~~~
wolfcrypt/src/sp_x86_64_asm.S:10360:14: error: register %xmm12 is only available in 64-bit mode
        pand %xmm12, %xmm1
             ^~~~~~
```

I tested with wolfSSL 5.3.0 (before the bignum overhaul) and there it apparently picks up that the target is i386 from the CFLAGS (`-m32`); no need to prefix the `configure` command with `setarch i386`.

SparkiDev, 2022-07-13:
This code is the 64-bit assembly code.
The configuration system doesn't know that you are compiling for 32-bit unless you give it a hint.
I would disable the sp_x86_64.c file if SP_WORD_SIZE isn't 64 but then the rest of the code is expecting an SP implementation that isn't being compiled.

Explicitly specify --disable-sp-asm and --enable-sp-asm.

Sean

guidovranken, 2022-07-25:
Hi Sean this is fixed as far as I'm concerned. Can you please merge https://github.com/wolfSSL/wolfssl/pull/5350 ?

dgarske, 2022-07-25:
> Hi Sean this is fixed as far as I'm concerned. Can you please merge #5350 ?

Merged

guidovranken, 2022-07-25:
Thanks @dgarske