wolfSSL/wolfssl #5349: How to accurately determine supported DH sizes/groups

tobiasbrunner, 2022-07-12:
### Version

5.4.0

### Description

Options, as used for our automated tests:

`./configure C_EXTRA_FLAGS="-DWOLFSSL_PUBLIC_MP -DWOLFSSL_DES_ECB -DHAVE_AES_ECB -DHAVE_ECC_BRAINPOOL -DWOLFSSL_MIN_AUTH_TAG_SZ=8" --disable-crypttests --disable-examples --enable-aesccm --enable-aesctr --enable-camellia --enable-curve25519 --enable-curve448 --enable-des3 --enable-ecccustcurves --enable-ed25519 --enable-ed448 --enable-keygen --enable-md4 --enable-rsapss --enable-sha3 --enable-shake256`

----

To detect which DH groups are supported, the _wolfssl_ plugin in strongSwan uses code like this (see [here](https://github.com/strongswan/strongswan/blob/110e8e6608b48dbefb9c5afd1f05818adabf9ace/src/libstrongswan/plugins/wolfssl/wolfssl_plugin.c#L273-L299) for the full code):

```c
	#if !defined(USE_FAST_MATH) || FP_MAX_BITS >= (3072 * 2)
			PLUGIN_PROVIDE(KE, MODP_3072_BIT),
	#endif
	#if !defined(USE_FAST_MATH) || FP_MAX_BITS >= (4096 * 2)
			PLUGIN_PROVIDE(KE, MODP_4096_BIT),
	#endif
	...
```
With 5.4.0 this doesn't seem to work reliably anymore as there is apparently some other kind of limitation because both MODP_6144 and MODP_8192 don't work (`wc_DhGenerateKeyPair()` returns `MP_EXPTMOD_E`).

Is there something else we can use to detect which groups are supported? Adding `--enable-heapmath` apparently makes the larger groups available again, but can we detect this limitation somehow if the library wasn't built with that option?

----

Another issue is that both MODP_1536 and MODP_768 cause the following error at runtime (whether `--enable-heapmath` is used or not doesn't matter):

    symbol lookup error: /usr/local/lib/libwolfssl.so.34: undefined symbol: pow

This doesn't happen for any of the other groups. Is this as intended? If so, can we detect that these groups are not supported for some reason? (I'm fine with removing support for them, but it's weird that it only happens for these two.)

embhorn, 2022-07-12:
Hi @tobiasbrunner 

Thanks for reaching out. We are reviewing and will post an update.

douzzer, 2022-07-12:
@tobiasbrunner I'm looking at the `pow()` lookup failure.  After a build on 5.4.0 with your exact config, I see this:
```
$ ldd src/.libs/libwolfssl.so
        linux-vdso.so.1 (0x00007fffae7eb000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f8acbd28000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f8acbb2f000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f8acbf0b000)
```
Is `libm.so` missing when you run that `ldd` command?

I'll need to run some experiments and do some consulting before I can advise on DH group support detection.

douzzer, 2022-07-12:
@tobiasbrunner I believe I have the root cause explanation for the DH group support detection glitch:

5.4.0 introduces a changed default for the bignum math back end -- equivalent to `--enable-sp-math-all`.  Because you don't explicitly designate the math back end with your config, unbeknownst to you your configuration effectively changed with 5.4.0.  `sp-math-all` is much better, and you should probably follow the new default, but you'll need to fix your code.

Your current code assumes that all key sizes are available `#if !defined(USE_FAST_MATH)`, but that's not true with sp-math-all.  In `sp-math-all`, a global max key size is set in `SP_INT_BITS`, and bignum buffers are allocated to be sufficient for that key size and can't be enlarged (no internal `realloc()`).

Note that `SP_INT_BITS` is the key size, not twice the key size as with `FP_MAX_BITS`.  So your new code would be a something like
```
    #if (defined(USE_FAST_MATH) && FP_MAX_BITS >= (3072 * 2)) \
        || (defined(WOLFSSL_SP_MATH_ALL) && SP_INT_BITS >= 3072) \
        || defined(USE_INTEGER_HEAP_MATH)
            PLUGIN_PROVIDE(KE, MODP_3072_BIT),
    #endif
```
Note this omits support for the special-purpose configuration `--enable-sp-math`, which despite the similar name is quite different from `--enable-sp-math-all`, supporting only certain preconfigured keysizes with tuned machine-generated implementations.

tobiasbrunner, 2022-07-13:
Thanks for the quick replies.

> Is `libm.so` missing when you run that `ldd` command?

Yes, I get this:

```
$ ldd src/.libs/libwolfssl.so
	linux-vdso.so.1 (0x00007ffee69f1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6ae04e7000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6ae02f5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f6ae05d9000)
```

> 5.4.0 introduces a changed default for the bignum math back end -- equivalent to `--enable-sp-math-all`. Because you don't explicitly designate the math back end with your config, unbeknownst to you [...]

Yes, I saw that in the changelog (it's what lead me to `--enable-heapmath`), but it wasn't immediately clear what to check for.

> Note that `SP_INT_BITS` is the key size, not twice the key size as with `FP_MAX_BITS`. So your new code would be a something like
>
> ```
>     #if (defined(USE_FAST_MATH) && FP_MAX_BITS >= (3072 * 2)) [...]

Makes sense. I've adapted our checks accordingly. Is `--with-max-rsa-bits` the correct way to tune this (if not using `--enable-heapmath`)?

tobiasbrunner, 2022-07-13:
> Yes, I get this:
> 
> ```
> $ ldd src/.libs/libwolfssl.so
> 	linux-vdso.so.1 (0x00007ffee69f1000)
> 	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6ae04e7000)
> 	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6ae02f5000)
> 	/lib64/ld-linux-x86-64.so.2 (0x00007f6ae05d9000)
> ```

Note that the above is on Ubuntu 20.04 with GCC 9.4.0. In our Debian testing environment with GCC 10.2.1 I get this:

```
$ ldd src/.libs/libwolfssl.so
	linux-vdso.so.1 (0x00007ffe677e8000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd3feba7000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd3feb85000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd3fe9c0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd3fedc4000)
```

douzzer, 2022-07-13:
@tobiasbrunner It sounds like there might be an autotools disparity between Ubuntu 20.04 and your testing environment.  Are you building from git source on Ubuntu 20.04, starting with `./autogen.sh`?  Same question, re building in your testing environment.

It could help to look at the generated `configure` and `Makefile` in each env -- can you share them?  And first, assuming you're building with `./autogen.sh`, it will help to see the output from
```
$ autoconf --version; automake --version
```

Another possibility is that the Ubuntu 20.04 gcc 9.4.0 default compiler configuration is somehow off, though that would be a real trip down the rabbit hole to debug.  Your best bet then would probably be to try upgrading the compiler (9.4.0 is pretty old now).

I've just done a _pro forma_ test build using the pregenerated `configure` script (from autoconf 2.69) in our commercial 5.4.0 release bundle, using your configure options.  It correctly produces a libwolfssl.so that links to libm, so that's good then.

douzzer, 2022-07-13:
>Is `--with-max-rsa-bits` the correct way to tune this (if not using `--enable-heapmath`)?

Yes, that will tune the supported keysize for both fastmath (the old default bignum math back end) and sp-math-all (the new default back end).

As for how to sense supported keysizes at build time in the application, I have started developing a patch that will add back-end-agnostic macros you can test, though obviously that won't help supporting release 5.4.0.

tobiasbrunner, 2022-07-14:
> Are you building from git source on Ubuntu 20.04, starting with `./autogen.sh`? Same question, re building in your testing environment.

Yes, both are built from the Git repository. After pushing the modified DH checks to our repository I noticed that the builds on GitHub that use Ubuntu 20.04 worked correctly. Only my local build seemed to fail. So I've compared the configure options again and I noticed that I locally added `-Werror` to `C_EXTRA_FLAGS`. This apparently breaks the `"checking for cos in -lm"` configure check (via `LT_LIB_M` macro):
```
conftest.c:67:6: error: conflicting types for built-in function 'cos'; expected 'double(double)' [-Werror=builtin-declaration-mismatch]
```
Also a few others with similar errors (e.g. for `memset` and `strftime`). After building without `-Werror`, `libm` was correctly linked.

tobiasbrunner, 2022-07-14:
> indeed autotools is super-brittle.

Yeah, just had a similar issue with newer versions of Autoconf and the `AC_PROG_LEX` macro, which failed if `CFLAGS` contained `-Werror`.

>  We include `-Werror` by default in configure.ac in any case.

Ah, good to know.

> Is all of the above resolved now? Seems like yes?

Yes, thanks.

closed 2022-07-14 by tobiasbrunner