wolfSSL/wolfssl #5354: [Bug]: Building with WOLFSSL_SP_ARM_ARCH=6 still has CLZ instruction

res0nance, 2022-07-13:
### Version

5.4.0

### Description

PR https://github.com/wolfSSL/wolfssl/pull/5240 introduced a new flag to build without the `clz` instruction. But https://github.com/wolfSSL/wolfssl/pull/5322 introduced a `clz` instruction without being guarded by the preprocessor macro as a result the compiler will use the definition with the `clz` instruction.

### Reproduction steps

Build with C_EXTRA_FLAGS=-DWOLFSSL_SP_ARM_ARCH=6

SparkiDev, 2022-07-13:
Hi @res0nance,

The sp_int.c assembly code does not take the WOLFSSL_SP_ARM_ARCH into account yet.
If you cannot use this implementation, then please define: WOLFSSL_SP_DIV_WORD_HALF

I have put up a pull request that I believe fixes the issue: #5358

Let me know if it works for you.

Sean

res0nance, 2022-07-14:
> Hi @res0nance,
> 
> The sp_int.c assembly code does not take the WOLFSSL_SP_ARM_ARCH into account yet. If you cannot use this implementation, then please define: WOLFSSL_SP_DIV_WORD_HALF
> 
> I have put up a pull request that I believe fixes the issue: #5358
> 
> Let me know if it works for you.
> 
> Sean

The pull request compiles. Thanks!

SparkiDev, 2022-07-14:
Hi @res0nance

The pull request has been merged and will be in the next release.
Thanks for letting us know.

Sean

closed 2022-07-14 by SparkiDev