wolfSSL/wolfssl #5366: [Bug]: WOLFSSL struct returned by wolfSSL_write_dup() uses incorrect TLS version

rvlad-patrascu, 2022-07-16:
### Contact Details

vladp@opensips.org

### Version

tested with 4.7.1r and 5.4.0

### Description

The `WOLFSSL` struct returned by `wolfSSL_write_dup()` seems to use the max TLS version inherited from the `WOLFSSL_CTX` instead of the one actually negotiated for the session. As such, after accepting a connection and getting the write duplicate structure, doing operations like `wolfSSL_write()`, `wolfSSL_shutdown()` etc. using this object may lead to errors. I've observed issues when the client requests a specific cipher that is not support in the higher TLS version used by the write dup object.

Doing the following work-around fixes the issue: check the TLS version of the original original `WOLFSSL` object and enforce it for the write duplicate object, with `wolfSSL_set_options()` and setting a flag such as `WOLFSSL_OP_NO_TLSv1_3` etc. But I'm  wondering if there may be other issues I'm missing with the write duplicate object.

### Reproduction steps

1. Create a server application where you configure the context, for example as such:
```
ctx = wolfSSL_CTX_new(wolfSSLv23_method())
wolfSSL_CTX_set_min_proto_version(ctx, TLS1_2_VERSION);
wolfSSL_CTX_set_max_proto_version(ctx, TLS1_3_VERSION);
```
2. `openssl s_client -connect <server_hostname> -tls1_2 -cipher ECDHE-RSA-AES128-SHA25`
3. error on client side: `139759000007936:error:1408F119:SSL routines:ssl3_get_record:decryption failed or bad record mac:ssl/record/ssl3_record.c:521:`
4. on the server application, observe that the TLS versions differ between the original `WOLFSSL` object and write duplicate, by checking with: `wolfSSL_version()`

### Relevant log output

```shell
-313, received alert fatal error
```

embhorn, 2022-07-18:
Hello @rvlad-patrascu 

Thanks for this excellent report. We will review and post an update.

Thanks,
Eric @ wolfSSL Support

embhorn, 2022-07-22:
Hi @rvlad-patrascu 

Please test with the fix in #5391 

Thanks,
@embhorn

rvlad-patrascu, 2022-08-02:
Hi @embhorn 

Looks like it's fixed now, thank you.

PR #5391 (2022-07-25): Fix to copy TLS version with wolfSSL_write_dup
# Description

`wolfSSL_write_dup`  was not copying the version structure from the SSL object.

Fixes #5366 

# Testing

Added server writedup example: https://github.com/wolfSSL/wolfssl-examples/pull/329

# Checklist

 - [X] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation