wolfSSL/wolfssl #5367: use ECDHE_ECDSA_WITH_AES256_CCM_8 in TLS 1.3

DongS2, 2022-07-17:
### Version

5.4.0

### Description

I want to use ECDHE_ECDSA_WITH_AES256_CCM_8 in TLS 1.3.

so i use like that on server and client:
`./configure --enable-ecc --enable-aescm`

[image]

It works well in TLS 1.2, but not works in TLS 1.3.
I captured client hello packet, it's supported_version is 1.2 and it has key_share extension.
I don't know why it doesn't work.

how can i solved this problem ??

SparkiDev, 2022-07-17:
Note that ECDHE-ECDSA-AES256-CCM-8 is not a TLS 1.3 cipher suite and you have asked the server to only do TLS 1.3.
The correct cipher suite for TLS 1.3 is: TLS13-AES128-CCM-8-SHA256

SparkiDev, 2022-07-25:
Hi @DongS2 

Have you been able to get TLS 1.3 to work with CCM_8?

Sean