wolfSSL/wolfssl #5386: [Bug]: possible typo or missing macro OR statement in sp_int.c

mpratt14, 2022-07-22:
### Contact Details

mcpratt@pm.me

### Version

5.4.0-stable

### Description

for `sp_rand_prime()` in sp_int.c

comment states that it is for RSA only, but the preprocessor macros used do not include RSA

```
#if defined(WOLFSSL_KEY_GEN) && (!defined(NO_DH) || !defined(NO_DSA)) && \
    !defined(WC_NO_RNG)
/* Generate a random prime for RSA only.
...
...
```

### Reproduction steps

building fails if NO_DH and NO_DSA are defined but RSA macros left untouched (RSA generally enabled)

### Relevant log output

```shell
see for more information
https://github.com/openwrt/openwrt/issues/10311
```

kareem-wolfssl, 2022-07-22:
Hi @mpratt14 ,

Thanks for the bug report, indeed I can reproduce this with the configure line from the OpenWRT issue: `./configure --enable-reproducible-build --enable-lighty --enable-opensslall --enable-opensslextra --enable-sni --enable-stunnel --enable-altcertchains --disable-crypttests --disable-examples --disable-jobserver --enable-ipv6 --disable-aesccm --disable-certgen --disable-chacha --disable-poly1305 --disable-dh --disable-arc4 --disable-tlsv10 --disable-tls13 --disable-session-ticket --disable-dtls --disable-curve25519 --disable-curve448 --disable-afalg --disable-openvpn --enable-devcrypto=no`

wolfSSL_BN_generate_prime_ex is using sp_rand_prime, but sp_rand_prime is being excluded from the build with NO_RSA undefined while wolfSSL_BN_generate_prime_ex is not.  Please try this PR and confirm it fixes the issue for you: https://github.com/wolfSSL/wolfssl/pull/5390

graysky2, 2022-07-22:
@kareem-wolfssl - I tested that PR and can confirm a successful build.

graysky2, 2022-07-22:
@kareem-wolfssl - if you agree that your PR is the official fix, will you be making a new stable point release immediately?  If not, I am happy to include your patch as a fix with 5.4.0.

kareem-wolfssl, 2022-07-22:
@graysky2 We don't make point releases except for major security issues.  This will be included in our next release 5.5.0, you will need to include it as a patch in the meantime.

PR #5390 (2022-07-24): Update sp_rand_prime's preprocessor gating to match wolfSSL_BN_generate_prime_ex's.
# Description

wolfSSL_BN_generate_prime_ex is using sp_rand_prime, but sp_rand_prime is being excluded from the build with NO_RSA undefined while wolfSSL_BN_generate_prime_ex is not.  Update preprocessor gating to match.

Fixes #5386

# Testing

Confirmed building and passing all tests

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-07-24 by SparkiDev