wolfSSL/wolfssl #5389: [Bug]: 8K keys are broken

vaintroub, 2022-07-22:
### Contact Details

wlad@mariadb.com

### Version

5.4.0

### Description

Found by trying to integrate new wolfSSL version into mariadb. 5.2.0 still worked fine with certificates that use large (8K) keys. 5.4.0 is [ broken in this regard](https://ci.appveyor.com/project/rasmushoj/server/builds/44236921?fullLog=true#L1690)

### Reproduction steps

To reproduce issue, using plain wolfssl build
1. clone wolfssl repo
2. mkdir xxx
3. cd xxx
4. cmake .. -DCMAKE_C_FLAGS="-DFP_MAX_BITS=16384"
5. make client server
6. Unpack the attached zip with certificates in the current directory  [certs.zip](https://github.com/wolfSSL/wolfssl/files/9169710/certs.zip)
7. cd ..
8. examples/server/server   -c ./xxx/server8k-cert.pem -k ./xxx/server8k-key.pem -A ./xxx/cacert.pem   &
9. examples/client/client    -c ./xxx/client-cert.pem  -k ./xxx/client-key.pem  -A ./xxx/cacert.pem

   

### Relevant log output

```shell
wolfSSL_connect error -155, ASN sig error, confirm failure
wolfSSL error: wolfSSL_connect failed
wlad@desktop:~/wolfssl$ SSL_accept error -143, ASN key init error, invalid input
wolfSSL error: SSL_accept failed
```

kareem-wolfssl, 2022-07-22:
Hi @vaintroub ,

I will work on reproducing this issue here to confirm, but at a glance it looks like the switch to SP math by default is the problem here.  If so you will need to use `-DRSA_MAX_SIZE=8192` instead.

Thanks,
Kareem

vaintroub, 2022-07-22:
I tried RSA_MAX_SIZE=8192 and this did not work. Kinda expected, because we (MariaDB) do not use SP math, and we do not use the CMakeLists.txt provided by wolfssl itself. [This is our CMakeLists.txt](https://github.com/MariaDB/server/blob/10.10/extra/wolfssl/CMakeLists.txt) .

SparkiDev, 2022-07-24:
Please try using both -DFP_MAX_BITS=16384 and -DRSA_MAX_SIZE=8192 and let us know if this works.

Thanks,
Sean

vaintroub, 2022-07-25:
-DFP_MAX_BITS=16384 -DRSA_MAX_SIZE=8192 seems to work.  Is it possible to have one preprocessor constant than causes large certificate to work, instead of 3?

SparkiDev, 2022-07-25:
Hi vaintroub,

I'll look into it and get back to you.
The RSA_MAX_SIZE is for the TLS layer and the others are for the maths implementations.
SP_INT_BITS is for the sp_int.c implementation so that won't be needed for you.

Sean

SparkiDev, 2022-07-27:
I've put PR up that makes FP_MAX_BITS and SP_INT_BITS set the rest.
MySQL is handled specially to ensure 8192 is supported.

closed 2022-07-28 by dgarske

SparkiDev, 2022-07-28:
Hi @vaintroub

The pull request has been merged.

You shouldn't need to do anything but define WOLFSSL_MYSQL_COMPATIBLE.

Let me know if this isn't true.

Sean

PR #5406 (2022-07-28): Maths bit size: ensure size is right
# Description

Ensure WOLFSSL_MAX_RSA_BITS works.
Fix SP math size defines.
Regression testing maths implementations.

Fixes #5389

# Testing

Regression tested maths implementations.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation