wolfSSL/wolfssl #5396: [Bug]: Android x86 builds return SSL Errors

res0nance, 2022-07-25:
### Version

5.4.0

### Description

Target Environment:

Android NDK 23c with `-march=i686 -msse3 -m32 -O3`.

The resulting build can no longer establish a TLS / DTLS connection. Using the option `--enable-heapmath` fixes the issue.

SparkiDev, 2022-07-25:
Hi @res0nance

Could you try using the tfm.c code with --enable-fastmath.
Also, instead, define: SP_WORD_SIZE=32

Sean

res0nance, 2022-07-25:
Setting `SP_WORD_SIZE=32` does not seem to change anything.

Enabling fastmath results in the following error
```
wolfcrypt/src/tfm.c:3257:11: error: inline assembly requires more registers than available
          SQRADD(A->dp[ix>>1], A->dp[ix>>1]);
          ^
./wolfcrypt/src/asm.c:701:6: note: expanded from macro 'SQRADD'
     "movl  %3,%%eax     \n\t"                            \
     ^
wolfcrypt/src/tfm.c:3251:11: error: inline assembly requires more registers than available
          SQRADD2(*tmpx++, *tmpy--);
          ^
./wolfcrypt/src/asm.c:712:6: note: expanded from macro 'SQRADD2'
     "movl  %3,%%eax     \n\t"                            \
     ^
wolfcrypt/src/tfm.c:3251:11: error: inline assembly requires more registers than available
./wolfcrypt/src/asm.c:712:6: note: expanded from macro 'SQRADD2'
     "movl  %3,%%eax     \n\t"                            \
     ^
wolfcrypt/src/tfm.c:3251:11: error: inline assembly requires more registers than available
./wolfcrypt/src/asm.c:712:6: note: expanded from macro 'SQRADD2'
     "movl  %3,%%eax     \n\t"                            \
     ^
4 errors generated.
```

SparkiDev, 2022-07-27:
Please try --enable-32bit and --enable-fastmath.

Thanks,
Sean

res0nance, 2022-07-27:
> Please try --enable-32bit and --enable-fastmath.
> 
> Thanks, Sean

This resulted in the exact same compiler error

SparkiDev, 2022-07-27:
Hi @res0nance 

I've put up a pull request that may help.
It appears clang isn't handling the inline assembly register/memory assignation.

Please give this a go and let me know if --enable-fastmath works for you.

Thanks,
Sean

res0nance, 2022-07-27:
> Hi @res0nance
> 
> I've put up a pull request that may help. It appears clang isn't handling the inline assembly register/memory assignation.
> 
> Please give this a go and let me know if --enable-fastmath works for you.
> 
> Thanks, Sean

I'll test this out tomorrow and report back, do we have any idea as to why the sp-math-all default implementation is causing errors?

SparkiDev, 2022-07-27:
No I don't know. It shouldn't be any different but there must be something in there that will be hard to track unless you give me more detail about the error that you see.

It would be good if you could run testwolfcrypt but I don't know if that's possible in your environment.

res0nance, 2022-07-28:
> No I don't know. It shouldn't be any different but there must be something in there that will be hard to track unless you give me more detail about the error that you see.
> 
> It would be good if you could run testwolfcrypt but I don't know if that's possible in your environment.

Tried both, it seems that fastmath is working now.

I tried to get `testwolfcrypt` to run by compiling it and pushing it to emulator via `adb push testwolfcrypt /data/local/tmp` Then running it, unfortunately all it does is return 255.

I think the `255` might be just `-1`. I can't get it to run with spmath or with fastmath.

res0nance, 2022-07-28:
@dgarske Yes that fixes `fastmath` but the default that wolf uses is `sp-math`, this option unfortunately does not work. Should platforms use this option instead? I would much rather see that the default option produces a working library so users would not have to deal with all the caveats.

dgarske, 2022-07-28:
Sorry it closed automatically with the PR merge. I'll keep this open for @SparkiDev to review.

SparkiDev, 2022-07-28:
I would too! :-)
But I can't diagnose the error from here.
Was there any output from the testwolfcrypt?
It could either be going to a log, if ANDROID is defined, or stdout.

res0nance, 2022-07-29:
To future readers, on the android platform for the `testwolfcrypt` binary the output goes to `logcat`

Firstly, I'd like to thank you for your patience in this process. Getting this all to work was quite confusing on my end.

Moving on to my findings.

The filesystem seems not to be able to read the cert despite me copying the entire wolfssl folder into the emulator (i think it doesn't work on a real device as well). So I had to workaround with `--disable-filesystem` and `USE_CERT_BUFFERS`

Side note: https://www.wolfssl.com/documentation/manuals/wolfssl/chapter02.html is out of date it lacks the `USE_CERT_BUFFERS_256`, `USE_CERT_BUFFERS_4096` and `USE_CERT_BUFFERS_3072` options

With USE_CERT_BUFFERS_1024, 2048, 3072 options the RSA test passes

But with 4096 spmath fails with

```
07-29 11:55:45.930  3069  3069 D [WOLFCRYPT]: ------------------------------------------------------------------------------
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]:  wolfSSL version 5.4.0
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: ------------------------------------------------------------------------------
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: error    test passed!
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: MEMORY   test passed!
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: base64   test passed!
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: asn      test passed!
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: RANDOM   test passed!
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: MD5      test passed!
07-29 11:55:45.931  3069  3069 D [WOLFCRYPT]: SHA      test passed!
07-29 11:55:45.932  3069  3069 D [WOLFCRYPT]: SHA-256  test passed!
07-29 11:55:45.933  3069  3069 D [WOLFCRYPT]: SHA-384  test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: SHA-512  test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: Hash     test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: HMAC-MD5 test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: HMAC-SHA test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: HMAC-SHA256 test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: HMAC-SHA384 test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: HMAC-SHA512 test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: HMAC-KDF    test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: TLSv1.3 KDF test passed!
07-29 11:55:45.935  3069  3069 D [WOLFCRYPT]: GMAC     test passed!
07-29 11:55:45.938  3069  3069 D [WOLFCRYPT]: Chacha   test passed!
07-29 11:55:45.938  3069  3069 D [WOLFCRYPT]: POLY1305 test passed!
07-29 11:55:45.938  3069  3069 D [WOLFCRYPT]: ChaCha20-Poly1305 AEAD test passed!
07-29 11:55:45.938  3069  3069 D [WOLFCRYPT]: AES      test passed!
07-29 11:55:45.938  3069  3069 D [WOLFCRYPT]: AES192   test passed!
07-29 11:55:45.938  3069  3069 D [WOLFCRYPT]: AES256   test passed!
07-29 11:55:45.951  3069  3069 D [WOLFCRYPT]: AES-GCM  test passed!
07-29 11:55:45.951  3069  3069 D [WOLFCRYPT]: RSA      test failed!
07-29 11:55:45.951  3069  3069 D [WOLFCRYPT]:  error = -7904
07-29 11:55:45.951  3069  3069 D [WOLFCRYPT]: Exiting main with return code: -1
```

This test will pass with `--enable-heapmath`

dgarske, 2022-07-29:
With SP math did you enable 4096-bit support using `WOLFSSL_SP_4096`? If cross compiling this may not get set with `--enable-sp` unless you do `--enable-sp=yes,4096 --enable-sp-math`.

RipleyTom, 2022-07-30:
We've been getting `mp_exptmod error state` errors from our software after connecting to a server with a RSA 4096 certificate after updating to 5.4.0.
Enabling fast-math also seems to have fixed this.
Options used at the time are listed here:
https://github.com/RPCS3/rpcs3/blob/c00d47cc99952c45996c0eab5f6a35c39e7a70fe/3rdparty/wolfssl/CMakeLists.txt

We're probably just going to turn fast-math on(tbh I can't remember why it was set to off, some build issues maybe?) but I assume this is no expected behaviour.

SparkiDev, 2022-07-31:
Hi @RipleyTom

SP math all is now the default maths implementation for wolfSSL.
This may have caused the change.
Please set SP_INT_BITS=4096 to allow for 4096 bit RSA keys.
There is a pull request out that should fix these types of issues.

SparkiDev, 2022-07-31:
Hi @res0nance,

The error appears to be happening in the decoding of the public key, which is odd.

Please try defining WOLFSSL_ASN_TEMPLATE and WOLFSSL_DEBUG_ASN_TEMPLATE.
Then run the test again.
This will change the ASN.1 parsing code and allow detailed debugging information.

Thanks,
Sean

res0nance, 2022-08-01:
@SparkiDev 

Here is the test output

```
08-01 11:35:06.562  6148  6148 D [WOLFCRYPT]: ------------------------------------------------------------------------------
08-01 11:35:06.562  6148  6148 D [WOLFCRYPT]:  wolfSSL version 5.4.0
08-01 11:35:06.562  6148  6148 D [WOLFCRYPT]: ------------------------------------------------------------------------------
08-01 11:35:06.563  6148  6148 D [WOLFCRYPT]: error    test passed!
08-01 11:35:06.563  6148  6148 D [WOLFCRYPT]: MEMORY   test passed!
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too big
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too big
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too big
08-01 11:35:06.563  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.564  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too big
08-01 11:35:06.564  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.564  6148  6148 I chatty  : uid=2000(shell) ./wolfcrypt/test/testwolfcrypt identical 10 lines
08-01 11:35:06.564  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.564  6148  6148 V [wolfSSL]: Bad end of line in Base64 Decode
08-01 11:35:06.564  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.566  6148  6148 I chatty  : uid=2000(shell) ./wolfcrypt/test/testwolfcrypt identical 27 lines
08-01 11:35:06.566  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too small
08-01 11:35:06.567  6148  6148 V [wolfSSL]: Bad Base64 Decode bad character
08-01 11:35:06.568  6148  6148 I chatty  : uid=2000(shell) ./wolfcrypt/test/testwolfcrypt identical 14 lines
08-01 11:35:06.568  6148  6148 V [wolfSSL]: Bad Base64 Decode bad character
08-01 11:35:06.569  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too big
08-01 11:35:06.575  6148  6148 I chatty  : uid=2000(shell) ./wolfcrypt/test/testwolfcrypt identical 131 lines
08-01 11:35:06.575  6148  6148 V [wolfSSL]: Bad Base64 Decode data, too big
08-01 11:35:06.575  6148  6148 D [WOLFCRYPT]: base64   test passed!
08-01 11:35:06.575  6148  6148 V [wolfSSL]: TEMPLATE: dateASN
08-01 11:35:06.575  6148  6148 V [wolfSSL]: wolfSSL Entering GetASN_Items
08-01 11:35:06.575  6148  6148 V [wolfSSL]:  0:    0   13    UTCTime
08-01 11:35:06.575  6148  6148 D [WOLFCRYPT]: asn      test passed!
08-01 11:35:06.576  6148  6148 D [WOLFCRYPT]: RANDOM   test passed!
08-01 11:35:06.576  6148  6148 D [WOLFCRYPT]: MD5      test passed!
08-01 11:35:06.576  6148  6148 D [WOLFCRYPT]: SHA      test passed!
08-01 11:35:06.577  6148  6148 D [WOLFCRYPT]: SHA-256  test passed!
08-01 11:35:06.578  6148  6148 D [WOLFCRYPT]: SHA-384  test passed!
08-01 11:35:06.579  6148  6148 D [WOLFCRYPT]: SHA-512  test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: Hash     test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: HMAC-MD5 test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: HMAC-SHA test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: HMAC-SHA256 test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: HMAC-SHA384 test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: HMAC-SHA512 test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: HMAC-KDF    test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: TLSv1.3 KDF test passed!
08-01 11:35:06.580  6148  6148 D [WOLFCRYPT]: GMAC     test passed!
08-01 11:35:06.584  6148  6148 D [WOLFCRYPT]: Chacha   test passed!
08-01 11:35:06.584  6148  6148 D [WOLFCRYPT]: POLY1305 test passed!
08-01 11:35:06.584  6148  6148 D [WOLFCRYPT]: ChaCha20-Poly1305 AEAD test passed!
08-01 11:35:06.584  6148  6148 D [WOLFCRYPT]: AES      test passed!
08-01 11:35:06.584  6148  6148 D [WOLFCRYPT]: AES192   test passed!
08-01 11:35:06.584  6148  6148 D [WOLFCRYPT]: AES256   test passed!
08-01 11:35:06.597  6148  6148 D [WOLFCRYPT]: AES-GCM  test passed!
08-01 11:35:06.597  6148  6148 V [wolfSSL]: TEMPLATE: &rsaPublicKeyASN[RSAPUBLICKEYASN_IDX_PUBKEY_RSA_SEQ]
08-01 11:35:06.597  6148  6148 V [wolfSSL]: wolfSSL Entering GetASN_Items
08-01 11:35:06.597  6148  6148 V [wolfSSL]:  0:    2    0 +    SEQUENCE              Tag=0x02 (INTEGER)
08-01 11:35:06.597  6148  6148 V [wolfSSL]: Bad tag
08-01 11:35:06.597  6148  6148 V [wolfSSL]: TEMPLATE: rsaPublicKeyASN
08-01 11:35:06.597  6148  6148 V [wolfSSL]: wolfSSL Entering GetASN_Items
08-01 11:35:06.597  6148  6148 V [wolfSSL]:  0:    2    0 +  SEQUENCE                Tag=0x02 (INTEGER)
08-01 11:35:06.598  6148  6148 V [wolfSSL]: Bad tag
08-01 11:35:06.598  6148  6148 V [wolfSSL]: TEMPLATE: &rsaPublicKeyASN[RSAPUBLICKEYASN_IDX_PUBKEY_RSA_SEQ]
08-01 11:35:06.598  6148  6148 V [wolfSSL]: wolfSSL Entering GetASN_Items
08-01 11:35:06.598  6148  6148 V [wolfSSL]: GetLength - value exceeds buffer length
[... 142 lines cut ...]
08-01 11:35:06.607  6148  6148 V [wolfSSL]:  2:    4    9      OBJECT ID
08-01 11:35:06.607  6148  6148 V [wolfSSL]:  3:   15    0      NULL
08-01 11:35:06.607  6148  6148 V [wolfSSL]:  4:   17    9     BIT STRING
08-01 11:35:06.607  6148  6148 V [wolfSSL]:  5:   20    6 +    SEQUENCE
08-01 11:35:06.607  6148  6148 V [wolfSSL]:  6:   22    1       INTEGER
[... cut ...]
08-01 11:35:06.608  6148  6148 V [wolfSSL]:  2:    7  513     INTEGER
08-01 11:35:06.608  6148  6148 V [wolfSSL]: Failed to read mp: -3
08-01 11:35:06.608  6148  6148 D [WOLFCRYPT]: RSA      test failed!
08-01 11:35:06.608  6148  6148 D [WOLFCRYPT]:  error = -7904
08-01 11:35:06.608  6148  6148 V [wolfSSL]: wolfSSL Entering wolfCrypt_Cleanup
08-01 11:35:06.608  6148  6148 D [WOLFCRYPT]: Exiting main with return code: -1
```

I'll try to get the output when this is run with our regular workload instead of the test later

SparkiDev, 2022-08-01:
Hi @res0nance

It definitely looks like the conversion from binary of the numbers is failing.
There isn't much that can fail in there!
Make sure SP_INT_BITS is 4096 and SP_WORD_SIZE is 32.

Also, try with -O2 instead of -03 or even debug.

Thanks,
Sean

res0nance, 2022-08-02:
I added `SP_INT_BITS=4096` and it now seems to be working. Looking around I found that on 64bit platforms this is automatically set which is probably why 64 bit doesn't have this issue.

> With SP math did you enable 4096-bit support using `WOLFSSL_SP_4096`? If cross compiling this may not get set with `--enable-sp` unless you do `--enable-sp=yes,4096 --enable-sp-math`.

Probably this is exactly the reason why. Not sure if this should be classified a bug but this did break when updating from 5.3.0 to 5.4.0. I'm going to try ensuring all our builds have these flags set for `--enable-sp`

SparkiDev, 2022-08-15:
Hi @res0nance,

Is there anything more for this issue?
If not I'll come it.

Thanks,
Sean

res0nance, 2022-08-16:
> Hi @res0nance,
> 
> Is there anything more for this issue? If not I'll come it.
> 
> Thanks, Sean

No I think its fine, Thanks for everything.

PR #5405 (2022-07-28): TFM: change inline x86 asm code to compile with clang
# Description

TFM not compiling for x86 with clang.

Fixes #5396

# Testing

./configure '--disable-shared' '--enable-32bit' '--enable-fastmath' 'CFLAGS=-m32 -msse3 -march=i686' 'CC=clang'

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-08-16 by res0nance