wolfSSL/wolfssl #5400: C++17 Standard

overkill74, 2022-07-26:
### Version

any / master

### Description

When I try link the library with a C++17 Std code, I get this error:

include/wolfssl/wolfcrypt/sha256.h:235: error: reference to ‘byte’ is ambiguous
WOLFSSL_API int wc_Sha256Final(wc_Sha256* sha256, byte* hash);

Repeated for any "byte" type variable.

It is possible to change the "byte", "word32" etc typedef with a more standard "uint8_t", "unint32_t" etc?
It could be possible typdef byte for backcompability.
Or there is another way to use this library with a C++17 code?

I can take care about this if you like

embhorn, 2022-07-26:
Hi @overkill74 

You can override the default types using `WOLFSSL_TYPES`, in which case you will need to provide the following typedefs:
https://github.com/wolfSSL/wolfssl/blob/a98642ba61f1848b819b307fd1ec084306c35680/wolfssl/wolfcrypt/types.h#L70-L88

Let us know if there are questions.

Kind regards,
@embhorn wolfSSL Support

overkill74, 2022-07-29:
Hi @embhorn,
thank you for your answer.

The problem is not how I can define my personal 'byte', but it is related to C++17 standard.
C++17 standard defines the class std::byte (https://en.cppreference.com/w/cpp/types/byte), so anytime the type 'byte' is found in the source code, the compiler does not know which one is the 'byte' type and generate the error.
So my suggesiton is to deprecate the use of 'byte' type int the WolfSSL code and use 'word8', for example, instead of it.

P.S.
Generally I personally prefer to use the standard types 'uint8_t' etc, in this way it not needed to redefine them, while compile to differents architecture (32/64bit), but it's just mine humble point of view. 'word8' is very good to solve the conflict.

embhorn, 2022-07-29:
HI @overkill74 

Obviously wolfSSL is not the only C library that has defined the byte type. Here is a good discussion of how users are working around this issue with C++ 17:
https://stackoverflow.com/questions/45125957/byte-and-ambiguous-symbol-due-to-using-declarations

closed 2022-07-29 by embhorn

dgarske, 2023-03-02:
Recommended workaround, which remaps the wolfssl `byte` when including wolfSSL headers in your C++ application.

```
typedef unsigned char byte_workaround;
#define byte byte_workaround
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/types.h>
#include <wolfssl/wolfcrypt/aes.h>
#include <wolfssl/wolfcrypt/ecc.h>
#include <wolfssl/wolfcrypt/random.h>
#undef byte
```

PR #6147 (2023-03-02): Fixes for building with C++17
# Description

Added note on byte redefinition issue when customer application sets default namespace to "std" or wants to use "byte".

Fixes 15729
Fixes #5400

# Testing

Tested using `./configure CC="g++ -std=c++17" --enable-all && make check`.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation