wolfSSL/wolfssl #5409: RFC 9266: Channel Bindings for TLS 1.3 support

Neustradamus, 2022-07-27:
### Version

last

### Description

Can you add the support of RFC 9266: Channel Bindings for TLS 1.3?
- https://datatracker.ietf.org/doc/html/rfc9266 

Little details, to know easily:
- tls-unique for TLS =< 1.2
- tls-exporter for TLS = 1.3 

Thanks in advance.

embhorn, 2022-07-27:
Hi @Neustradamus 

I'd be happy to create a formal feature request. Please send an email to support@wolfssl.com to initiate the process.

Thanks,
@embhorn wolfSSL Support

embhorn, 2022-07-29:
Handling feature request in ZD, so closing this GH issue.

closed 2022-07-29 by embhorn

Neustradamus, 2023-02-14:
@embhorn: Have you progressed on it?

embhorn, 2023-02-14:
Hi @Neustradamus 

Your feature request was added to our internal queue. As you never specified a location or project details, we cannot prioritize your request.

We try to work on requested features whenever we get free engineering cycles unless a customer decides to back the effort for a finite implementation timeline. If at any time a feature becomes a progress blocker for a project it can be accelerated. That process can be started by contacting the account business manager.

Email: [facts@wolfssl.com](mailto:facts@wolfssl.com)

Neustradamus, 2023-11-19:
Dear @wolfSSL team, @embhorn,

I think that you have seen the jabber.ru MITM and Channel Binding is the solution:
- https://notes.valdikss.org.ru/jabber.ru-mitm/
- https://snikket.org/blog/on-the-jabber-ru-mitm/
- https://www.devever.net/~hl/xmpp-incident
- https://blog.jmp.chat/b/certwatch

Neustradamus, 2025-11-13:
Dear @wolfSSL team, dear all,

After several years of patience, I recall you that:
- tls-unique is not secure, vulnerable
- tls-server-end-point support is always missing
- tls-exporter support is always missing

Thanks in advance.

Neustradamus, 2026-09-30:
Dear @cconlon, @LinuxJedi, @danielinux, @anhu, @dgarske, @Frauschi, @mattia-moffa, @AlexLanzano, @tamasan238, @embhorn, @julek-wolfssl, @aidankeefe2022, @stenslae, @aidangarske, and others,

Have you progressed on it?

Security is important.

Thanks in advance.

julek-wolfssl, 2026-09-30:
Hi @Neustradamus ,

the `tls-server-end-point` hash can be accessed through the compatibility layer in the latest release. Its a little convoluted so I would recommend `tls-exporter` which is implemented since release 4.7.0 (via `wolfSSL_export_keying_material` API).

Sincerely
Juliusz

Neustradamus, 2026-09-30:
@julek-wolfssl: Thanks for your answer but I do not find tls-unique/tls-server-end-point/tls-exporter in the source code:
- https://github.com/search?q=repo%3AwolfSSL%2Fwolfssl%20tls-exporter&type=code
- https://github.com/search?q=repo%3AwolfSSL%2Fwolfssl%20tls-server-end-point&type=code 
- https://github.com/search?q=repo%3AwolfSSL%2Fwolfssl%20tls-unique&type=code

Can you add it?

julek-wolfssl, 2026-09-30:
"exporter" exists https://github.com/search?q=repo%3AwolfSSL%2Fwolfssl+exporter&type=code as does "tls_unique" https://github.com/search?q=repo%3AwolfSSL%2Fwolfssl+tls_unique&type=code . "tls-server-end-point" currently doesn't have a direct API so it won't be mentioned in the code.

Neustradamus, 2026-09-30:
@julek-wolfssl: Can you add comments in source code to have?
- tls-unique
- tls-exporter
- tls-server-end-point (when it will be added)

It is important to find easily in source code like other softwares, projects, ...

julek-wolfssl, 2026-09-30:
@Neustradamus the channel bindings are sufficiently commented and documented as-is.

Neustradamus, 2026-09-30:
Example in GnuTLS, it is clear...
- https://github.com/search?q=repo%3Agnutls%2Fgnutls%20tls-unique&type=code
- https://github.com/search?q=repo%3Agnutls%2Fgnutls%20tls-server-end-point&type=code
- https://github.com/search?q=repo%3Agnutls%2Fgnutls%20tls-exporter&type=code