wolfSSL/wolfssl #5436: TLS1.3 extensions are inconsistency with  RFC 8446

melissa-cjt, 2022-08-04:
### Version

master

### Description

Hello,
I  find that in RFC 8446 it says the extension should appear in the specified message. While, the implementation of the following three extensions are not comply with RFC.

   | server_name [RFC6066]                                            |      CH, EE    |
   | supported_groups [RFC7919]                                   |      CH, EE    |
   | application_layer_protocol_negotiation [RFC7301]  |      CH, EE    |

In src/tls.c  TLSX_Parse  function, it allowed these extensions appear in msgtype==server_hello.  Are these inconsistency bug ?

          case TLSX_SERVER_NAME:
                if (IsAtLeastTLSv1_3(ssl->version) &&
                        msgType != client_hello &&
                        msgType != server_hello &&
                        msgType != encrypted_extensions) {
                    return EXT_NOT_ALLOWED;
                }

         case TLSX_SUPPORTED_GROUPS:

                if (IsAtLeastTLSv1_3(ssl->version) &&
                        msgType != client_hello &&
                        msgType != server_hello &&
                        msgType != encrypted_extensions) {
                    return EXT_NOT_ALLOWED;
                }

            case TLSX_APPLICATION_LAYER_PROTOCOL:
                WOLFSSL_MSG("ALPN extension received");

                if (IsAtLeastTLSv1_3(ssl->version) &&
                        msgType != client_hello &&
                        msgType != server_hello &&
                        msgType != encrypted_extensions) {
                    return EXT_NOT_ALLOWED;
                }

SparkiDev, 2022-08-04:
Hi @melissa-cjt 

You are correct!
I believe it was a copy and paste error.
A PR has been put up to fix this.

Thanks for taking the time to let us know!

Sean

closed 2022-08-05 by julek-wolfssl

julek-wolfssl, 2022-08-05:
@melissa-cjt Thank you for your report. https://github.com/wolfSSL/wolfssl/pull/5439 has been merged.

PR #5439 (2022-08-05): TLS 1.3: restrict extension validity by message
# Description

Extensions ServerName, SupportedGroups and ALPN must not appear in
server_hello.
Removed server_hello from the valid checks.

Fixes #5436

# Testing

Standard testing

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation