wolfSSL/wolfssl #5446: [Bug]: Not all private PKCS#8 keys could be used with WolfSSL

vvvictor07, 2022-08-09:
### Contact Details

vchernikh@tourmalinecore.com

### Version

v5.4.0-stable, master latest (7004157869e5445b086045ea9e9f86833fb2e330)

### Description

I try to load the private (not encrypted) PKCS#8 key into CTX  with `wolfSSL_CTX_use_PrivateKey_buffer` or just load RsaKey with `wc_RsaPrivateKeyDecode` but it fails (`ToTraditionalInline_ex` fails to skip whole sequence while `GetAlgoId`).

So the main problem is wrong work of the `ToTraditionalInline_ex` that can't find right position of octet string of private key in PKCS#8 file (`AlgorithmIdentifier` could have [parameters field that could be ANY DEFINED BY algorithm OPTIONAL](https://www.itu.int/rec/T-REC-X.509-198811-S) and the problem is when it is Sequence). 

Unsupported key structure:
[image]

### Reproduction steps

1. Download [ps256_cert.zip](https://github.com/wolfSSL/wolfssl/files/9287037/ps256_cert.zip) (_Archive with RSAssaPSS unencrypted private key and cert chain_) and unpack  `ps256.der` key file into `certs` folder.
2. Change `tests/api.c` like below:
```diff
--- a/tests/api.c
+++ b/tests/api.c
@@ -26737,12 +26737,12 @@ static int test_ToTraditional(void)
      defined(OPENSSL_EXTRA_X509_SMALL))
 
     XFILE   f;
-    byte    input[TWOK_BUF];
+    byte    input[FOURK_BUF];
     word32  sz;
 
     printf(testingFmt, "ToTraditional()");
 
-    f = XFOPEN("./certs/server-keyPkcs8.der", "rb");
+    f = XFOPEN("./certs/ps256.der", "rb");
     AssertTrue((f != XBADFILE));
     sz = (word32)XFREAD(input, 1, sizeof(input), f);
     XFCLOSE(f);
-- 
```
3. Run unit_test and get fail

### Relevant log output

```shell
ToTraditional(): failed
test_ToTraditional failed.

ERROR - /tmp/tmp.UavVdGm55a/tests/api.c line 57936 failed with:
    expected: ret == 0
    result:   -132 != 0
```

SparkiDev, 2022-08-10:
Hi @vvvictor07 

Support for RSA-PSS certificates is coming!
We have had some infrastructure issues of late but the handling of RSA-PSS and the parameters is something I've implemented just recently.

If you would like to try it out the PR is: https://github.com/wolfSSL/wolfssl/pull/5397

Sean

SparkiDev, 2022-08-15:
Hi @vvvictor07 

Let us know if there is anything more here.
Otherwise I'll close the ticket.

Thanks,
Sean

vvvictor07, 2022-08-16:
Hi all, I have checked this bug was fixed in #5397
Have a nice day)

closed 2022-08-16 by vvvictor07