wolfSSL/wolfssl #5450: [Bug]: wc_HmacSetKey corrupts memory

Ziemien, 2022-08-10:
### Version

5.4.0

### Description

osx 12.4, clang 14.0.6, used with cmake 

```
set(wolfSSL_DIR ${_OSX_HOMEBREW_PATH}/opt/wolfssl)

add_library(wolfssl UNKNOWN IMPORTED)

set_target_properties(
        wolfssl
        PROPERTIES
        IMPORTED_LOCATION "${wolfSSL_DIR}/lib/libwolfssl.dylib"
        INTERFACE_INCLUDE_DIRECTORIES "${wolfSSL_DIR}/include"
)

add_definitions(-DWC_NO_HARDEN -DLARGE_STATIC_BUFFERS)
```

### Reproduction steps

I followed the example from: https://www.wolfssl.com/documentation/manuals/wolfssl/chapter10.html#hmac

```c++
byte key[24];        /*fill key with keying material*/
  Hmac hmac;
  wc_HmacSetKey(&hmac, SHA256, key, sizeof(key));

  wc_HmacUpdate(&hmac, reinterpret_cast<const byte *>(input.data()), input.size());
  wc_HmacFinal(&hmac, reinterpret_cast<byte *>(buffer_));

  return {buffer_, SHA256_DIGEST_SIZE};
```

`wc_HmacSetKey` corrupts program memory causing segfaults

Ziemien, 2022-08-10:
I traded this on the Amazon Linux 2, Clang 14.0.6 with libc++

```
<?xml version="1.0" encoding="UTF-8"?>
<Catch name="test" filters="emotional damage">
  <Group name="test">
    <TestCase name="emotional damage" tags="[wolfssl]" filename="file.cxx" line="6">
      <FatalErrorCondition filename="file.cxx" line="6">
        SIGSEGV - Segmentation violation signal
      </FatalErrorCondition>
      <OverallResult success="false" durationInSeconds="8.5e-05"/>
    </TestCase>
    <OverallResults successes="0" failures="1" expectedFailures="0"/>
    <OverallResultsCases successes="0" failures="1" expectedFailures="0"/>
  </Group>
  <OverallResults successes="0" failures="1" expectedFailures="0"/>
  <OverallResultsCases successes="0" failures="1" expectedFailures="0"/>
</Catch>Segmentation fault
```

dgarske, 2022-08-10:
Hi @Ziemien ,

The reason for the crash is the stack defined for Hmac structure does not match the size that wolfSSL expects.

It sounds like your application is not including the `wolfssl/options.h` file. This is a generated file containing the build settings used at compile-time. It is important your application also includes / defines the same build settings.

See the FAQ item 1 here:
https://www.wolfssl.com/docs/frequently-asked-questions-faq/#How_do_I_manage_the_build_configuration_for_wolfSSL?

Thanks,
David Garske, wolfSSL

Ziemien, 2022-08-10:
thank you, is `--enable-fips` require to access SHA256?

dgarske, 2022-08-10:
Hi @Ziemien ,

The SHA256 is on by default unless --disable-sha256 or `NO_SHA256` is used.

The issue you are seeing is a data type mismatch trying to call from C++ to C. Specifically the `SHA256` to `int`.

Can you try using `WC_SHA256` and casting to `(int)`? This value comes from an enum, so it is not an `int`. See sha256.h -> `WC_SHA256              =  WC_HASH_TYPE_SHA256,`.

Thanks,
David Garske, wolfSSL

Ziemien, 2022-08-10:
`WC_SHA256` works fine (at compile time) but not the `SHA256` is suggested in the documentation

dgarske, 2022-08-10:
Hi @Ziemien ,

The `SHA256 ` was the old / original name. The `SHA256` is a macro that maps to `WC_SHA256`. For some older FIPS releases I think it still uses `SHA256`. Documentation has been updated here:
https://github.com/wolfSSL/wolfssl/pull/5454

Let me know if you have any additional feedback. Feel free to close this issue if there is nothing further.

Thanks,
David Garske, wolfSSL

Ziemien, 2022-08-12:
thank you David

closed 2022-08-12 by Ziemien