wolfSSL/wolfssl #5462: Does not build cleanly on Apple M1

res0nance, 2022-08-12:
### Version

5.4.0

### Description

Run the following on apple M1

```
./autogen.sh
./configure
make
```

The make command will fail with the following error
```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -j9  all-recursive
make[2]: warning: -jN forced in submake: disabling jobserver mode.
make[2]: ./build-aux/install-sh: Permission denied
make[2]: ./build-aux/install-sh: Permission denied
make[2]: *** [wolfcrypt/src/.dirstamp] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [wolfcrypt/src/.deps/.dirstamp] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

embhorn, 2022-08-12:
Hello @res0nance 

We tried unsuccessfully to reproduce this issue. Since it is a permission problem, there may have been an accidental sudo in front of the autogen or configure command. This would require a clean up of the permissions in the generated files.

It would be easier to just start from scratch by first deleting the existing folder (may need sudo since the permissions were changed).

Then grab a fresh clone and try the install again (without sudo).

res0nance, 2022-08-12:
> Hello @res0nance
>
> We tried unsuccessfully to reproduce this issue. Since it is a permission problem, there may have been an accidental sudo in front of the autogen or configure command. This would [...]

The issue is not because I used sudo.
With a fresh clone, without sudo and it doesn't work with sudo it works. This has been tested across 3 m1 Macs
Note I've only been able to reproduce this with m1 Macs and no intel mac seems to be able to reproduce this.

embhorn, 2022-08-12:
> with sudo it works

So if you build with `sudo make`, it does work? This would seem to indicate an issue outside the wolfSSL build system. Do you have the same issue building other projects that use autotools?

I found a couple of posts that sounds relevant:
https://github.com/caskformula/homebrew-caskformula/issues/6
https://trac.macports.org/ticket/24431

res0nance, 2022-08-15:
Figured this out, the hint of the issue not being related to wolfSSL made me look into autoconf and make.

To all future readers, if you have an m1 mac, the issue seems to be with the default make.

Steps I used to fix the issue
- Install latest make via homebrew `brew install make`
- Use `gmake` to build the repo instead of the default `make`

closed 2022-08-15 by res0nance