wolfSSL/wolfssl #5463: [Bug]:

icing, 2022-08-12:
### Contact Details

stefan.eissing@gmail.com

### Version

5.4.0-dev (master)

### Description

On a `master` checkout of wolfSSL, built with `--enable-all`, the TLSv1.3 resumption and early data handling runs into interop problems with a OpenSSL client. The client used was:

```
OpenSSL 3.0.3+quic 3 May 2022 (Library: OpenSSL 3.0.3+quic 3 May 2022)
```

The client reports errors like

```
0036CF0B01000000:error:0A000119:SSL routines:ssl3_get_record:decryption failed or bad record mac:ssl/record/ssl3_record.c:614:
```

and aborts the connection when re-using a sesion ticket against a restarted wolfssl example server. Resumption against a wolfSSL server that was kept running works.

The issue was found by @tatsuhiro-t when testing the integration of wolfSSL into ngtcp2. After testing with the "plain" wolfSSL server, he discovered that the problem applies to TLSv1.3 connection as well.

### Reproduction steps

The issue mentioned above happens with regular TLSv1.3 connection over TCP. Steps to repro:

* Run wolfssl examples server: examples/server/server -0 -v 4 -d -C10000
* Run openssl client: openssl s_client -connect localhost:11111 -sess_out session.txt -early_data data.txt
* Stop openssl client (ctrl-C)
* Stop wolfssl examples server
* Start wolfssl server again with the same options.
* Run openssl client: openssl s_client -connect localhost:11111 -sess_out session.txt -sess_in session.txt -early_data data.txt
* Now I get 140059142509952:error:1408F119:SSL routines:ssl3_get_record:decryption failed or bad record mac:../ssl/record/ssl3_record.c:676:

* After some time, do step 5 and 6 (needs step 5 because wolfssl example server stops as a result of step 6) and I get 140609374721408:error:1421A06E:SSL routines:tls_parse_stoc_early_data:bad extension:../ssl/statem/extensions_clnt.c:1946:

### Relevant log output

```shell
> /opt/quic/bin/openssl s_client -connect localhost:11111 -sess_out session.txt -sess_in session.txt -early_data data.txt
CONNECTED(00000003)
0006580801000000:error:0A000119:SSL routines:ssl3_get_record:decryption failed or bad record mac:ssl/record/ssl3_record.c:614:
---
no peer certificate available
---
No client certificate CA names sent
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 159 bytes and written 594 bytes
Verification error: self-signed certificate in certificate chain
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was rejected
Verify return code: 19 (self-signed certificate in certificate chain)
---

> ./examples/server/server -0 -v 4 -d -C10000
Early Data was accepted
SSL_accept error -397, Peer closed underlying transport Error
wolfSSL error: SSL_accept failed
```

embhorn, 2022-08-12:
Hi @icing 

Thanks for the report! We are reviewing.

dgarske, 2022-08-16:
Any updates on this @SparkiDev or @embhorn . @icing did any of your recent PR's resolve this?

SparkiDev, 2022-08-16:
Hi Stefan,

Session tickets are encrypted on the server side and sent to the client.
The client sends them back and the server must have the key to decrypt.
The wolfSSL server creates a new key at startup.
This is due to the way the session ticket encryption callback works. The keys are only persisted in memory.

If this is a required use case then it can be supported by modify the default session ticket callback to persist the keys to disk.
Alternatively, test.h has MyTicketEncCb that persists the key to memory as well which can be enhanced to persist to disk.

Sean

icing, 2022-08-17:
Hi Sean,

the resumption on an old session is not the problem. That works fine between the wolfSSL example server and an openssl client. It is only when the `-early_data data.txt` is added to the client that the handshake fails with openssl complaining about `decryption failed or bad record mac` when parsing the server reply.

tl;dr

* no session: connect+early data works
* prev session, same server: connect+early data works
* prev session, new server: connect works
* prev session, new server: connect+early data fails

icing, 2022-08-17:
Looking at `./scripts/openssl.test` where this could be reproduced, but a bit discouraged to work on that code. How about a `pytest` example of how such test cases could work?

I don't want to start a technology battle here, however my experiences with that test framework are superb. Just throwing it out here...(and one could get into python version issues on the various platforms you run, so it might not be a good idea.)

icing, 2022-08-17:
Further digging revealed: on an unsuccessful resumption by the client, the wolfSSL server still generates a EARLY_DATA Indicator in the EncryptedExtensions and this makes an OpenSSL client error and drop the connection.

I tried various places where to suppress sending that extension. But was not successful. It gets added in `TLSX_EarlyData_Parse()`, but there it is not reliable know if a resumption is happening and if it will not be rejected. I also tried adding a removal in `CheckPreSharedKeys()`(I think that is the place resumption is actually decided on), but it had no effect as well.

If someone (@SparkiDev ?) can give me some guidance, I can make a PR for this. Thanks!

SparkiDev, 2022-08-19:
Hi @icing ,

I put up a pull request that should fix the issue: https://github.com/wolfSSL/wolfssl/issues/5463
The response flag was being set always.
The correct check is being done and response flag set but it was already on.

Let me know if there are still issues.

Sean

icing, 2022-08-19:
Hi @SparkiDev ,

thanks for the proposed fix. That one had some errors in testing. I improved on your PR somewhat in https://github.com/wolfSSL/wolfssl/pull/5487 and that passes.

On a failed resume attempt, a client no longer see EDI in the server's EncryptedExtensions.

SparkiDev, 2022-08-21:
Great! I've closed my pull request.
At least I gave you somewhere to look.

icing, 2022-08-22:
I can confirm that 26c61f8e0fb1bc3aad5e3759de23fa0d8b40baa5 fixed the issue.

dgarske, 2022-08-22:
Thanks @icing. Marking this closed.

closed 2022-08-22 by dgarske