wolfSSL/wolfssl #5466: Linker error LNK2001 - wolfSSLv3_server_method

LionHackerman, 2022-08-13:
### Version

Latest

### Description

OS: Windows.
IDE: VS2022 (C++)
I've linked the project and successfully built wolfssl which includes ssl.c. I've also linked the .lib in Additional Library Directories.
I'm trying to compile my application but I keep getting a linker error:
Code:
wolfSSL_CTX_new(wolfSSLv3_server_method()); 
...
Error: unresolved external symbol wolfSSLv3_server_method

embhorn, 2022-08-15:
Hi @LionHackerman 

SSLv3 is disabled by default. You can enable it by adding `#define WOLFSSL_ALLOW_SSLV3` to the config header:
`https://github.com/wolfSSL/wolfssl/blob/master/IDE/WIN/user_settings.h`

Also make sure `NO_OLD_TLS` is not defined.

Thanks,
Eric @ wolfSSL Support

LionHackerman, 2022-08-16:
Thanks, worked perfectly. +rep

closed 2022-08-16 by LionHackerman