wolfSSL/wolfssl #5479: Doubt on using FALCON/Dilithium for TLS in the STM32f board

samyuktha123sets, 2022-08-18:
### Version

wolfssl/wolfssl-examples/pq

### Description

How do i use Falcon and Kyber in wolfssl for authentication and key exchange in STM32 board. The default example given is p251-kyber combination. But I want to execute falcon-kyber combination in board. Kindly help at the earliest
[image]

anhu, 2022-08-18:
Hello @samyuktha123sets , 

Thank you for your inquiry and interest in post-quantum cryptography on STM32 using wolfSSL.  

Please see our announcement here: https://www.wolfssl.com/post-quantum-tls-1-3-key-establishment-comes-stm32-cortex-m4/

Please note that Falcon is a signature scheme that we have not yet ported to STM32. You mention p251-kyber. Perhaps you are thinking of p256_kyber_level1  which is a hybrid of ECDHE over curve P-256 and Kyber at NIST Level 1. We have also not ported the hybrid to STM32 yet. 

Currently, only non-hybridized Kyber NIST Level 1 is supported on STM32.  If you would like, you can send a message to facts@wolfssl.com and support@wolfssl.com requesting a feature to add hybrid kyber and/or an authentication scheme. I would think that Dilithium would be more appropriate for embedded systems. 

Can you please let me know some details about your project?  

Warm regards, Anthony

samyuktha123sets, 2022-08-19:
Hello,

Thank you for the prompt response. Our work is to set up a digital certificate generation and verification in STM32 Board. We thought of using wolfssl for the same. For the certificate generation, do you have any of the PQ Digital Signature scheme integrated with wolfssl.

Thanks and Regards,
Samyuktha

anhu, 2022-08-19:
Conversation has moved to  ZD 14692 .

closed 2022-08-19 by anhu