wolfSSL/wolfssl #5546: [Bug]: Release artifacts for 5.5.0 missing PGP signatures

dvzrv, 2022-09-02:
### Version

5.5.0

### Description

Hi! I'm maintaining this project as a package for Arch Linux.
For packaging we use the source tarballs attached as artifacts to the releases on github *and* verify them using the attached PGP signature to mitigate supply chain attacks.

For 5.5.0 the PGP signatures are missing, which prevents me from updating to that version.

### Reproduction steps

1. Download https://github.com/wolfSSL/wolfssl/releases/download/v5.4.0-stable/wolfssl-5.4.0-stable.tar.gz.asc and https://github.com/wolfSSL/wolfssl/archive/refs/tags/v5.4.0-stable.tar.gz for 5.4.0
2. Attempt to download https://github.com/wolfSSL/wolfssl/releases/download/v5.5.0-stable/wolfssl-5.5.0-stable.tar.gz.asc

dgarske, 2022-09-02:
@JacobBarthelmeh did we me the release step to sign the bundle?

JacobBarthelmeh, 2022-09-02:
Updated signatures for the release bundles, thanks for the report!

closed 2022-09-02 by JacobBarthelmeh