wolfSSL/wolfssl #5553: [Bug]: 5.5.0 wolfSSL_writev - error: ‘staticBuffer’ may be used uninitialized [-Werror=maybe-uninitialized]

dvzrv, 2022-09-05:
### Contact Details

dvzrv@archlinux.org

### Version

5.5.0

### Description

Hi! I'm packaging this project for Arch Linux.
When trying to build 5.5.0 I ran into issues during build due to `wolfSSL_writev` (#4048 looks related).

We are currently using gcc 12.2.0 and glibc 2.36.

Full build log: 
[wolfssl-5.5.0-1-x86_64-build.log](https://github.com/wolfSSL/wolfssl/files/9490732/wolfssl-5.5.0-1-x86_64-build.log)

### Reproduction steps

```
  cmake \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_BUILD_TYPE=None \
    -DWOLFSSL_CURVE25519=ON \
    -DWOLFSSL_CURVE448=ON \
    -DWOLFSSL_ED25519=ON \
    -DWOLFSSL_ED448=ON \
    -DWOLFSSL_REPRODUCIBLE_BUILD=ON \
    -Wno-dev \
    -B build \
    -S $pkgname-$pkgver-stable
  cmake --build build --verbose
```

### Relevant log output

```shell
[ 66%] Linking C shared library libwolfssl.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/wolfssl.dir/link.txt --verbose=1
/usr/bin/cc -fPIC -Wall -Wextra -Wno-unused -Werror -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/wolfssl/src=/usr/src/debug -flto=auto -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -flto=auto -shared -Wl,-soname,libwolfssl.so.35 -o libwolfssl.so.35.0.0 CMakeFiles/wolfssl.dir/wolfcrypt/src/hmac.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/hash.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/cpuid.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/kdf.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/random.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha256.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/rsa.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sp_int.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/aes.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha512.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha3.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/logging.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/wc_port.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/error.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/wc_encrypt.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/signature.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/wolfmath.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/memory.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/dh.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/asn.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/coding.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/poly1305.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/md5.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/pwdbased.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/pkcs12.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/chacha.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/chacha20_poly1305.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/integer.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ecc.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/curve25519.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ed25519.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/fe_operations.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ge_operations.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/curve448.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ed448.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/fe_448.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ge_448.c.o CMakeFiles/wolfssl.dir/src/internal.c.o CMakeFiles/wolfssl.dir/src/wolfio.c.o CMakeFiles/wolfssl.dir/src/keys.c.o CMakeFiles/wolfssl.dir/src/ssl.c.o CMakeFiles/wolfssl.dir/src/tls.c.o CMakeFiles/wolfssl.dir/src/tls13.c.o  -lm
/usr/src/debug/wolfssl-5.5.0-stable/src/ssl.c: In function ‘wolfSSL_writev’:
/usr/src/debug/wolfssl-5.5.0-stable/src/ssl.c:14702:19: error: ‘staticBuffer’ may be used uninitialized [-Werror=maybe-uninitialized]
/usr/src/debug/wolfssl-5.5.0-stable/src/ssl.c:2369:5: note: by argument 2 of type ‘const void *’ to ‘wolfSSL_write’ declared here
/usr/src/debug/wolfssl-5.5.0-stable/src/ssl.c:14669:20: note: ‘staticBuffer’ declared here
lto1: all warnings being treated as errors
make[3]: *** [/tmp/ccVXov7g.mk:23: /tmp/ccgVYYSj.ltrans7.ltrans.o] Error 1
make[3]: *** Waiting for unfinished jobs....
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:785: libwolfssl.so.35.0.0] Error 1
make[2]: Leaving directory '/build/wolfssl/src/build'
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/wolfssl.dir/all] Error 2
make[1]: Leaving directory '/build/wolfssl/src/build'
make: *** [Makefile:136: all] Error 2
```

embhorn, 2022-09-06:
@douzzer 
I see you updated a pragma to try and address this:
https://github.com/wolfSSL/wolfssl/pull/4359
https://github.com/wolfSSL/wolfssl/commit/4cf1826c8f8ae615ff638d4c70f69a6c615ee49a#diff-34d62efe7620eac9235c878aafc6716d1ef20a2246a35fac977f34bacce0481c

@dvzrv 
Would this be fixed by simply initializing with a value?
```
@@ -14679,6 +14679,8 @@ int wolfSSL_set_compression(WOLFSSL* ssl)
 
             WOLFSSL_ENTER("wolfSSL_writev");
 
+            staticBuffer[1] = 0;
+
             for (i = 0; i < iovcnt; i++)
                 sending += (int)iov[i].iov_len;
```

embhorn, 2022-09-06:
@Uriah-wolfSSL will reproduce and investigate a fix.

Uriah-wolfSSL, 2022-09-08:
Hi @dvzrv

Have you tried the previous suggestion by @douzzer ?

To note: I have been unable to reproduce this problem. I have tried with gcc 11 and gcc 12 and both compile cleanly and unit tests pass. This is using wolfssl-5.5.0-stable on Ubuntu 22.04 LTS. I have verified that the section of code in question is getting compiled. I have also double checked the command line options shown in the provided log file, though I could have missed something of course.

Here is an example of the command line I am using. It is pretty detailed. Does anything jump out to you as being different from what you are attempting?

`--tag=CC   --mode=compile gcc-12 -DHAVE_CONFIG_H -I.  -DBUILDING_WOLFSSL   -DBUILDING_WOLFSSL -fvisibility=hidden -DHAVE_THREAD_LS -DNDEBUG -DNO_DO178 -D_POSIX_THREADS -pthread -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DWOLFSSL_USE_ALIGN -DWOLFSSL_SHA224 -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DHAVE_HKDF -DNO_DSA -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DHAVE_CURVE25519 -DHAVE_CURVE448 -DHAVE_ED448 -DWC_RSA_PSS -DWOLFSSL_PSS_LONG_SALT -DWOLFSSL_BASE64_ENCODE -DWOLFSSL_SHA3 -DWOLFSSL_NO_SHAKE128 -DWOLFSSL_SHAKE256 -DHAVE_POLY1305 -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_FFDHE_2048 -DHAVE_SUPPORTED_CURVES -DWOLFSSL_TLS13 -DHAVE_TLS_EXTENSIONS -DHAVE_EXTENDED_MASTER -DWOLFSSL_SMALL_STACK -DNO_RC4 -DHAVE_ENCRYPT_THEN_MAC -DNO_PSK -DNO_MD4 -DWOLFSSL_SP_MATH_ALL -DWOLFSSL_SP_X86_64 -DWOLFSSL_X86_64_BUILD -DWC_NO_ASYNC_THREADING -DHAVE_DH_DEFAULT_PARAMS -DHAVE_ED25519 -DNO_DES3 -DGCM_TABLE_4BIT -DHAVE_AESGCM -DHAVE_TLS_EXTENSIONS -DHAVE_SERVER_RENEGOTIATION_INFO -Wall -Wno-unused -O2 -DHAVE___UINT128_T=1 -DHAVE_WC_INTROSPECTION -include ./.build_params  -DLIBWOLFSSL_GLOBAL_EXTRA_CFLAGS="\" \"" -DBUILDING_WOLFSSL -DECC_SHAMIR -DECC_TIMING_RESISTANT -DGCM_TABLE_4BIT -DHAVE_AESGCM -DHAVE_CHACHA -DHAVE_CONFIG_H -DHAVE_CURVE25519 -DHAVE_CURVE448 -DHAVE_DH_DEFAULT_PARAMS -DHAVE_ECC -DHAVE_ED25519 -DHAVE_ED448 -DHAVE_ENCRYPT_THEN_MAC -DHAVE_EXTENDED_MASTER -DHAVE_FFDHE_2048 -DHAVE_HASHDRBG -DHAVE_HKDF -DHAVE_ONE_TIME_AUTH -DHAVE_POLY1305 -DHAVE_PTHREAD -DHAVE_REPRODUCIBLE_BUILD -DHAVE_SUPPORTED_CURVES -DHAVE_THREAD_LS -DHAVE_TLS_EXTENSIONS -DNO_DES3 -DNO_DSA -DNO_MD4 -DNO_PSK -DNO_RC4 -DTFM_ECC256 -DTFM_TIMING_RESISTANT -DWC_NO_ASYNC_THREADING -DWC_RSA_BLINDING -DWC_RSA_PSS -DWOLFSSL_BASE64_ENCODE -DWOLFSSL_DLL -DWOLFSSL_IGNORE_FILE_WARN -DWOLFSSL_PSS_LONG_SALT -DWOLFSSL_SHA224 -DWOLFSSL_SHA3 -DWOLFSSL_SHA384 -DWOLFSSL_SHA512 -DWOLFSSL_SHAKE256 -DWOLFSSL_TLS13 -DWOLFSSL_USE_ALIGN -D_POSIX_THREADS -Dwolfssl_EXPORTS -I./ -I./wolfssl/wolfcrypt -Wall -Wextra -Wno-unused -Werror -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -Werror=maybe-uninitialized -fstack-clash-protection -fcf-protection -fPIC -Wall -Wextra -Wno-unused -Werror -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions   -Wno-pragmas -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv -MT src/libwolfssl_la-ssl.lo -MD -MP -MF src/.deps/libwolfssl_la-ssl.Tpo -c -o src/libwolfssl_la-ssl.lo `test -f 'src/ssl.c' || echo './'`src/ssl.c`

dvzrv, 2022-09-08:
> @dvzrv Would this be fixed by simply initializing with a value?
>
> ```
> @@ -14679,6 +14679,8 @@ int wolfSSL_set_compression(WOLFSSL* ssl)
>
>              WOLFSSL_ENTER("wolfSSL_writev");
>
> + [...]

No, this leads to the next issue then:

```
[ 66%] Linking C shared library libwolfssl.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/wolfssl.dir/link.txt --verbose=1
/usr/bin/cc -fPIC -Wall -Wextra -Wno-unused -Werror -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/wolfssl/src=/usr/src/debug -flto=auto -O3 -DNDEBUG -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -flto=auto -shared -Wl,-soname,libwolfssl.so.35 -o libwolfssl.so.35.0.0 CMakeFiles/wolfssl.dir/wolfcrypt/src/hmac.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/hash.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/cpuid.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/kdf.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/random.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha256.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/rsa.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sp_int.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/aes.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha512.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/sha3.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/logging.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/wc_port.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/error.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/wc_encrypt.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/signature.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/wolfmath.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/memory.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/dh.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/asn.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/coding.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/poly1305.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/md5.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/pwdbased.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/pkcs12.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/chacha.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/chacha20_poly1305.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/integer.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ecc.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/curve25519.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ed25519.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/fe_operations.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ge_operations.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/curve448.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ed448.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/fe_448.c.o CMakeFiles/wolfssl.dir/wolfcrypt/src/ge_448.c.o CMakeFiles/wolfssl.dir/src/internal.c.o CMakeFiles/wolfssl.dir/src/wolfio.c.o CMakeFiles/wolfssl.dir/src/keys.c.o CMakeFiles/wolfssl.dir/src/ssl.c.o CMakeFiles/wolfssl.dir/src/tls.c.o CMakeFiles/wolfssl.dir/src/tls13.c.o  -lm
In function ‘Hmac_UpdateFinal_CT’,
    inlined from ‘TLS_hmac’ at /usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1197:23:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:995:50: error: writing 16 bytes into a region of size 0 [-Werror=stringop-overflow=]
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c: In function ‘TLS_hmac’:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1133:12: note: at offset 784 into destination object ‘hmac’ of size 784
In function ‘Hmac_UpdateFinal_CT’,
    inlined from ‘TLS_hmac’ at /usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1197:23:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:995:50: error: writing 16 bytes into a region of size 0 [-Werror=stringop-overflow=]
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c: In function ‘TLS_hmac’:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1133:12: note: at offset 800 into destination object ‘hmac’ of size 784
In function ‘Hmac_UpdateFinal_CT’,
    inlined from ‘TLS_hmac’ at /usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1197:23:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:995:50: error: writing 16 bytes into a region of size 0 [-Werror=stringop-overflow=]
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c: In function ‘TLS_hmac’:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1133:12: note: at offset 816 into destination object ‘hmac’ of size 784
In function ‘Hmac_UpdateFinal_CT’,
    inlined from ‘TLS_hmac’ at /usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1197:23:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:995:50: error: writing 16 bytes into a region of size 0 [-Werror=stringop-overflow=]
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c: In function ‘TLS_hmac’:
/usr/src/debug/wolfssl-5.5.0-stable/src/tls.c:1133:12: note: at offset 832 into destination object ‘hmac’ of size 784
lto1: all warnings being treated as errors
make[3]: *** [/tmp/cctGCpy7.mk:26: /tmp/cceCLJ2Y.ltrans8.ltrans.o] Error 1
make[3]: *** Waiting for unfinished jobs....
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:785: libwolfssl.so.35.0.0] Error 1
make[2]: Leaving directory '/build/wolfssl/src/build'
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/wolfssl.dir/all] Error 2
make[1]: Leaving directory '/build/wolfssl/src/build'
make: *** [Makefile:136: all] Error 2
```

What sticks out there is `lto-wrapper: fatal error: make returned 2 exit status`. Is link time optimization not supported? Even if I disable our now distribution wide default (LTO is used by default), it still leads to that error message.

dvzrv, 2022-09-08:
@Uriah-wolfSSL Hm, not sure what the difference there is supposed to be.

FWIW: I can still build 5.4.0 just fine, so there definitely is a regression somewhere.

dvzrv, 2022-09-08:
f

dvzrv, 2022-09-08:
Ah... okay, I guess I have found the issue. As soon as `-flto=auto` is used, it breaks the build with 5.5.0 (but this still works with 5.4.0)!

dvzrv, 2022-09-08:
I have bisected this and the first broken commit is e38290531190dbc3f9683766eebed2f0dd4aca0c.

It adds `-Wall -Wextra -Wno-unused -Werror` to CFLAGS, which is the cause of this issue.
Something like `-Werror` should really not be included by default as it *will* break as soon as someone uses a newer compiler version than the one you have tested with! After removing it I can compile successfully.

Uriah-wolfSSL, 2022-09-22:
Hi @dvzrv,

I need to inform you that wolfSSL does not support Link Time Optimization. It negatively affects some of the algorithms used within wolfSSL. I suggest not using -flto=auto.

As far as the -Werror comand line option, this will not be removed. This is a requirement of our process.

I'll take another look at reproducing this on my side.

dvzrv, 2022-09-22:
@dgarske Thanks for the in-depth feedback. I have made the package work with LTO by patching out the use of `-Werror` (which represents the underlying issue, see https://github.com/wolfSSL/wolfssl/issues/5553#issuecomment-1241253523).

@Uriah-wolfSSL I understand that this is useful in the context of your CI to ensure code quality etc., but it is prone to break for everyone else not using the exact same compiler as you do. This is why flags such as `-Werror` are usually not applied by default (i.e. for everyone trying to compile the project) but are used specifically in CI to have one or more pipelines that test whether the build would fail on any warnings.

My suggestion would therefore be to only apply the strict compiler flags, that you deem necessary for your CI, in the setup of your CI, but not in the build system itself, as the current situation will lead to a lot of opened issues by users with differing build environments, chasing red herrings and requiring all of them to eventually manually modify the build system to remove the use of `-Werror`.

If you are absolutely sure, that you need to have `-Werror` included in the build system, please offer an option to toggle it.

Uriah-wolfSSL, 2022-09-22:
@dvzrv I understand your point and having `-Werror` as a toggle option sounds like a good compromise. We will discuss this internally make a determination if such an option will be added in a future release.

dgarske, 2022-12-08:
Hi @dvzrv ,

It looks like we provided a solution to this in CMake by adding a new `WARNING_C_FLAGS`.

See PR https://github.com/wolfSSL/wolfssl/pull/5683(specifically commit https://github.com/wolfSSL/wolfssl/pull/5683/commits/e48e406def79de91d13199b923091e333368bbb3). 

CMakeLists.txt: add handling for WARNING_C_FLAGS in setup for CMAKE_C_FLAGS, to allow override warning config, e.g. without -Werror.

Thanks,
David Garske, wolfSSL

dvzrv, 2022-12-31:
Thanks! Using `-DWARNING_C_FLAGS="$CFLAGS"` fixes this for me.

closed 2022-12-31 by dvzrv