wolfSSL/wolfssl #5629: [Bug]: DoTls13HandShakeMsgType stack smashing in the wolfSSL 5.5.0 powered web server on aarch64

ynezz, 2022-09-24:
### Contact Details

ynezz@true.cz

### Version

5.4.0 and 5.5.0

### Description

Several OpenWrt [users have reported](https://github.com/openwrt/luci/issues/5962), that wolfSSL powered HTTPS server crashes when accessed from Chrome 105 web browser. Issue happens with wolfSSL versions 5.4.0 and 5.5.0.

### Reproduction steps

1. run latest OpenWrt 22.03.0 release (wolfSSL 5.4.0) or latest development snapshot (wolfSSL 5.5.0) on aarch64 target
2. use Chrome 105 web browser to access web user interface served over HTTPS 

### Relevant log output

```shell
Core was generated by `/usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -l /cgi-bin/luci -L /usr/lib'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000ffff8f0c72a0 in a_crash () at ./src/internal/atomic.h:250
250     ./src/internal/atomic.h: No such file or directory.
(gdb) bt
#0  0x0000ffff8f0c72a0 in a_crash () at ./src/internal/atomic.h:250
#1  __stack_chk_fail () at src/env/__stack_chk_fail.c:26
#2  0x0000ffff8ee76744 in DoTls13HandShakeMsgType (ssl=0xffff8ebec170, input=0xffff8ebd46e0 "\001", inOutIdx=<optimized out>, type=<optimized out>, size=<optimized out>, totalSz=<optimized out>) at src/tls13.c:9783
#3  0x0000ffff8ee1ce18 in DoTls13HandShakeMsg (totalSz=529, inOutIdx=0xffff8ebec414, input=0xffff8ebd46e0 "\001", ssl=0xffff8ebec170) at src/tls13.c:10074
#4  ProcessReplyEx (allowSocketErr=0, ssl=0xffff8ebec170) at src/internal.c:19337
#5  ProcessReply (ssl=0xffff8ebec170) at src/internal.c:18593
#6  0x0213021302130213 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
```

dgarske, 2022-09-25:
Hi @ynezz,

Thank you for this report. Have you tried with latest master? I've asked @rizlik to review this report. Please pull in @SparkiDev if needed.

Thanks, David Garske, wolfSSL

ynezz, 2022-09-26:
> Have you tried with latest master?

Unfortunately I'm not able to reproduce it myself, so [I've just asked](https://github.com/openwrt/luci/issues/5962#issuecomment-1257754365) one of the affected users @kien-truong to help with that check.

rizlik, 2022-09-26:
@ynezz 
I also tried to reproduce with armvirt-64 and qemu without luck. The only similar target I've at hand is a raspberryPi 4 and I'll try to reproduce there.

kien-truong, 2022-09-26:
@ynezz  I tried WolfSSL master from your patch, and Luci hasn't crashed so far.  I also notice 1 thing odd about my Chrome TLS handshake: it has a duplicate in its Cipher Suite

```
Cipher Suite: TLS_AES_256_GCM_SHA384 (0x1302)
Cipher Suite: TLS_AES_256_GCM_SHA384 (0x1302)
```

There's a new commit in WolfSSL that seems related to this https://github.com/wolfSSL/wolfssl/commit/79d85f6c136c63b66ab5acc8015c5d6e9c7c9c48

ynezz, 2022-09-26:
> and Luci hasn't crashed so far

Thanks for such a quick response, thats a good news!

> There's a new commit in WolfSSL that seems related to this https://github.com/wolfSSL/wolfssl/commit/79d85f6c136c63b66ab5acc8015c5d6e9c7c9c48

And probably 5e945f94b4033806f0d8435c1b4554dcac444663 ?

kien-truong, 2022-09-26:
It does seem possible to write pass the end of the `suites` array in case of duplicates, before https://github.com/wolfSSL/wolfssl/commit/5e945f94b4033806f0d8435c1b4554dcac444663 ,  thus corrupting the stack.

ynezz, 2022-09-27:
@dgarske BTW are you going to request CVE for this security issue having RCE potential?

ynezz, 2022-09-29:
Fixed in 5.5.1.

closed 2022-09-29 by ynezz

ynezz, 2022-09-29:
> @dgarske BTW are you going to request CVE# for this security issue having RCE potential?

CVE-2022-39173