wolfSSL/wolfssl #5659: [Bug]: TLS connection fails with error -188 when compiled with --disable-asm, but not otherwise

jawj, 2022-10-04:
### Contact Details

george@mackerron.co.uk

### Version

5.5.1

### Description

I've put together a simple test program that requests a single web page over TLS 1.2/1.3 and HTTP 1.0.

When I compile with these settings, it works fine:

```
./configure \
  --disable-filesystem --disable-examples --disable-oldtls \
  --enable-sni --enable-tls13 --enable-altcertchains

make
make install
```

But when I add `--disable-asm` to this, the same request fails with error -188. The last bytes sent are:

```
15 03 03 00 02 02 2a
15 03 03 00 02 02 30
```

I am using `--disable-asm` because I need to compile with emscripten, but I get the same result when I compile with these settings to native code on Mac.

### Reproduction steps

```
./configure \
  --disable-filesystem --disable-examples --disable-oldtls \
  --enable-sni --enable-tls13 --enable-altcertchains --disable-asm
```

(I can clean up the C file to share if necessary)

### Relevant log output

```shell
./tls-test neon.tech 443 / 
send: 16 03 03 01 07 01 00 01 03 03 03 68 35 73 1e b4 3f 66 56 17 69 96 4d 12 9b 0b 95 ff 47 97 83 b5 f0 da 44 9e 59 81 ad ef 0b fc 59 00 00 36 13 01 13 02 13 03 c0 2c c0 2b c0 30 c0 2f 00 9f 00 9e cc a9 cc a8 cc aa c0 27 c0 23 c0 28 c0 24 c0 0a c0 09 c0 14 c0 13 00 6b 00 67 00 39 00 33 cc 14 cc 13 cc 15 01 00 00 a4 00 33 00 47 00 45 00 17 00 41 04 66 20 ff 5f 99 b2 b8 50 7a 20 51 4d 1c 87 06 4c cd c0 e8 b8 b5 06 b0 5e 1e 14 4a 8a 12 1a 4d c0 87 aa 63 92 4b 6e 8f a5 e8 29 fb 88 3c e3 41 b5 c6 b9 3a 89 99 75 5e ce 2e c7 a5 22 77 1b 44 e8 00 2b 00 05 04 03 04 03 03 00 0d 00 1c 00 1a 06 03 05 03 04 03 08 06 08 0b 08 05 08 0a 08 04 08 09 06 01 05 01 04 01 03 01 00 0b 00 02 01 00 00 0a 00 0c 00 0a 00 19 00 18 00 17 00 15 01 00 00 16 00 00 00 00 00 0e 00 0c 00 00 09 6e 65 6f 6e 2e 74 65 63 68 00 17 00 00
sent 268 bytes to 3

recv: 16 03 03 00 5c
received 5 bytes from 3

recv: 02 00 00 58 03 03 88 a1 b5 6f 1d 7a a7 09 ba 3a 21 3a 45 e0 81 c8 ff 5d b1 ee 6c 41 de 0c b1 62 d4 d4 b3 3d a3 bb 20 04 63 0b 32 89 bb 6a 59 6e 07 d1 6c a1 25 c4 7c 0a b4 bc a6 32 ca e5 9e 82 cd 25 8f 88 0c 56 44 c0 2f 00 00 10 00 00 00 00 00 0b 00 04 03 00 01 02 00 17 00 00
received 92 bytes from 3

recv: 16 03 03 0f a9
received 5 bytes from 3

recv: 0b 00 0f a5 00 0f a2 00 05 1b 30 82 05 17 30 82 03 ff a0 03 02 01 02 02 12 03 5e e3 f5 53 68 c7 e3 c6 97 6a c1 1d f9 e7 01 22 10 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 32 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 16 30 14 06 03 55 04 0a 13 0d 4c 65 74 27 73 20 45 6e 63 72 79 70 74 31 0b 30 09 06 03 55 04 03 13 02 52 33 30 1e 17 0d 32 32 30 39 31 30 31 39 30 37 31 38 5a 17 0d 32 32 31 32 30 39 31 39 30 37 31 37 5a 30 14 31 12 30 10 06 03 55 04 03 13 09 6e 65 6f 6e 2e 74 65 63 68 30 82 01 22 30 0d 06 09 2a 86 48 86 f7 0d 01 01 01 05 00 03 82 01 0f 00 30 82 01 0a 02 82 01 01 00 f2 60 b1 c4 1e 40 0f 4e e5 31 2d 0c c0 90 00 92 c8 57 c8 d9 4b b1 2e 55 62 22 cb 04 8b 00 dd 54 49 93 05 79 1d 4e cb 95 cc 4b d5 bb da 9d f6 df 80 4d bb 3c a7 4a b0 5d aa 3c 7f 85 1d db 4b 05 57 76 2a 31 f0 c8 a1 a7 5e 70 cb 5b 45 99 98 94 05 ba 0b bf cb dc ff 4c d6 46 e8 e6 2f ee 6d c1 14 18 7d 63 d6 c0 a0 f6 ed 1b 17 d8 c9 17 4b ad 53 b7 2f 9b 24 e5 17 89 9c 2e 80 0a 2c 58 33 b2 79 62 ed 2f dc e5 66 06 99 f1 b0 b4 ad 50 61 08 dd f8 17 fa 33 32 56 58 8a 91 ea c2 bf b3 1c ac f9 e6 ef f5 a3 4f 2f ed cf 67 8d 17 dc 5c c6 5e ea e4 9f 04 76 18 92 e4 de 87 b3 17 a9 d9 90 b6 40 05 ee 0c d9 68 36 48 75 55 e1 73 5b d7 6e d1 bb 93 7c 6a 14 e6 7d e3 7a a1 37 8b a2 7b 9b 1e af 84 49 3a 69 bf 53 90 80 1c 0b 21 79 d4 eb 48 63 3f 7d e0 91 23 f9 21 19 11 0b e1 11 06 58 4f 02 03 01 00 01 a3 82 02 43 30 82 02 3f 30 0e 06 03 55 1d 0f 01 01 ff 04 04 03 02 05 a0 30 1d 06 03 55 1d 25 04 16 30 14 06 08 2b 06 01 05 05 07 03 01 06 08 2b 06 01 05 05 07 03 02 30 0c 06 03 55 1d 13 01 01 ff 04 02 30 00 30 1d 06 03 55 1d 0e 04 16 04 14 fd 68 25 e5 df 12 c4 61 a2 8a 8e 3e 1c 21 84 e7 f2 53 18 fe 30 1f 06 03 55 1d 23 04 18 30 16 80 14 14 2e b3 17 b7 58 56 cb ae 50 09 40 e6 1f af 9d 8b 14 c2 c6 30 55 06 08 2b 06 01 05 05 07 01 01 04 49 30 47 30 21 06 08 2b 06 01 05 05 07 30 01 86 15 68 74 74 70 3a 2f 2f 72 33 2e 6f 2e 6c 65 6e 63 72 2e 6f 72 67 30 22 06 08 2b 06 01 05 05 07 30 02 86 16 68 74 74 70 3a 2f 2f 72 33 2e 69 2e 6c 65 6e 63 72 2e 6f 72 67 2f 30 14 06 03 55 1d 11 04 0d 30 0b 82 09 6e 65 6f 6e 2e 74 65 63 68 30 4c 06 03 55 1d 20 04 45 30 43 30 08 06 06 67 81 0c 01 02 01 30 37 06 0b 2b 06 01 04 01 82 df 13 01 01 01 30 28 30 26 06 08 2b 06 01 05 05 07 02 01 16 1a 68 74 74 70 3a 2f 2f 63 70 73 2e 6c 65 74 73 65 6e 63 72 79 70 74 2e 6f 72 67 30 82 01 03 06 0a 2b 06 01 04 01 d6 79 02 04 02 04 81 f4 04 81 f1 00 ef 00 75 00 41 c8 ca b1 df 22 46 4a 10 c6 a1 3a 09 42 87 5e 4e 31 8b 1b 03 eb eb 4b c7 68 f0 90 62 96 06 f6 00 00 01 83 29 03 7f aa 00 00 04 03 00 46 30 44 02 20 50 a9 08 00 9f 68 0e e4 26 7c 22 82 9b 08 7f 66 b2 e6 4e 57 9e 22 da af 41 20 a1 93 72 d5 4c 5e 02 20 12 5b 15 52 cf e8 36 7d 21 a8 86 5f df 86 30 ff 56 46 a5 9e 24 4f 17 e0 12 28 27 df ad 21 c8 4a 00 76 00 29 79 be f0 9e 39 39 21 f0 56 73 9f 63 a5 77 e5 be 57 7d 9c 60 0a f8 f9 4d 5d 26 5c 25 5d c7 84 00 00 01 83 29 03 7f 95 00 00 04 03 00 47 30 45 02 21 00 84 91 b6 08 3b c3 d6 a2 2c 21 04 65 8f 00 eb 3d a2 58 5b c9 f5 2d e4 46 2e 90 e5 06 af 71 a3 7b 02 20 49 7f ac 27 26 8c 5e dc 79 17 5e ef d4 08 dc a9 d5 83 39 fc 93 02 57 02 49 13 6d a4 39 c8 b7 c0 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 03 82 01 01 00 9a 31 6e 70 d8 24 65 08 e7 91 f1 bc f5 46 2e 23 d6 61 d6 9e 8c c6 21 a0 0a 2f 25 66 35 69 b2 bf cc e8 a0 b0 11 a9 2a 26 7c a9 32 26 29 a6 0f e7 b5 af de aa 56 1e fc 3a dd 3e 43 12 b2 e2 51 80 df 20 15 8d 30 64 c0 d6 b5 74 4f 64 58 e0 56 ce ca 76 8e 86 6f 50 86 f1 37 95 15 8d 1a 96 50 86 0c 34 9d 67 3e 48 7d 9e b4 53 a3 11 a8 a3 e0 bd 3a db cc 4e 6a be 5b b1 8f d6 e5 95 9d 0b 0f 44 b3 80 09 a8 62 e5 84 07 76 15 38 e4 d0 df 44 55 38 ee 7d fc 3b 97 9c 31 94 2d 45 21 54 70 07 5f 0c 35 dd c1 e7 4f a7 65 ce cd a2 3f 05 a7 e0 2f 58 3f 13 31 e8 53 fd 11 cc 2f f2 77 47 c0 0c 1e c4 f8 08 e0 98 1e 10 2c 1d c8 a6 98 ba b1 d1 ba ba 9d 13 a4 dc 9f f0 08 07 e7 d0 4a dd d7 7c f0 8d 08 23 d6 94 b1 54 42 bc 6c e2 63 04 ef 9c e4 c9 df 82 31 cf 52 16 21 a2 7a 62 f4 69 d3 35 52 00 05 1a 30 82 05 16 30 82 02 fe a0 03 02 01 02 02 11 00 91 2b 08 4a cf 0c 18 a7 53 f6 d6 2e 25 a7 5f 5a 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 4f 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 29 30 27 06 03 55 04 0a 13 20 49 6e 74 65 72 6e 65 74 20 53 65 63 75 72 69 74 79 20 52 65 73 65 61 72 63 68 20 47 72 6f 75 70 31 15 30 13 06 03 55 04 03 13 0c 49 53 52 47 20 52 6f 6f 74 20 58 31 30 1e 17 0d 32 30 30 39 30 34 30 30 30 30 30 30 5a 17 0d 32 35 30 39 31 35 31 36 30 30 30 30 5a 30 32 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 16 30 14 06 03 55 04 0a 13 0d 4c 65 74 27 73 20 45 6e 63 72 79 70 74 31 0b 30 09 06 03 55 04 03 13 02 52 33 30 82 01 22 30 0d 06 09 2a 86 48 86 f7 0d 01 01 01 05 00 03 82 01 0f 00 30 82 01 0a 02 82 01 01 00 bb 02 15 28 cc f6 a0 94 d3 0f 12 ec 8d 55 92 c3 f8 82 f1 99 a6 7a 42 88 a7 5d 26 aa b5 2b b9 c5 4c b1 af 8e 6b f9 75 c8 a3 d7 0f 47 94 14 55 35 57 8c 9e a8 a2 39 19 f5 82 3c 42 a9 4e 6e f5 3b c3 2e db 8d c0 b0 5c f3 59 38 e7 ed cf 69 f0 5a 0b 1b be c0 94 24 25 87 fa 37 71 b3 13 e7 1c ac e1 9b ef db e4 3b 45 52 45 96 a9 c1 53 ce 34 c8 52 ee b5 ae ed 8f de 60 70 e2 a5 54 ab b6 6d 0e 97 a5 40 34 6b 2b d3 bc 66 eb 66 34 7c fa 6b 8b 8f 57 29 99 f8 30 17 5d ba 72 6f fb 81 c5 ad d2 86 58 3d 17 c7 e7 09 bb f1 2b f7 86 dc c1 da 71 5d d4 46 e3 cc ad 25 c1 88 bc 60 67 75 66 b3 f1 18 f7 a2 5c e6 53 ff 3a 88 b6 47 a5 ff 13 18 ea 98 09 77 3f 9d 53 f9 cf 01 e5 f5 a6 70 17 14 af 63 a4 ff 99 b3 93 9d dc 53 a7 06 fe 48 85 1d a1 69 ae 25 75 bb 13 cc 52 03 f5 ed 51 a1 8b db 15 02 03 01 00 01 a3 82 01 08 30 82 01 04 30 0e 06 03 55 1d 0f 01 01 ff 04 04 03 02 01 86 30 1d 06 03 55 1d 25 04 16 30 14 06 08 2b 06 01 05 05 07 03 02 06 08 2b 06 01 05 05 07 03 01 30 12 06 03 55 1d 13 01 01 ff 04 08 30 06 01 01 ff 02 01 00 30 1d 06 03 55 1d 0e 04 16 04 14 14 2e b3 17 b7 58 56 cb ae 50 09 40 e6 1f af 9d 8b 14 c2 c6 30 1f 06 03 55 1d 23 04 18 30 16 80 14 79 b4 59 e6 7b b6 e5 e4 01 73 80 08 88 c8 1a 58 f6 e9 9b 6e 30 32 06 08 2b 06 01 05 05 07 01 01 04 26 30 24 30 22 06 08 2b 06 01 05 05 07 30 02 86 16 68 74 74 70 3a 2f 2f 78 31 2e 69 2e 6c 65 6e 63 72 2e 6f 72 67 2f 30 27 06 03 55 1d 1f 04 20 30 1e 30 1c a0 1a a0 18 86 16 68 74 74 70 3a 2f 2f 78 31 2e 63 2e 6c 65 6e 63 72 2e 6f 72 67 2f 30 22 06 03 55 1d 20 04 1b 30 19 30 08 06 06 67 81 0c 01 02 01 30 0d 06 0b 2b 06 01 04 01 82 df [... cut ...] e 57 e8 36 0e f0 0e 23 aa fd 88 97 d0 e3 5c 0e 94 49 b5 b5 17 35 d2 2e bf 4e 85 ef 18 e0 85 92 eb 06 3b 6c 29 23 09 60 dc 45 02 4c 12 18 3b e9 fb 0e de dc 44 f8 58 98 ae ea bd 45 45 a1 88 5d 66 ca fe 10 e9 6f 82 c8 11 42 0d fb e9 ec e3 86 00 de 9d 10 e3 38 fa a4 7d b1 d8 e8 49 82 84 06 9b 2b e8 6b 4f 01 0c 38 77 2e f9 dd e7 39
received 4009 bytes from 3

send: 15 03 03 00 02 02 2a
sent 7 bytes to 3

send: 15 03 03 00 02 02 30
sent 7 bytes to 3

ERROR: failed to connect to wolfSSL, error -188
```

philljj, 2022-10-04:
HI @jawj,

Could you share your C file and more reproduction steps? I could not reproduce on my M1 Mac.

Also, what architecture is your Mac?

jawj, 2022-10-05:
Sure. 

Please see this repo: https://github.com/jawj/wolfssl-disable-asm-trouble

My Mac is an Intel (the README of the repo has more details).

philljj, 2022-10-07:
Hello @jawj, this looks to be a configure options issue.

On my Mac the issue is resolved with this additional configure option:
`--enable-fastmath`

Please let me know if this resolves your problem.

jawj, 2022-10-07:
Thanks, yes: adding `--enable-fastmath` does make it work for me both natively on my Mac and using emscripten.

I guess this should be either documented alongside `--disable-asm` or made automatic?