wolfSSL/wolfssl #5673: wolfSSL_set_msg_callback callback only called once

ilikdoge, 2022-10-07:
### Version

master

### Description

./configure --enable-fast-rsa --enable-intelasm --enable-sp --enable-sp-math --enable-sp-asm --enable-fastmath --enable-tls13 --enable-sni --enable-aesni --enable-alpn --enable-aesgcm  --enable-harden --enable-opensslextra --enable-aesgcm-stream

The callback set by the **wolfSSL_set_msg_callback** function only gets called once
I did a bit of digging and it seems like everything works internally, but

[image]

the encryptionOn stops all future callbacks. This doesn't seem to be the behavior with OpenSSL

ilikdoge, 2022-10-07:
This only happens with TLSv1.3 (and other places that set encryptionOn to 1)
With TLSv1.2 it works as expected

embhorn, 2022-10-07:
Hi @ilikdoge 

Right, encrypted messages should not be passed to the callback. This check was added in https://github.com/wolfSSL/wolfssl/pull/5492

Thanks,
@embhorn wolfSSL Support

ilikdoge, 2022-10-08:
Ok, interesting. I would like the data to be passed if possible, similar to OpenSSL. I looked more into the docs and found the connect_ex which seems to contain information about the individual packets, but I can't seem to enable the WOLFSSL_CALLBACKS define.

Is there a way to do this? I would prefer not to use OpenSSL if possible.

Thanks

closed 2022-10-09 by ilikdoge