wolfSSL/wolfssl #5685: OpenSSL License missed in openssl compatibility layer

KanPlus, 2022-10-11:
### Version

v5.5.1

### Description

The declaration in openssl compatibility layer match with the OpenSSL, but the OpenSSL License not added currently, this may violate the  OpenSSL License.

embhorn, 2022-10-11:
Hello @KanPlus 

The wolfSSL implementation of the compatibility layer was developed independently by us under our copyright.

Could you tell us a bit about your project?

Kind regards,
@embhorn wolfSSL Support

KanPlus, 2022-10-12:
> 
The project need evaluate whether the  compatibility layer should be used.
The compatibility layer will help the application migrate from OpenSSL with rebuild. but some declaration and meaning matched with OpenSSL.

embhorn, 2022-10-12:
Hi @KanPlus 

Thanks for sharing these details. So you are trying to build an application that links both wolfSSL (with the compatibility layer enabled) and OpenSSL? 

We do support a configuration with wolfSSL and OpenSSL (`--enable-opensslcoexist`), but the compatibility layer must be disabled, as that would obviously cause naming conflicts.

You are welcome to send an email to support@wolfssl.comm if you'd like to discuss you application goals in a less public forum.

Thanks,
@embhorn wolfSSL Support

KanPlus, 2022-10-13:
Hi @embhorn 
Thanks for sharing the informations.

closed 2022-10-13 by KanPlus