wolfSSL/wolfssl #5693: [Bug]: wolfssl 5.5.1 compilation fails on WOLFSSL_HAS_DEVCRYPTO_AES

zokl, 2022-10-12:
### Contact Details

zokl@atlas.cz

### Version

5.5.1

### Description

Compilation of the wolfssl 5.5.1 fails if devcrypto support is enabled (WOLFSSL_HAS_DEVCRYPTO_AES etc.).

Problem was described here: https://github.com/openwrt/openwrt/issues/10944 

Working patch:

```
diff --git a/wolfcrypt/src/port/devcrypto/wc_devcrypto.c b/wolfcrypt/src/port/devcrypto/wc_devcrypto.c
index 2a21d676c899..5c8d99217910 100644
--- a/wolfcrypt/src/port/devcrypto/wc_devcrypto.c
+++ b/wolfcrypt/src/port/devcrypto/wc_devcrypto.c
@@ -122,7 +122,7 @@ int wc_DevCryptoCreate(WC_CRYPTODEV* ctx, int type, byte* key, word32 keySz)
         case CRYPTO_SHA2_512_HMAC:
             ctx->sess.cipher = 0;
             ctx->sess.mac    = type;
-            ctx->sess.mackey    = (u_int8_t*)key;
+            ctx->sess.mackey    = (uint8_t*)key;
             ctx->sess.mackeylen = keySz;
             break;

```

### Relevant log output

```shell
-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-i
nitializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result
 -Wunused-variable -Wwrite-strings -fwrapv -MT wolfcrypt/src/port/devcrypto/src_libwolfssl_la-wc_devcrypto.lo -MD -MP -MF wolfcrypt/src/port/devcrypto/.deps/src_libwolfssl_la-wc_devcrypto.Tpo -c wolfcrypt/src/port/devcrypt
o/wc_devcrypto.c  -fPIC -DPIC -o wolfcrypt/src/port/devcrypto/.libs/src_libwolfssl_la-wc_devcrypto.o
In file included from ./wolfssl/wolfcrypt/wc_port.h:572,
                 from ./wolfssl/wolfcrypt/types.h:35,
                 from ./wolfssl/wolfcrypt/error-crypt.h:34,
                 from wolfcrypt/src/port/devcrypto/wc_devcrypto.c:33:
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h: In function 'snprintf':
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h:101:2: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
  return __orig_snprintf(__s, __n, __f, __builtin_va_arg_pack());
  ^~~~~~
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h: In function 'sprintf':
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h:110:3: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
   __r = __orig_snprintf(__s, __b, __f, __builtin_va_arg_pack());
   ^~~
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h:114:3: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
   __r = __orig_sprintf(__s, __f, __builtin_va_arg_pack());
   ^~~
wolfcrypt/src/port/devcrypto/wc_devcrypto.c: In function 'wc_DevCryptoCreate':
wolfcrypt/src/port/devcrypto/wc_devcrypto.c:125:36: error: 'u_int8_t' undeclared (first use in this function); did you mean 'uint8_t'?
             ctx->sess.mackey    = (u_int8_t*)key;
                                    ^~~~~~~~
                                    uint8_t
wolfcrypt/src/port/devcrypto/wc_devcrypto.c:125:36: note: each undeclared identifier is reported only once for each function it appears in
wolfcrypt/src/port/devcrypto/wc_devcrypto.c:125:45: error: expected expression before ')' token
             ctx->sess.mackey    = (u_int8_t*)key;
                                             ^
```

embhorn, 2022-10-12:
Hi @zokl 

Thanks for bringing this to our attention. I was unable to reproduce (my platform has these types defined in `sys/types.h`). That said, I see no reason for using the sys types here over the std C type (`uint8_t`).

I created #5695 to address this issue.

zokl, 2022-10-12:
Thank you very much.

ynezz, 2022-10-13:
>  I was unable to reproduce

I assume its related to cross compilation.

```
OpenWrt-libtool: compile:  x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -DBUILDING_WOLFSSL -I/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -DBUILDING_WOLFSSL -fvisibility=hidden -DHAVE_THREAD_LS -DNDEBUG -DNO_DO178 -DHAVE_REPRODUCIBLE_BUILD -DUSE_FAST_MATH -DWOLFSSL_X86_64_BUILD -pthread -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DTEST_IPV6 -DWOLFSSL_IPV6 -DWOLFSSL_WPAS -DHAVE_SECRET_CALLBACK -DWOLFSSL_PUBLIC_ECC_ADD_DBL -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_ALWAYS_KEEP_SNI -DHAVE_EX_DATA -DHAVE_EXT_CACHE -DWOLFSSL_EITHER_SIDE -DOPENSSL_EXTRA_X509_SMALL -DWOLFSSL_PUBLIC_MP -DWOLFSSL_DER_LOAD -DATOMIC_USER -DWOLFSSL_KEY_GEN -DWOLFSSL_DES_ECB -DKEEP_OUR_CERT -DKEEP_PEER_CERT -DHAVE_KEYING_MATERIAL -DNO_SESSION_CACHE_REF -DWOLFSSL_VALIDATE_ECC_IMPORT -DWC_CTC_NAME_SIZE=128 -DFORTRESS -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_AES_COUNTER -DWOLFSSL_AES_DIRECT -DWOLFSSL_DER_LOAD -DWOLFSSL_KEY_GEN -DHAVE_AESCCM -DWOLFSSL_USE_ALIGN -DWOLFSSL_DEVCRYPTO -DWOLFSSL_DEVCRYPTO_AES -DWOLFSSL_DEVCRYPTO_CBC -DWOLFSSL_AES_DIRECT -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DSESSION_CERTS -DHAVE_HKDF -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DHAVE_CURVE25519 -DWOLFSSL_ALLOW_TLSV10 -DWC_RSA_PSS -DWOLFSSL_PSS_LONG_SALT -DHAVE_ANON -DWOLFSSL_BASE64_ENCODE -DWOLFSSL_CMAC -DWOLFSSL_AES_DIRECT -DWOLFSSL_SHA3 -DWOLFSSL_NO_SHAKE128 -DWOLFSSL_NO_SHAKE256 -DHAVE_POLY1305 -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_OPENSSL_CMD -DHAVE_TLS_EXTENSIONS -DHAVE_CERTIFICATE_STATUS_REQUEST -DHAVE_TLS_EXTENSIONS -DHAVE_CERTIFICATE_STATUS_REQUEST_V2 -DHAVE_CRL -DHAVE_TLS_EXTENSIONS -DHAVE_SNI -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_FFDHE_2048 -DHAVE_SUPPORTED_CURVES -DWOLFSSL_TLS13 -DHAVE_TLS_EXTENSIONS -DHAVE_TLS_EXTENSIONS -DHAVE_SESSION_TICKET -DHAVE_EXTENDED_MASTER -DHAVE_TLS_EXTENSIONS -DHAVE_SNI -DHAVE_MAX_FRAGMENT -DHAVE_TRUNCATED_HMAC -DHAVE_ALPN -DHAVE_TRUSTED_CA -DHAVE_SUPPORTED_CURVES -DWOLFSSL_ALT_CERT_CHAINS -DHAVE_LIGHTY -DHAVE_WOLFSSL_SSL_H=1 -DHAVE_EX_DATA -DOPENSSL_ALL -DWOLFSSL_KEY_GEN -DOPENSSL_NO_SSL2 -DOPENSSL_NO_COMP -DOPENSSL_NO_SSL3 -DSINGLE_THREADED -DWOLFSSL_OPENVPN -DHAVE_KEYING_MATERIAL -DWOLFSSL_DES_ECB -DHAVE_EX_DATA -DWOLFSSL_KEY_GEN -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_ALWAYS_KEEP_SNI -DKEEP_OUR_CERT -DKEEP_PEER_CERT -DHAVE_EXT_CACHE -DHAVE_EX_DATA -DWOLFSSL_CERT_GEN -DHAVE_ENCRYPT_THEN_MAC -DWOLFSSL_KEY_GEN -DHAVE_STUNNEL -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_ALWAYS_KEEP_SNI -DHAVE_EX_DATA -DWOLFSSL_DES_ECB -DWOLFSSL_SIGNER_DER_CERT -DOPENSSL_COMPATIBLE_DEFAULTS -DWOLFSSL_TICKET_HAVE_ID -DWOLFSSL_ENCRYPTED_KEYS -DWOLFSSL_DEVCRYPTO_HASH_KEEP -DWC_NO_ASYNC_THREADING -DHAVE_AES_KEYWRAP -DWOLFSSL_AES_DIRECT -DHAVE_DH_DEFAULT_PARAMS -DWOLFSSL_TRUST_PEER_CERT -DNO_SESSION_CACHE_REF -DWOLFSSL_TLS13_NO_PEEK_HANDSHAKE_DONE -DWOLFSSL_ALT_CERT_CHAINS -DWOLFSSL_KEY_GEN -DWOLFSSL_CERT_GEN -DWOLFSSL_CERT_EXT -DHAVE_OCSP -DOPENSSL_ALL -DWOLFSSL_EITHER_SIDE -DWC_RSA_NO_PADDING -DWC_RSA_PSS -DWOLFSSL_PSS_LONG_SALT -DWOLFSSL_TICKET_HAVE_ID -DWOLFSSL_ERROR_CODE_OPENSSL -DWOLFSSL_CERT_NAME_ALL -DOPENSSL_EXTRA -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_VERIFY_CB_ALL_CERTS -DWOLFSSL_EXTRA_ALERTS -DHAVE_EXT_CACHE -DWOLFSSL_FORCE_CACHE_ON_TICKET -DWOLFSSL_AKID_NAME -DHAVE_CTS -DGCM_TABLE_4BIT -DHAVE_AESGCM -DHAVE_TLS_EXTENSIONS -DHAVE_SERVER_RENEGOTIATION_INFO -DHAVE_COMP_KEY -DWOLFSSL_ALLOW_RC4 -Wall -Wno-unused -O2 -DHAVE___UINT128_T=1 "-DLIBWOLFSSL_GLOBAL_EXTRA_CFLAGS=\" \"" -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/build_dir/target-x86_64_musl/wolfssl-regular/wolfssl-5.5.1-stable=wolfssl-5.5.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_NAMES -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv -MT wolfcrypt/src/port/devcrypto/src_libwolfssl_la-wc_devcrypto.lo -MD -MP -MF wolfcrypt/src/port/devcrypto/.deps/src_libwolfssl_la-wc_devcrypto.Tpo -c wolfcrypt/src/port/devcrypto/wc_devcrypto.c  -fPIC -DPIC -o wolfcrypt/src/port/devcrypto/.libs/src_libwolfssl_la-wc_devcrypto.o
In file included from ./wolfssl/wolfcrypt/wc_port.h:572,
                 from ./wolfssl/wolfcrypt/types.h:35,
                 from ./wolfssl/wolfcrypt/error-crypt.h:34,
                 from wolfcrypt/src/port/devcrypto/wc_devcrypto.c:33:
/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdio.h: In function 'snprintf':
/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdio.h:101:9: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
  101 |         return __orig_snprintf(__s, __n, __f, __builtin_va_arg_pack());
      |         ^~~~~~
/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdio.h: In function 'sprintf':
[... cut ...] w -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_NAMES    -Itoolchain-x86_64_gcc-11.3.0_musl/usr/include -Itoolchain-x86_64_gcc-11.3.0_musl/include/fortify -Itoolchain-x86_64_gcc-11.3.0_musl/include  conftest.c >&5
configure:8987: $? = 0
configure:8987: result: yes
```

> That said, I see no reason for using the sys types here over the std C type (uint8_t).

Indeed.

```
wolfssl.git$ git grep uint8_t | wc -l
1040

wolfssl.git$ git grep u_int8_t | wc -l
1
```

ynezz, 2022-10-13:
> I assume its related to cross compilation.

Seems like I wasn't that far off. IMO the culprit here is, that you're simply relying on implicit `<sys/types.h>` include in glibc's `<stdlib.h>`:

```
#include <stdlib.h>
# 33 "./wolfssl/wolfcrypt/memory.h"
# 1 "/usr/include/stdlib.h" 1 3 4
...
#include <sys/types.h>
# 394 "/usr/include/stdlib.h" 3 4
```

but musl's [<stdlib.h>](https://git.musl-libc.org/cgit/musl/tree/include/stdlib.h) doesn't include `<sys/types.h>` so the compilation fails.

closed 2022-12-01 by embhorn