wolfSSL/wolfssl #5732: AES GCM decryption reveals plaintext if tag verification fails

embyoshiv, 2022-10-25:
### Version

master (e7a121325bbb51560cf954dec66cba069aa199af)

### Description

The function AES_GCM_decrypt_C in the file wolfcrypt/src/aes.c decrypts the ciphertext, even if the authentication tag verification fails. The resulting plaintext can be found inside the output buffer. 

If I have read the NIST specification correctly (and various security proofs), the plaintext is not supposed to leak in case of an verification failure. 

As the GHASH is computed before anything is decrypted, the problem can be solved without any perfomance degradation by simply verifying the authentication tag earlier.

embhorn, 2022-10-25:
Hello @embyoshiv 

Could you please provide a PoC that reproduces the issue you are reporting?

embyoshiv, 2022-10-25:
I made a small POC for my problem. The configuration file and code should be attached (as normal text files). I took the configurations from an application for a cortex-m0+ MCU. Though it compiles and runs on a PC.

While making the POC, I noticed that we should have handled the verification differently in our code. However, I still wonder why the function even attempts to decrypt when an incorrect tag is seen.   

[aes_gcm_config.txt](https://github.com/wolfSSL/wolfssl/files/9863940/aes_gcm_config.txt)
[aes_gcm_poc_c.txt](https://github.com/wolfSSL/wolfssl/files/9863941/aes_gcm_poc_c.txt)

SparkiDev, 2022-10-25:
Hi @embyoshiv 

The code was changed as part of an effort to make it constant time.
I've added support for a define to enable early failure when the authentication tag is bad: WC_AES_GCM_DEC_AUTH_EARLY
(see #5735)
Use this define if you are not going to memset the output yourself.

Sean

embyoshiv, 2023-04-20:
> Hi @embyoshiv <https://github.com/embyoshiv>,
>
> I would like to close this issue.
> Please let us know if you have any more for this issue.
> If I do not hear back from you within a week I will have to close this
> issue.
>
> Sean

Hi,

You can close this issue.
Changes solved our problem.
Thank you.

Kind regards

SparkiDev, 2023-04-20:
Thanks @embyoshiv for helping us improve wolfSSL.

Sean :-)

closed 2023-04-20 by SparkiDev