wolfSSL/wolfssl #5756: Compilation failed with mingw64 on Windows 10

reapercreap, 2022-10-30:
### Version

v5.5.2-stable

### Description

I am failing to build wolfcrypt and I don't know what to do.

Configured with:
`./configure --enable-cryptonly --enable-static --host=x86_64-w64-mingw32`

Build:
```$ make
make -j13  all-recursive
make[1]: Entering directory '/home/Flo/wolfssl'
make[2]: Entering directory '/home/Flo/wolfssl'
make[2]: warning: -j13 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-sp_int.lo
  CC       wolfcrypt/src/src_libwolfssl_la-asn.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-kdf.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
wolfcrypt/src/sp_int.c:579:64: error: "_MSC_VER" is not defined, evaluates to 0 [-Werror=undef]
  579 | #if !defined(WOLFSSL_SP_DIV_WORD_HALF) && (!defined(_WIN64) || _MSC_VER >= 1920)
      |                                                                ^~~~~~~~
wolfcrypt/src/sp_int.c: In function 'sp_div_word':
wolfcrypt/src/sp_int.c:5787:5: error: unknown type name 'sp_int_word'; did you mean 'sp_isword'?
 5787 |     sp_int_word w;
      |     sp_isword
wolfcrypt/src/sp_int.c:5790:11: error: 'sp_int_word' undeclared (first use in this function); did you mean 'sp_div_word'?
 5790 |     w = ((sp_int_word)hi << SP_WORD_SIZE) | lo;
      |           ^~~~~~~~~~~
      |           sp_div_word
wolfcrypt/src/sp_int.c:5790:11: note: each undeclared identifier is reported only once for each function it appears in
wolfcrypt/src/sp_int.c:5790:23: error: expected ')' before 'hi'
 5790 |     w = ((sp_int_word)hi << SP_WORD_SIZE) | lo;
      |         ~             ^~
      |                       )
wolfcrypt/src/sp_int.c:5747:56: error: unused parameter 'hi' [-Werror=unused-parameter]
 5747 | static WC_INLINE sp_int_digit sp_div_word(sp_int_digit hi, sp_int_digit lo,
      |                                           ~~~~~~~~~~~~~^~
wolfcrypt/src/sp_int.c: In function '_sp_div_3':
wolfcrypt/src/sp_int.c:5831:18: error: variable 'l' set but not used [-Werror=unused-but-set-variable]
 5831 |     sp_int_digit l = 0;
      |                  ^
wolfcrypt/src/sp_int.c: In function '_sp_div_10':
wolfcrypt/src/sp_int.c:5892:18: error: variable 'l' set but not used [-Werror=unused-but-set-variable]
 5892 |     sp_int_digit l = 0;
      |                  ^
wolfcrypt/src/sp_int.c: In function '_sp_div_small':
wolfcrypt/src/sp_int.c:5955:18: error: variable 'l' set but not used [-Werror=unused-but-set-variable]
 5955 |     sp_int_digit l = 0;
      |                  ^
cc1.exe: all warnings being treated as errors
make[2]: *** [Makefile:5838: wolfcrypt/src/src_libwolfssl_la-sp_int.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/Flo/wolfssl'
make[1]: *** [Makefile:6947: all-recursive] Error 1
make[1]: Leaving directory '/home/Flo/wolfssl'
make: *** [Makefile:4025: all] Error 2
```
What am I missing? If more information is needed tell me pls. I want to get this to work under windows too for a school project. 

Thank you

reapercreap, 2022-10-31:
Going with:
`./configure --host=x86_64-w64-mingw32 --enable-fastmath --enable-cryptonly && make`
compiles fine. BUT:
```
#include <stdio.h>

#define HAVE_ECC

#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/settings.h>
#include <wolfssl/wolfcrypt/ecc.h>

void __attribute((constructor)) init(void){
    wolfCrypt_Init();
    printf("%s\n", "INIT");

}

void __attribute((destructor)) destroy(void){
    printf("%s\n", "CLEANUP");
    wolfCrypt_Cleanup();
}

int main(void) {
   return 0; 
}
```
compiles with no error but executing fails and it fails at `wolfCrypt_Init();` because it does not print out INIT.
After commenting `wolfCrypt_Init()` and  `wolfCrypt_Cleanup()` INIT and CLEANUP print correctly.

`gdb .\main.exe run` says:
```
[New Thread 3444.0x3280]
[New Thread 3444.0x1468]
[Thread 3444.0x3280 exited with code 3221225781]
[Thread 3444.0x1020 exited with code 3221225781]
During startup program exited with code 0xc0000135. 
```
I tried to compile my program with `-g` but same output from gdb as above.
I tried to configure and comfile with CFLAG like so: 
`./configure --host=x86_64-w64-mingw32 --enable-fastmath --enable-cryptonly CFLAGS="-g" && make`

But I can not see any debugging information in gdb.

I put the  `wolfCrypt_Init()` and  `wolfCrypt_Cleanup()` inside the main() function but this did not make a change.

embhorn, 2022-10-31:
Hello @reapercreap 

Please add `--enable-debug` to your configure line. Next add a call to start the debug log before calling the init function:
`wolfSSL_Debugging_ON();`

You should add configuration options to the configure command, not as a define in the application:
```
#define HAVE_ECC
```
ECC can be enabled with `./configure --enable-ecc`

Thanks,
@embhorn wolfSSL Support

reapercreap, 2022-10-31:
Hello @embhorn 

I added it: `./configure --host=x86_64-w64-mingw32 --enable-fastmath --enable-cryptonly --enable-debug && make`

I put configured options on pastebin so i dont spam the section here: https://pastebin.com/WiJ1xByh

```
#include <stdio.h>

#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/logging.h>
#include <wolfssl/wolfcrypt/settings.h>
#include <wolfssl/wolfcrypt/ecc.h>

/* void __attribute((constructor)) init(void){
    
    wolfCrypt_Init();
    printf("%s\n", "INIT");

}

void __attribute((destructor)) destroy(void){
    printf("%s\n", "CLEANUP");
    wolfCrypt_Cleanup();
} */

int main(void) {
    printf("%d\n", wolfSSL_Debugging_ON());

    wolfCrypt_Init();

    // code 

    wolfCrypt_Cleanup();

    return 0; 
}
```
I get no output from `printf("%d\n", wolfSSL_Debugging_ON());`
gdb:
``` 
[New Thread 6060.0x4504]
[New Thread 6060.0x16d8]
[Thread 6060.0x4504 exited with code 3221225781]
[Thread 6060.0x16d8 exited with code 3221225781]
During startup program exited with code 0xc0000135.
```

Pls let me know if you have an idea what to do.

embhorn, 2022-10-31:
Hi @reapercreap 

I see that the wolfCrypt test was built. Try executing that:
`./wolfcrypt/test/testwolfcrypt`

You should see lots of debug output and this message at the end:

```
Test complete
wolfSSL Entering wolfCrypt_Cleanup
Exiting main with return code: 0
```

reapercreap, 2022-10-31:
```
$ ./wolfcrypt/test/testwolfcrypt.exe
------------------------------------------------------------------------------
 wolfSSL version 5.5.2
------------------------------------------------------------------------------
error    test passed!
MEMORY   test passed!
[lots of Bad Base64 Decode data, too big]
[lots of Bad Base64 Decode data, too small]
[lots of Bad Base64 Decode bad character]
base64   test passed!
asn      test passed!
RANDOM   test passed!
MD5      test passed!
SHA      test passed!
SHA-224  test passed!
SHA-256  test passed!
SHA-384  test passed!
SHA-512  test passed!
SHA-3    test passed!
Hash     test passed!
HMAC-MD5 test passed!
HMAC-SHA test passed!
HMAC-SHA224 test passed!
HMAC-SHA256 test passed!
HMAC-SHA384 test passed!
HMAC-SHA512 test passed!
HMAC-SHA3   test passed!
GMAC     test passed!
Chacha   test passed!
POLY1305 test passed!
ChaCha20-Poly1305 AEAD test passed!
AES      test passed!
AES192   test passed!
AES256   test passed!
AES-GCM  test passed!
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
can't open ./certs/client-key.der, Please run from wolfSSL home dir error = -40
RSA      test failed!
 error = -7901
wolfSSL Entering wolfCrypt_Cleanup
Exiting main with return code: -1
```
```
$ pwd && ls certs/client-key.der
/home/Flo/wolfssl
certs/client-key.der
```
I dont find any information this error code.

chr0meice2, 2022-11-05:
I have a similar problem and also stuck: https://github.com/wolfSSL/wolfssl/issues/5775

If find a solution will post back in this thread to let you know.  I have been using the exact same way to compile wolfSSL for a long time now in Windows and now it is broken for anything after 5.5.0

chr0meice2, 2022-11-05:
@reapercreap did you find a solution to your issue?  I was able to get everything working with 5.5.3 described here:
https://github.com/wolfSSL/wolfssl/issues/5775

reapercreap, 2022-11-05:
@chr0meice2  I tried it again with the LDFLAGS but I never got those errors like you but these with sp_int.c . I recompiled it but this code failed if I call something with no debugging info for gdb.

```
#include <stdio.h>

#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/settings.h>

#include <wolfssl/wolfcrypt/logging.h>
/*#include <wolfssl/wolfcrypt/ecc.h> */

/* void __attribute((constructor)) init(void){
    
    wolfCrypt_Init();
    printf("%s\n", "INIT");

}

void __attribute((destructor)) destroy(void){
    printf("%s\n", "CLEANUP");
    wolfCrypt_Cleanup();
} */

int main(void) {
    printf("%d\n", wolfSSL_Debugging_ON());
    printf("hoi\n");

    // code 

    return 0; 
}
```

Same is for wolfCrypt_Init(). Do I miss something? I configure with:
`./configure --host=x86_64-w64-mingw32 --enable-fastmath --enable-cryptonly --enable-debug LDFLAGS="-lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections"`
I only need wolfcrypt unlike you but I dont know why it is failing because inside the `testwolfcrypt.exe` wolfCrypt_Init() is called to and it works.

I saw that i got the error with `testwolfcrypt.exe` because the CERT_ROOT definition is not using the path in `test_paths.h` and also there I dont know how to activate that. If i copy the `testwolfcrypt.exe` into the directory of wolfssl where the directory `certs` is, the exe is not starting with no error. 

I DONT KNOOOOW XD
@chr0meice2 maybe you could try to compile it with my options and compile my "testprogram"? 

Thx for trying to help.

embhorn, 2022-11-08:
Hi @reapercreap 

The issue with SP math was resolved. If you would like to try the SP math speedups with MinGW you can try with this patch: https://github.com/wolfSSL/wolfssl/pull/5776

For the path issues, you can build with `--disable-filesystem` to use the cert buffers instead of the cert files.

Let us know if there are still questions.

Thanks,

embhorn, 2022-11-30:
Closing this issue as resolved, but please feel free to reopen if there are still questions.

closed 2022-11-30 by embhorn