wolfSSL/wolfssl #5759: undefined reference to `RAND_status' and few more

Jpgig, 2022-10-31:
### Version

wolfssl/INSTALL

### Description

I am following steps noted in 15 as below in the link(https://github.com/wolfSSL/wolfssl/blob/master/INSTALL)
15. Building with liboqs for TLS 1.3 [EXPERIMENTAL]
    $ mkdir ~/oqs
    $ cd ~/oqs
    $ git clone --single-branch https://github.com/open-quantum-safe/liboqs.git
    $ cd liboqs/
    $ git checkout af76ca3b1f2fbc1f4f0967595f3bb07692fb3d82
    $ mkdir build
    $ cd build
    $ cmake -DOQS_USE_OPENSSL=0 ..
    $ make all
    $ sudo make install

    And then for building wolfssl, the following is sufficient:

    $ cd wolfssl
    $ ./autogen.sh (Might not be necessary)
    $ ./configure --with-liboqs
    $ make all

When I am performing wolfssl make all command getting below errors:

/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_status'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_bytes'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_free'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptUpdate'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_aes_256_ecb'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptInit_ex'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_poll'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_new'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `ERR_print_errors_fp'
collect2: error: ld returned 1 exit status
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_status'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_bytes'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_free'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptUpdate'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_aes_256_ecb'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptInit_ex'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_poll'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_new'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `ERR_print_errors_fp'
make[2]: *** [Makefile:5214: wolfcrypt/test/testwolfcrypt] Error 1
make[2]: *** Waiting for unfinished jobs....
collect2: error: ld returned 1 exit status

anhu, 2022-10-31:
Hi @Jpgig , 

Thank you so much for your message! The only place that those functions are used is in liboqs when the usage of OpenSSL is enabled. For example, the only place that `RAND_status()` appears is in liboqs in `src/common/rand/rand.c` .  Please see the following snippet: 

```
#ifdef OQS_USE_OPENSSL
#define OQS_RAND_POLL_RETRY 3 // in case failure to get randomness is a temporary problem, allow some repeats
void OQS_randombytes_openssl(uint8_t *random_array, size_t bytes_to_read) {
        int rep = OQS_RAND_POLL_RETRY;
        SIZE_T_TO_INT_OR_EXIT(bytes_to_read, bytes_to_read_int)
        do {
                if (RAND_status() == 1) {
                        break;
                }
                RAND_poll();
        } while (rep-- >= 0);
        if (RAND_bytes(random_array, bytes_to_read_int) != 1) {
                fprintf(stderr, "No OpenSSL randomness retrieved. DRBG available?\n");
                // because of void signature we have no other way to signal the problem
                // we cannot possibly return without randomness
                exit(EXIT_FAILURE);
        }
}
#endif
```
Its usage is guarded by `OQS_USE_OPENSSL` but that is undefined as we used `cmake -DOQS_USE_OPENSSL=0 ..` .  

Perhaps you need to rebuild liboqs.  Might I suggest completely deleting the build directory:

```
rm -rf build
```
@Jpgig  can you please let us know a bit about yourself and your project?  At wolfSSL, we're always very interested to know how our code is being used.  

Warm regards, Anthony

anhu, 2022-10-31:
Perhaps you need to remove all previous work? 
```
$ sudo rm -rf /usr/local/lib/*
$ rm -rf ~/oqs
$ mkdir ~/oqs
$ cd ~/oqs
$ git clone --single-branch https://github.com/open-quantum-safe/liboqs.git
$ cd liboqs/
$ git checkout af76ca3b1f2fbc1f4f0967595f3bb07692fb3d82
$ mkdir build
$ cd build
$ cmake -DOQS_USE_OPENSSL=0 ..                       #check the output of this step carefully
$ make all
$ sudo make install
$ cd ~
$ git clone https://github.com/wolfSSL/wolfssl
$ cd wolfssl
$ ./autogen.sh
$ ./configure --with-liboqs
$ make all
```

Jpgig, 2022-10-31:
I have done all above steps as noted still facing the same issue:
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_status'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_bytes'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_free'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptUpdate'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_aes_256_ecb'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptInit_ex'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_poll'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_new'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `ERR_print_errors_fp'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:5208: wolfcrypt/benchmark/benchmark] Error 1
make[2]: *** Waiting for unfinished jobs....
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_status'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_bytes'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_free'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptUpdate'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_aes_256_ecb'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_EncryptInit_ex'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `RAND_poll'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `EVP_CIPHER_CTX_new'
/usr/bin/ld: src/.libs/libwolfssl.so: undefined reference to `ERR_print_errors_fp'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:5214: wolfcrypt/test/testwolfcrypt] Error 1
make[2]: Leaving directory '/home/jiten/wolfssl'
make[1]: *** [Makefile:7160: all-recursive] Error 1
make[1]: Leaving directory '/home/jiten/wolfssl'
make: *** [Makefile:4232: all] Error 2

anhu, 2022-10-31:
This is very odd.  I've just run through all the steps on my own machine and it built fine.  I even purposely added a syntax error in some code guarded by `OQS_USE_OPENSSL` to make sure it didn't get built in . Perhaps there is something unique about your system.  I'm running on a fairly standard Ubuntu OS with nothing really special installed. 

May I ask what you are running?

Jpgig, 2022-10-31:
I am also using Ubuntu 20.04 LTS.

anhu, 2022-10-31:
I cannot reproduce what you are seeing.  Can you add a syntax error to code that is guarded by `OQS_USE_OPENSSL` to ensure that you are correctly doing the configuration step?  In my first message, I give you a great place to add such an error.

Jpgig, 2022-11-01:
I followed below steps:
1) I added a syntax error to code that is guarded by OQS_USE_OPENSSL
2) cmake -DOQS_USE_OPENSSL=0 ..
3) make all
Compilation went successful.

To reproduce the syntax error I change as below
2) cmake -DOQS_USE_OPENSSL=1 ..
3) make all
Compilation failed at the line where I added syntax error.

This seems liboqs configuration is correct.

I can build correctly with given steps in Kali linux.

I found that in Ubantu with "--with-liboqs" setting testwolfcrypt file is not generated in wolfcrypt/test folder

anhu, 2022-11-01:
Hi @Jpgig , 

Thank you for all your steps you've performed and all the information you've provided. I'm sorry that you were not able to get it working on Ubuntu, but it seems it is a problem that is unique to your Ubuntu machine's configuration as you are able to proceed in Kali Linux.

Now that you have a path forward, I will close this issue. Best of luck!! 

Warm regards, Anthony

closed 2022-11-01 by anhu