wolfSSL/wolfssl #5775: Compilation failed

chr0meice2, 2022-11-05:
### Version

5.5.3

### Description

I have been using the same method to compile OpenSSL in mingw64 and it now fails in version 5.5.3

FreezeLvl7@C4 MSYS /c/wolfssl-5.5.3
$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-12.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/include --libexecdir=/mingw64/lib --enable-bootstrap --enable-checking=release --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --enable-lto --enable-libgomp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev6, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --disable-libstdcxx-debug --with-boot-ldflags=-static-libstdc++ --with-stage1-ldflags=-static-libstdc++
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.0 (Rev6, Built by MSYS2 project)

Errors on make:

FreezeLvl7@C4 MSYS /c/wolfssl-5.5.3
$ make
make -j5  all-recursive
make[1]: Entering directory '/c/wolfssl-5.5.3'
make[2]: Entering directory '/c/wolfssl-5.5.3'
make[2]: warning: -j5 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-kdf.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sp_int.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
wolfcrypt/src/sp_int.c:579:64: warning: "_MSC_VER" is not defined, evaluates to 0 [-Wundef]
  579 | #if !defined(WOLFSSL_SP_DIV_WORD_HALF) && (!defined(_WIN64) || _MSC_VER >= 1920)
      |                                                                ^~~~~~~~
wolfcrypt/src/sp_int.c: In function 'sp_div_word':
wolfcrypt/src/sp_int.c:5787:5: error: unknown type name 'sp_int_word'; did you mean 'sp_isword'?
 5787 |     sp_int_word w;
      |     ^~~~~~~~~~~
      |     sp_isword
wolfcrypt/src/sp_int.c:5790:11: error: 'sp_int_word' undeclared (first use in this function); did you mean 'sp_div_word'?
 5790 |     w = ((sp_int_word)hi << SP_WORD_SIZE) | lo;
      |           ^~~~~~~~~~~
      |           sp_div_word
wolfcrypt/src/sp_int.c:5790:11: note: each undeclared identifier is reported only once for each function it appears in
wolfcrypt/src/sp_int.c:5790:23: error: expected ')' before 'hi'
 5790 |     w = ((sp_int_word)hi << SP_WORD_SIZE) | lo;
      |         ~             ^~
      |                       )
wolfcrypt/src/sp_int.c:5747:56: warning: unused parameter 'hi' [-Wunused-parameter]
 5747 | static WC_INLINE sp_int_digit sp_div_word(sp_int_digit hi, sp_int_digit lo,
      |                                           ~~~~~~~~~~~~~^~
wolfcrypt/src/sp_int.c: In function '_sp_div_3':
wolfcrypt/src/sp_int.c:5831:18: warning: variable 'l' set but not used [-Wunused-but-set-variable]
 5831 |     sp_int_digit l = 0;
      |                  ^
wolfcrypt/src/sp_int.c: In function '_sp_div_10':
wolfcrypt/src/sp_int.c:5892:18: warning: variable 'l' set but not used [-Wunused-but-set-variable]
 5892 |     sp_int_digit l = 0;
      |                  ^
wolfcrypt/src/sp_int.c: In function '_sp_div_small':
wolfcrypt/src/sp_int.c:5955:18: warning: variable 'l' set but not used [-Wunused-but-set-variable]
 5955 |     sp_int_digit l = 0;
      |                  ^
make[2]: *** [Makefile:6052: wolfcrypt/src/src_libwolfssl_la-sp_int.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/c/wolfssl-5.5.3'
make[1]: *** [Makefile:7161: all-recursive] Error 1
make[1]: Leaving directory '/c/wolfssl-5.5.3'
make: *** [Makefile:4233: all] Error 2

It was configured with this, never had a problem in the past with other versions:

./configure --enable-tls13 --disable-errorstrings --disable-oldtls --enable-static --disable-crypttests --disable-examples

Please let me know if any tips how to fix

Thanks for the help

chr0meice2, 2022-11-05:
OK I think I got it to compile OK, will test if the lib works in my C++ program next

Here is the command line that works:

./configure --host=x86_64-w64-mingw32 --enable-fastmath --enable-tls13 --disable-errorstrings --disable-oldtls --enable-static --disable-crypttests --disable-examples

Note, this is probably but and should be fixed... I had to manually link to -lcrypt32 to get rid of this:

  CCLD     src/libwolfssl.la
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/.libs/libwolfssl_la-ssl.o:ssl.c:(.text+0x47d0): undefined reference to `__imp_CertEnumCertificatesInStore'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/.libs/libwolfssl_la-ssl.o:ssl.c:(.text+0x47f0): undefined reference to `__imp_CertOpenSystemStoreA'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/.libs/libwolfssl_la-ssl.o:ssl.c:(.text+0x48c7): undefined reference to `__imp_CertCloseStore'

I had to add this to my make file:

LDFLAGS =  -lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections

ls2_32 is added by wolfSSL but shouldn't it also add lcrypt32 if it can't compile without it?

My cflags have -O2
AM_FLAGS I add these like usual: -DWOLFSSL_STATIC_DH -DWOLFSSL_STATIC_RSA -DNO_WOLFSSL_SERVER

I will report back my c++ works like usual , but appears to compile now

chr0meice2, 2022-11-05:
Confirmed everything works OK now.  Recommend updating the build process to automatically link to lcrypt32 similar to how lws2_32 is being added now=)

SparkiDev, 2022-11-06:
Hi @chr0meice2,

sp_int.c hasn't been tested on mingw64 with the latest changes.
Would you be willing to work through the changes needed to make it work?

Thanks,
Sean

chr0meice2, 2022-11-06:
Hey @SparkiDev , yeah sure please let me know what tests to try / etc.

SparkiDev, 2022-11-07:
Hi @chr0meice2,

Thanks for looking at this with me!

I've put a pull request up: #5776 

Let me know if this fixes things.

Thanks.
Sean :-)

vszakats, 2022-11-23:
@SparkiDev: I can confirm that #5776 fixed the compiler error. Thanks!

What remains is the missing `-lcrypt32` from `LIBS` when building with autotools.

This lines isn't understood by mingw-w64 to pull that lib automatically like MSVC does,
also resulting in a warning. That could be fixed by guarding it for `_MSC_VER`:
```
./wolfssl/src/ssl.c:168:9: warning: unknown pragma ignored [-Wunknown-pragmas]
#pragma comment(lib, "crypt32")
```

There are also these warnings with llvm/clang:
```
./wolfssl/src/ssl.c:168:9: warning: unknown pragma ignored [-Wunknown-pragmas]
#pragma comment(lib, "crypt32")
        ^
```
```
./wolfssl/src/ssl.c:8099:28: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
            while (certCtx = CertEnumCertificatesInStore(handle,
                   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./wolfssl/src/ssl.c:8099:28: note: place parentheses around the assignment to silence this warning
            while (certCtx = CertEnumCertificatesInStore(handle,
                           ^
                   (
./wolfssl/src/ssl.c:8099:28: note: use '==' to turn this assignment into an equality comparison
            while (certCtx = CertEnumCertificatesInStore(handle,
                           ^
                           ==
```
```
In file included from ../wolfssl/wolfcrypt/error-crypt.h:34:
../wolfssl/wolfcrypt/types.h:764:59: warning: format string is not a string literal [-Wformat-nonliteral]
                        ret = XVSNPRINTF(buffer, bufsize, format, ap);
```
```
../src/wolfio.c:1139:17: warning: result of comparison of constant 18446744073709551615 with expression of type 'SOCKET_T' (aka 'unsigned int') is always false [-Wtautologi
    if (*sockfd == SOCKET_INVALID)
        ~~~~~~~ ^  ~~~~~~~~~~~~~~
../src/wolfio.c:1177:19: warning: implicit conversion from 'SOCKET' (aka 'unsigned long long') to 'SOCKET_T' (aka 'unsigned int') changes value from 18446744073709551615 to
        *sockfd = SOCKET_INVALID;
                ~ ^~~~~~~~~~~~~~
../wolfssl/wolfio.h:363:32: note: expanded from macro 'SOCKET_INVALID'
        #define SOCKET_INVALID INVALID_SOCKET
                               ^~~~~~~~~~~~~~
/usr/local/opt/mingw-w64/toolchain-x86_64/x86_64-w64-mingw32/include/psdk_inc/_socket_types.h:16:24: note: expanded from macro 'INVALID_SOCKET'
#define INVALID_SOCKET  (SOCKET)(~0)
                        ^~~~~~~~~~~~
[...]
```

dgarske, 2022-11-28:
@philljj will you review these build warnings?

philljj, 2022-12-01:
I can reproduce the latest issues reported by @vszakats, I'll put together a PR.

landhb, 2022-12-03:
I think there is also a header include issue when cross compiling from Linux with mingw. Unless I'm doing something wrong, since Linux is case sensitive, when cross compiling:

```diff
diff --git a/src/ssl.c b/src/ssl.c
index 173d7e5e0..558de7126 100644
--- a/src/ssl.c
+++ b/src/ssl.c
@@ -166,7 +166,7 @@

 #ifdef _WIN32
 #include <windows.h>
-#include <Wincrypt.h>
+#include <wincrypt.h>
 #pragma comment(lib, "crypt32")
 #endif
 ```

philljj, 2022-12-03:
Thanks, I'll look at mingw issues on linux next.

closed 2022-12-06 by SparkiDev

vszakats, 2022-12-06:
The merged patch uses `__MINGW64__` wrongly in two places I believe (and in two more places it's unnecessary). See: https://github.com/wolfSSL/wolfssl/pull/5848#issuecomment-1340150007

vszakats, 2022-12-07:
This patch silences the `-Wparentheses` warning above:
```diff
--- a/ssl.c
+++ b/ssl.c
@@ -8101,8 +8101,8 @@
          i < sizeof(storeNames)/sizeof(*storeNames); ++i) {
         handle = CertOpenSystemStoreA(hProv, storeNames[i]);
         if (handle != NULL) {
-            while (certCtx = CertEnumCertificatesInStore(handle,
-                   certCtx)) {
+            while ((certCtx = CertEnumCertificatesInStore(handle,
+                   certCtx)) != NULL) {
                 if (certCtx->dwCertEncodingType == X509_ASN_ENCODING) {
                     if (ProcessBuffer(ctx, certCtx->pbCertEncoded,
                           certCtx->cbCertEncoded, WOLFSSL_FILETYPE_ASN1,
```

PR #5776 (2022-11-07): SP int: support mingw64 again
# Description

Check for _MSC_VER to determine that we are compiling for MSVC.

Fixes #5775

# Testing

Standard

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation