wolfSSL/wolfssl #5808: Bug Report

zhangslice, 2022-11-22:
### Version

Certificate verification Failed

### Description

 the ROOT certificate fails to be used to check the level-2 certificate, the return value is: err = -188, ASN no signer error to confirm failure. The same code can be successfully used with openssl, can you help to analyze the possible reasons?[https://gitee.com/zhangslice/kms](url)

kareem-wolfssl, 2022-11-22:
Hi @zhangslice,

Can you attach your wolfSSL build settings (user_settings.h or ./configure line)?
-188 means we were unable to find a CA certificate to validate your certificate's authenticity, most likely there is an issue with how your cert chain is being registered if only level-2 certificates are failing.  I am reviewing your code to confirm.

zhangslice, 2022-11-23:
Ok, thank you very much. I have uploaded the certificate including user_setting.h to the factory repository. Please help me check the possible causes[https://gitee.com/zhangslice/kms](url)

zhangslice, 2022-11-29:
Hi, @kareem-wolfssl  
      Could you help me analyze the possible causes?

kareem-wolfssl, 2022-12-02:
Hi @zhangslice

Thanks for your patience on this.
2 issues:
1)  You must include our build settings (<wolfssl/wolfcrypt/settings.h>) before all other SSL headers when building with wolfSSL.
2)  X509_STORE_CTX_trusted_stack does not currently work with X509_verify_cert in our OpenSSL compatibility layer.  You can instead use our CertManager API to load your trusted cert, then your example works correctly.
This patch corrects both issues, note that I've only changed CertChainCheck, you need to update any other uses of X509_STORE_CTX_trusted_stack in your code:

```
diff --git a/main.c b/main.c
index 88eb895..c8b5f23 100644
--- a/main.c
+++ b/main.c
@@ -2,8 +2,16 @@
 #include "stdio.h"
 #include <stdbool.h>
 #include <stdint.h>
-#include "openssl/pem.h"
-#include "openssl/x509.h"
+#ifdef HAVE_CONFIG_H
+    #include <config.h>
+#endif
+
+#ifndef WOLFSSL_USER_SETTINGS
+    #include <wolfssl/options.h>
+#endif
+#include <wolfssl/wolfcrypt/settings.h>
+#include <openssl/pem.h>
+#include <openssl/x509.h>
 #include "pl.h"
 //////////////////////////////////////////////////////////////////////////////
 //
@@ -344,7 +352,13 @@ bool CertChainCheck(const char* cert, char* certchain,int certchainLen, char * r
                sk_X509_push(certChain, root);
        }

-       X509_STORE_CTX_trusted_stack(csc, certChain);
+       //X509_STORE_CTX_trusted_stack(csc, certChain);
+       ret = wolfSSL_CertManagerLoadCABuffer(csc->store->cm, rootCert, certchainLen, SSL_FILETYPE_PEM);
+       if (ret != WOLFSSL_SUCCESS) {
+               ulErr = ERR_get_error();
+               pTmp = ERR_error_string(ulErr, err);
+               return ret;
+       }

        int i = X509_verify_cert(csc);
        if (i)
```

Here is a complete example of verifying a cert chain with wolfSSL: https://github.com/wolfSSL/wolfssl-examples/blob/master/certmanager/certloadverifybuffer.c

Let me know if you have any further questions.

zhangslice, 2022-12-05:
Hi, @kareem-wolfssl
       Thank you very much. The problem has been solved.

kareem-wolfssl, 2022-12-05:
Happy to help, and glad to hear the problem was solved.