wolfSSL/wolfssl #5813: [Bug]: DTLS1.3 - Server doesn't include key_share extention in hello_retry_request

Franko5040, 2022-11-23:
### Contact Details

fkrenzel@redhat.com

### Version

v5.5.3-stable

### Description

The client sends key_share extension but the server doesn't reply with it. As you can see in the log bellow, the server receives the extension from the first client hello, yet it is not send back possibly not even handled. It is only later during Server Hello that the extension is replied back to client.

```
client                                                              server
ClientHello
+key_update
+...
				-------->
					               Hello Retry Request
							           +cookie
                                                               -key_update
						    	              +...
				<--------
ClientHello
+cookie
+key_update
+...
				-------->
						              Server Hello
							        +key_share
							              +...
				<--------
```
* the `-` denotes the missing extention

reference:
* [rfc 8446 (TLS1.3  4.2.8)](https://www.rfc-editor.org/rfc/rfc8446#section-4.2.8) Key share
* [rfc 8446 (TLS1.3  4.1.4)](https://www.rfc-editor.org/rfc/rfc8446#section-4.1.4) Hello retry request

### Reproduction steps

./configure --enable-dtls --enable-dtls13 --enable-tls13 
make
./examples/client/client -u -v 4 127.0.0.1
./examples/server/server -u -v 4

### Relevant log output

```shell
Server log:
wolfSSL Entering wolfSSL_Init
wolfSSL Entering wolfCrypt_Init
wolfSSL Entering DTLSv1_3_server_method_ex
wolfSSL Entering wolfSSL_CTX_new_ex
wolfSSL Entering wolfSSL_CertManagerNew
wolfSSL Leaving WOLFSSL_CTX_new, return 0
wolfSSL Entering wolfSSL_CTX_use_certificate_chain_file
Getting dynamic buffer
wolfSSL Entering PemToDer
Growing Tmp Chain Buffer
Processing Cert Chain
wolfSSL Entering PemToDer
   Consumed another Cert in Chain
Finished Processing Cert Chain
Checking cert signature type
wolfSSL Entering GetExplicitVersion
wolfSSL Entering wc_GetSerialNumber
Got Cert Header
wolfSSL Entering GetObjectId()
Got Algo ID
[... 419 lines cut ...]
wolfSSL Entering SendAlert
growing output buffer
wolfSSL Entering BuildMessage
wolfSSL Entering BuildTls13Message
wolfSSL Entering EncryptTls13
wolfSSL Leaving BuildTls13Message, return 0
wolfSSL Entering EmbedSendTo()
Shrinking output buffer
wolfSSL Leaving SendAlert, return 0
wolfSSL Leaving SSL_shutdown(), return 2
wolfSSL Entering SSL_free
Free'ing server ssl
wolfSSL Entering DtlsMsgPoolReset()
wolfSSL Entering ClientSessionToSession
wolfSSL Entering Dtls13RtxFlushAcks
wolfSSL Entering Dtls13RtxFlushBuffered
CTX ref count not 0 yet, no free
wolfSSL Leaving SSL_free, return 0
wolfSSL Entering SSL_CTX_free
CTX ref count down to 0, doing full free
wolfSSL Entering wolfSSL_CertManagerFree
wolfSSL Leaving SSL_CTX_free, return 0
wolfSSL Entering wolfSSL_Cleanup
wolfSSL Entering wolfCrypt_Cleanup
```

dgarske, 2022-11-23:
@Franko5040 is it possible this issue is solved with PR https://github.com/wolfSSL/wolfssl/pull/5716 ?
@rizlik can you review this bug report?

rizlik, 2022-11-23:
Hi @Franko5040 
Thanks for the report. When the HelloRetryRequest is used to perform a return-routability check (as in this case) the KeyShare is not mandatory. Moreover, if the client already provided a good KeyShare the server MUST NOT send a KeyShare extension asking for the same key share in the HelloRetryRequest. Otherwise the (2) in :
In https://www.rfc-editor.org/rfc/rfc8446#section-4.2.8:
```
Upon receipt of this extension in a HelloRetryRequest, the client
   MUST verify that (1) the selected_group field corresponds to a group
   which was provided in the "supported_groups" extension in the
   original ClientHello and (2) the selected_group field does not
   correspond to a group which was provided in the "key_share" extension
   in the original ClientHello.
```
will be violated.
Of course, if the client doesn't provide a good key share in the first ClientHello, the KeyShare extension needs to be present in the HelloRetryRequest (together with the cookie). 

Does it make sense to you?
Marco

Franko5040, 2022-12-05:
I see,
Thanks for the explanation and you time!

closed 2022-12-05 by Franko5040