wolfSSL/wolfssl #5837: [Bug]: Library lacks `sp_rshd`

guidovranken, 2022-12-01:
### Version

Git repository checkout

### Description

`sp_rshd` is included in the headers, but lacking from the library if compiled with `--enable-sp --enable-sp-math`. This has been the case for 2-3 days.

### Reproduction steps

```
./configure --enable-static --enable-sp --enable-sp-math && make
```

Then compile this program:

```
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/integer.h>
int main(void)
{
    mp_rshd(0, 0);
    return 0;
}
```

### Relevant log output

```shell
/usr/bin/ld: /tmp/ccJkBKVi.o: in function `main':
x.c:(.text+0x13): undefined reference to `sp_rshd'
collect2: error: ld returned 1 exit status
```

SparkiDev, 2022-12-01:
Hi Guido,

When building with --enable-sp-math, this is asking that a minimal set of MP APIs are made available - only enough to support operations not being performed with --enable-sp.

When I reworked the sp_int.c code, I noticed that nothing was calling sp_rshd or even mp_rshd.
Therefore it no longer is available with --enable-sp-math.

It is available with --enable-sp-math-all which is the option to enable all MP APIs.

Please remove the mp_rshd from testing when using --enable-sp-math.

I've put up a pull request to not have sp_rshd available in the header when --enable-sp-math; #5839

Sean

guidovranken, 2022-12-05:
Thanks, fixed in my fuzzer.

closed 2022-12-05 by guidovranken