wolfSSL/wolfssl #5844: [Bug]: crash when using haproxy + wolfSSL

wlallemand, 2022-12-02:
### Contact Details

wlallemand@haproxy.com

### Version

1388e66

### Description

We experienced random crashes when injecting a lot of traffic  (about 70k req/s) into HAProxy in server mode. The process has 48 threads so it could be related to thread contention.

WolfSSL was built this way:
`./configure --enable-haproxy --enable-quic  --enable-sp=yes,asm --enable-intelasm --prefix=/opt/wolfssl-1388e66-intelasm`

the HAProxy version is commit  9d5e116 from master.

A coredump with the binary and libraries  is available here: http://master.formilux.org/pub/people/wlallemand/haproxy-wolfssl-crash.tgz

### Reproduction steps

The setup is quite difficult to reproduce, we could add some detail on how to inject if needed.

HAProxy was built this way:

```make  V=1 TARGET=linux-glibc USE_OPENSSL=1 USE_OPENSSL_WOLFSSL=1 WOLFSSL_INC=/opt/wolfssl-1388e66/include/ WOLFSSL_LIB=/opt/wolfssl-1388e66/lib ADDLIB='-Wl,-rpath=/opt/wolfssl-1388e66/lib'```

Started this way
```./haproxy -f ./h.conf -D```

And the traffic is injected on the port 4433.

### Relevant log output

```shell
This is the backtrace extracted from the coredump:

(gdb) bt
#0  0x00007fc09e8fe44d in wc_GetCurrentIdx () from ./opt/wolfssl-1388e66-intelasm/lib/libwolfssl.so.35
#1  0x00007fc09e97bf13 in wolfSSL_ERR_peek_error_line_data () from ./opt/wolfssl-1388e66-intelasm/lib/libwolfssl.so.35
#2  0x000055fcb4918563 in ssl_sock_handshake (flag=134217728, conn=<optimized out>) at src/ssl_sock.c:6180
#3  ssl_sock_io_cb (t=0x7fc00c0b1e90, context=0x7fc00c07be40, state=<optimized out>) at src/ssl_sock.c:6508
#4  0x000055fcb4a69e9e in run_tasks_from_lists (budgets=<optimized out>) at src/task.c:596
#5  0x000055fcb4a6a96c in process_runnable_tasks () at src/task.c:861
#6  0x000055fcb4a390ea in run_poll_loop () at src/haproxy.c:2891
#7  0x000055fcb4a396c9 in run_thread_poll_loop (data=<optimized out>) at src/haproxy.c:3080
#8  0x00007fc09ed01609 in start_thread () from ./lib/x86_64-linux-gnu/libpthread.so.0
#9  0x00007fc09e73f133 in clone () from ./lib/x86_64-linux-gnu/libc.so.6
```

dgarske, 2022-12-02:
@icing can you look into this?

icing, 2022-12-02:
Will do on Monday.

icing, 2022-12-06:
We have the problem analyzed and I am now working on a fix.

icing, 2022-12-06:
I opened PR #5856 to address this issue. Let's see what the wolfSSL CI says about the changes.

If you are eager to test this yourself, please note that you need to `autoreconf/configure` your build for the changes to take effect. In your configure output, you should see the line:

```
   * ERR Queues per Thread:      yes
```

wlallemand, 2022-12-06:
Thanks, I'm going to test this today.

wlallemand, 2022-12-06:
We've been injecting traffic for 2h with the same setup and your PR. We don't have crashes anymore, thanks for the quick fix!

icing, 2022-12-08:
@wlallemand FYI: I need to redo the implementation, since we encountered memory leaks on thread exit. The PR is now updated and passes our CI. If you could find the time to rerun your tests, that would be much appreciated.

wlallemand, 2022-12-08:
No pb, I'll try to do it this afternoon or tomorrow and let you the results, thanks!

icing, 2022-12-08:
> No pb, I'll try to do it this afternoon or tomorrow and let you know the results, thanks!

The PR has been merged to master.

wlallemand, 2022-12-09:
I just tested traffic injection on the master branch (85dc7a61c) and it seems to run smoothly. I'm closing the issue. Thanks !

closed 2022-12-09 by wlallemand