wolfSSL/wolfssl #5863: ClientHello extensions parse lack of inspection of repeatability

SmallTown123, 2022-12-08:
### Contact Details

small_town_123@163.com

### Version

5.5.1

### Description

When we tested TLS1.2 on the TLS handshake implementation part of wolfSSL,
we found that the Server lack of inspection of repeatability when parsing the
Extensions field of the ClientHello message, but responded normally.

This is inconsistent with the specification "There MUST NOT be more than one
extension of the same type." in section 7.4.1.4 of the RFC5246 documentation.

The resolution ambiguity is easily caused, and if the processing of an extension
is complex and consumes large resources, the malicious Client may attack the
Server by repeat someone Extensions.

### Reproduction steps

The problem can be verified by simply sending a ClientHello message to wolfSSL Server with duplicate extensions.

kareem-wolfssl, 2022-12-08:
Hi @SmallTown123 ,

Thanks for the report.  I am assigning this to another engineer who will look into it further.
It would be helpful if you can include your build settings (./configure or user_settings.h), and confirm if you can reproduce this on the latest master.

Thanks,
Kareem

SparkiDev, 2022-12-09:
@SmallTown123 

Thanks for the report!

A pull request has been put up to address the issue.
Feedback on the fix would be appreciated.

Thanks,
Sean

SmallTown123, 2022-12-09:
Hi @kareem-wolfssl,

Sorry for not replying to you at the first time. 
And the issue has been fixed by Sean. 

Thanks,
Town

SparkiDev, 2022-12-11:
@SmallTown123

Thanks for the feedback in the pull request!
The PR will be merged when I address review comments.

Sean :-)

PR #5872 (2022-12-12): TLS: detect duplicate known extensions
# Description

TLS specification requires that there not be more than one extension of the same type in a given extension block. E.g. ClientHello

Fixes #5863

# Testing

Added test case test_tls_exr_duplicate().

# Checklist

 - [x] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-12-11 by SparkiDev