wolfSSL/wolfssl #5880: [Bug]: Can't work correctly with BIO mem buffers

wlallemand, 2022-12-12:
### Contact Details

wlallemand@haproxy.com

### Version

50c5d61

### Description

We are having issue with HAProxy updating the certificates over the CLI. The CLI code massively uses the BIO__new_mem_buf() which seems to have problems with BIO_reset().

 ./configure --prefix=/opt/wolfssl --enable-haproxy --enable-quic

### Reproduction steps

Here a reproducer:

test_wolfssl.c
```
#ifndef OPENSSL
#include <wolfssl/options.h>
#endif
#include <openssl/ssl.h>

#include <openssl/bn.h>
#include <openssl/crypto.h>
#include <openssl/ssl.h>
#include <openssl/x509.h>
#include <openssl/x509v3.h>
#include <openssl/err.h>
#include <openssl/rand.h>
#include <openssl/hmac.h>

int main(int argc, char *argv[])
{
	BIO *in = NULL;
	SSL_CTX *ctx;
	int ret = 1;
	X509 *ca;
	X509 *cert = NULL;
	EVP_PKEY *key = NULL;

	const char buf_pem[] = "-----BEGIN CERTIFICATE-----\n"
"MIIBfzCCAQWgAwIBAgIUYDgleyiLJSKbSWzlU3PTCB/PPYIwCgYIKoZIzj0EAwIw\n"
"FDESMBAGA1UEAwwJbG9jYWxob3N0MB4XDTE5MTIxOTA5MzExMloXDTIwMDExODA5\n"
"MzExMlowFDESMBAGA1UEAwwJbG9jYWxob3N0MHYwEAYHKoZIzj0CAQYFK4EEACID\n"
"YgAEHNNG/ZSuS7CXvL03ye/Y+LpWnX818mnYkxqUQdFO2N1CO0p6kSIMHrzMQIRe\n"
"v3+j2g6drKehMGjBmeZJwsbD6nYyUO1z+0MatW5UiTMWFmPq4v08TDDtd8sNcWgs\n"
"SWrToxgwFjAUBgNVHREEDTALgglsb2NhbGhvc3QwCgYIKoZIzj0EAwIDaAAwZQIw\n"
"N2BdTJOH3BZlJ7HRIJNRC7jjByI9+QYAHiBoXmJVi9aoKd7OIz1Nb2DPe3QS1sDw\n"
"AjEA9KzI8BVIZJEmsVA6rs+vRjX0tUfBhD7BCHKas0roOny9Smj/TkBFxVTNnjzM\n"
"8iLn\n"
"-----END CERTIFICATE-----\n"
"-----BEGIN PRIVATE KEY-----\n"
"MIG2AgEAMBAGByqGSM49AgEGBSuBBAAiBIGeMIGbAgEBBDDZMkuztqaUgCAC9/7P\n"
"CsmlC2ac7rWerq5+NKbP0Cz1+mao6+F5Hc8DKNXHgi5GPr2hZANiAAQc00b9lK5L\n"
"sJe8vTfJ79j4uladfzXyadiTGpRB0U7Y3UI7SnqRIgwevMxAhF6/f6PaDp2sp6Ew\n"
"aMGZ5knCxsPqdjJQ7XP7Qxq1blSJMxYWY+ri/TxMMO13yw1xaCxJatM=\n"
"-----END PRIVATE KEY-----\n\n";

#ifndef OPENSSL
	wolfSSL_Init();
	wolfSSL_Debugging_ON();
#endif

	/* reading from a buffer */
	in = BIO_new_mem_buf(buf_pem, -1);
	if (in == NULL) {
		fprintf(stderr, "Can't allocate memory\n");
		goto end;
	}

	if (BIO_reset(in) == -1) {
		fprintf(stderr, "an error occurred while reading the file.\n");
		goto end;
	}

	cert = PEM_read_bio_X509_AUX(in, NULL, NULL, NULL);
	if (cert == NULL) {
		ret = ERR_get_error();
		fprintf(stderr, "unable to load certificate from buf : %s.\n", ERR_reason_error_string(ret));
		goto end;
	}

	if (BIO_reset(in) == -1) {
		fprintf(stderr, "an error occurred while reading the file.\n");
		goto end;
	}

	/* private key */
	key = PEM_read_bio_PrivateKey(in, NULL, NULL, NULL);
	if (key == NULL) {
		fprintf(stderr, "unable to load private key from buf: %s.\n", ERR_reason_error_string(ret));
		goto end;
	}

	/* creating the SSL_CTX */

	ctx = SSL_CTX_new(SSLv23_server_method());
	if (!ctx) {
		fprintf(stderr, "unable to allocate SSL context for cert.\n");
		goto end;
	}

	if (!SSL_CTX_use_certificate(ctx, cert)) {
		fprintf(stderr, "unable to load SSL certificate into SSL Context.\n");
		goto end;
	}

	if (SSL_CTX_use_PrivateKey(ctx, key) <= 0) {
		int ret;

		ret = ERR_get_error();
		fprintf(stderr, "unable to load SSL private key into SSL Context: %s.\n", ERR_reason_error_string(ret));
		goto end;
	}

	return 0;

end:

	return 1;
}
```

With OpenSSL:
```
$ gcc test_wolfssl.c -DOPENSSL -lssl  -lcrypto -o test_wolfssl
$ ./test_wolfssl && echo $?
0
```
With WolfSSL:
```
$ gcc test_wolfssl.c -I/opt/wolfssl/include/ -I/opt/wolfssl/include/wolfssl -L/opt/wolfssl/lib/ '-Wl,-rpath=/opt/wolfssl/lib' -lwolfssl -o  test_wolfssl
$ ./test_wolfssl && echo $?
unable to load certificate from buf : input state error.       
```

Removing the BIO_reset seems to fix the problem somehow. However, when swapping the read_bio_X509() with the read_bio_PrivateKey and removing the BIO_reset seems to provoke the same error.

philljj, 2022-12-13:
Thank you @wlallemand for the example code, I have reproduced this and it is a bug. I'll make a PR to address this.

tmael, 2022-12-14:
@wlallemand, Thanks for confirming the fix. Are there any other remaining issues we should look at?

wlallemand, 2022-12-14:
@tmael we still have 12 reg-tests ( `make reg-tests reg-tests/ssl/` in haproxy repository)  that are not working, but I didn't dig into them and I don't have reproducers for now.

tmael, 2022-12-14:
@wlallemand, Thanks for letting me know about the remaining issues. I'd be happy to dig into them. I just followed up with you in an email asking for some additional info.

tmael, 2023-02-23:
Fixed via https://github.com/wolfSSL/wolfssl/pull/5887

closed 2023-02-23 by tmael