wolfSSL/wolfssl #5883: Target-specific autogen? Espressif in particular.

gojimmypi, 2022-12-13:
### Version

Latest master branch

### Description

## Current State
Currently there are several different mechanisms to use wolfSSL in the Espressif ESP-IDF. The most common is to have wolfSSL installed to the `components` directory of either the local project or the ESP-IDF directory.

One can run [setup](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/setup.sh), which is certainly a reliable mechanism to copy only the required files, using typically just 10% of the entire repo size. The problem is that the installed library is then "disconnected" from GitHub. 

I encountered a problem recently where my work was lost when making changes to wolfSSL in my current project (the components directory), the re-running setup which over-wrote the component contents. With no source control, there's an obvious difficulty in both detecting changes as well as refreshing to recent code.

## Alternative Install

I also created a [install_wolfssl.cmd](https://github.com/gojimmypi/wolfssl/blob/Espressif_Multichip/IDE/Espressif/ESP-IDF/libs/install_wolfssl.cmd) that uses Git to fetch files with `fetch --depth 1` and sparse checkout. Although it has some merits such as being common to Windows, Mac and Linux, I've not created a PR with it as it is still a multi-step _install_.

I'd like to simply have the install be a `git clone` and `autogen` then voila, it's working.

## Proposed State

I'd like to update the wolfSSL [autogen.sh](https://github.com/wolfSSL/wolfssl/blob/master/autogen.sh) to detect the ESP-IDF environment and add a few extra steps to turn a fresh clone of wolfSSL from the respective `./components` directory into an operational Espressif component.

I'm considering adding something like this to the end of `autogen.sh`:

```
if [ -e "../../components/wolfssl/README.md" ]; then
    if [ "$IDF_PATH" == "" ]; then
        echo "It looks like autogen.sh is running from an Espressif component"
        echo "directory. However, the IDF_PATH is not set. If you would like"
        echo "to configure wolfSSL as an Espressif component, run the export."
    else
        echo "Found Espressif IDF_PATH=$IDF_PATH"
        echo "setup component makefiles here..."
    fi
else
    echo "Not an Espressif component"
fi
```

Any thoughts on this modification?  The line starting at "_setup component makefiles here_" would be code to add CMake files and such to make the wolfSSL clone look and work like an Espressif component. 

I'm looking for feedback before I actually implement this since there's nothing like it in the current `autogen`.

cc: @dgarske @kaleb-himes @JacobBarthelmeh

douzzer, 2023-01-19:
This sounds generally reasonable.  Though `echo "Not an Espressif component"` doesn't sound like the right idea -- `autogen.sh` shouldn't emit distracting noise.

Also, the test `-e "../../components/wolfssl/README.md"` looks fidgity -- is there a more decisive test you could substitute for that, preferably with a path that has "espressif" in it, or something similarly obvious and specific?

gojimmypi, 2023-01-29:
Provided https://github.com/wolfSSL/wolfssl/pull/6018 is merged, I think I'll abandon this idea of modification of the `autogen.sh` for an install. 

The "_it just works with no install_" with a [specially-crafted Cmakefiles.txt](https://github.com/gojimmypi/wolfssl/blob/Espressif_No_Install/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/CMakeLists.txt) is by far more preferable.

gojimmypi, 2023-02-16:
The intention of this PR was to minimize installation fuss for Espressif examples. Now that https://github.com/wolfSSL/wolfssl/pull/6018 has been merged, there's no reason to clutter `autogen.sh` with installation setups. 

The [Espressif ESP-IDF examples](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples) now work immediately after a `git clone https://github.com/wolfSSL/wolfssl.git` command with no [setup](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF#setup-for-linux) needed.

Note that other examples, such as [wolfssl-examples/ESP32](https://github.com/wolfSSL/wolfssl-examples/tree/master/ESP32) and [wolfssh-examples/Espressif](https://github.com/wolfSSL/wolfssh-examples/tree/main/Espressif) will still need to have the components installed until updated with the latest [components/wolfssl/CMakeLists.txt](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl) local project component files.

I'll leave this opened as a reminder.

gojimmypi, 2023-05-04:
I'm closing this issue as now obsolete and irrelevant. 

Instead, I've created a local project [CMakeLists.txt](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/CMakeLists.txt) added in https://github.com/wolfSSL/wolfssl/pull/6018 that allows for Espressif ESP-IDF projects to just work without any install or setup. See also https://github.com/wolfSSL/wolfssl/issues/6118 that will allow for project-specific `user_settings.h`.

See examples:

- [wolfssl_benchmark](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_benchmark) 
- [wolfssl_test](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test)  
- TLS [wolfssl_client](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_client) 
- TLS [wolfssl_server](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_server)

Sneak peek: upcoming [CMakeLists.txt](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/wolfssl_benchmark/components/wolfssl/CMakeLists.txt)

I've added https://github.com/wolfSSL/wolfssl-examples/issues/385 and https://github.com/wolfSSL/wolfssh/issues/514 as reminders to update those examples.

See https://github.com/wolfSSL/wolfssl/issues/6234 for a roadmap of Espressif updates.

closed 2023-05-04 by gojimmypi